eclipse-milo/milo #622: How 通

0DennisLee0, 2020-04-07:


kevinherron, 2020-04-07:
If you mean from inside the server, programmatically, yes it's easy: you just add nodes to your NodeManager or otherwise make them available in your AddressSpace implementation.

If you want a client to be able to add nodes you have to implement the NodeManagement services in your AddressSpace. There are not currently any examples of how to do this.

0DennisLee0, 2020-04-10:
> If you mean from inside the server, programmatically, yes it's easy: you just add nodes to your NodeManager or otherwise make them available in your AddressSpace implementation.
>
> If you want a [...]

Thanks for your kind replay~

closed 2020-04-10 by 0DennisLee0