eclipse-milo/milo #628: jaxb-api imported version incompatibility with OSGi and java 8

matteoporru, 2020-04-15:
**Describe the bug**
When upgrading from version 0.2.x to version 0.3.x a restriction on version of library jab-api has been introduced in the MANIFEST.MF files of artifacts bsd-parser-core, sdk-client, stack-core 
The new restriction is the following: javax.xml.bind;version=**"[2.3,3)"**
In previous versions such restriction was not specified.

**This restriction brakes the compatibility with OSGi+java8.**

Say I start my OSGi environment with my own bundle importing milo packages (typical use case).
The combination of OSGi+java8 brings natively the package javax.xml.bind in version 2.2.x (that to the best of my knowledge may not be hidden to OSGi classloader).
In order to start milo I have now to add also javax.xml.bind version 2.3.0.
This produces an "uses contraint violation" and my own bundle may not be started.
My own bundle does not have any necessity of using version 2.2.x. I have tried forcing it loading version 2.3.x with no difference.

**Expected behavior**
Unless there is a clear reason of excluding previous versions of jaxb-api, the restrictions on the version should be avoided in the Import section.
One possible fix that comes to my mind is by changing the file pom.xml (line 308) as follows (not tested):
```
                <plugin>
                    <groupId>org.apache.felix</groupId>
                    <artifactId>maven-bundle-plugin</artifactId>
                    <version>3.2.0</version>
                    <executions>
                        <execution>
                            <id>generate-manifest</id>
                            <phase>process-classes</phase>
                            <goals>
                                <goal>manifest</goal>
                            </goals>
                            <configuration>
                                <supportedProjectTypes>
                                    <supportedProjectType>jar</supportedProjectType>
                                    <supportedProjectType>bundle</supportedProjectType>
                                </supportedProjectTypes>
                                <instructions>
                                    <Import-Package>
                                        com.sun.management.*;resolution:=optional,
                                        !javax.annotation.*,
                                        javax.xml.bind,
                                        *
                                    </Import-Package>
```

**Logs and Packet Captures**
This is the error:
```
org.osgi.framework.BundleException: Could not resolve module: mybundle [265]
  Bundle was not resolved because of a uses contraint violation.
  org.osgi.service.resolver.ResolutionException: Uses constraint violation. Unable to resolve resource mybundle [osgi.identity; osgi.identity="mybundle"; type="osgi.bundle"; version:Version="4.55.0.SNAPSHOT"] because it is exposed to package 'javax.xml.bind' from resources org.eclipse.osgi [osgi.identity; osgi.identity="org.eclipse.osgi"; type="osgi.bundle"; version:Version="3.11.3.v20170209-1843"; singleton:="true"] and jaxb-api [osgi.identity; osgi.identity="jaxb-api"; type="osgi.bundle"; version:Version="2.3.0"] via two dependency chains.

Chain 1:
  mybundle [osgi.identity; osgi.identity="mybundle"; type="osgi.bundle"; version:Version="4.55.0.SNAPSHOT"]
    import: (&(osgi.wiring.package=javax.xml.bind)(&(version>=0.0.0)(!(version>=2.3.0))))
     |
    export: osgi.wiring.package: javax.xml.bind
  org.eclipse.osgi [osgi.identity; osgi.identity="org.eclipse.osgi"; type="osgi.bundle"; version:Version="3.11.3.v20170209-1843"; singleton:="true"]

Chain 2:
  mybundle [osgi.identity; osgi.identity="mybundle"; type="osgi.bundle"; version:Version="4.55.0.SNAPSHOT"]
    import: (&(osgi.wiring.package=org.eclipse.milo.opcua.sdk.client)(&(version>=0.3.0)(!(version>=1.0.0))))
     |
    export: osgi.wiring.package=org.eclipse.milo.opcua.sdk.client; uses:=org.eclipse.milo.opcua.binaryschema.parser
  org.eclipse.milo.sdk-client [osgi.identity; osgi.identity="org.eclipse.milo.sdk-client"; type="osgi.bundle"; version:Version="0.3.8"]
    import: (&(osgi.wiring.package=org.eclipse.milo.opcua.binaryschema.parser)(&(version>=0.3.0)(!(version>=1.0.0))))
     |
    export: osgi.wiring.package=org.eclipse.milo.opcua.binaryschema.parser; uses:=javax.xml.bind
  org.eclipse.milo.bsd-parser-core [osgi.identity; osgi.identity="org.eclipse.milo.bsd-parser-core"; type="osgi.bundle"; version:Version="0.3.8"]
    import: (&(osgi.wiring.package=javax.xml.bind)(&(version>=2.3.0)(!(version>=3.0.0))))
     |
    export: osgi.wiring.package: javax.xml.bind
  jaxb-api [osgi.identity; osgi.identity="jaxb-api"; type="osgi.bundle"; version:Version="2.3.0"]
	at org.eclipse.osgi.container.Module.start(Module.java:444)
	at org.eclipse.osgi.container.ModuleContainer$ContainerStartLevel.incStartLevel(ModuleContainer.java:1620)
	at org.eclipse.osgi.container.ModuleContainer$ContainerStartLevel.incStartLevel(ModuleContainer.java:1600)
	at org.eclipse.osgi.container.ModuleContainer$ContainerStartLevel.doContainerStartLevel(ModuleContainer.java:1571)
	at org.eclipse.osgi.container.ModuleContainer$ContainerStartLevel.dispatchEvent(ModuleContainer.java:1514)
	at org.eclipse.osgi.container.ModuleContainer$ContainerStartLevel.dispatchEvent(ModuleContainer.java:1)
	at org.eclipse.osgi.framework.eventmgr.EventManager.dispatchEvent(EventManager.java:230)
	at org.eclipse.osgi.framework.eventmgr.EventManager$EventThread.run(EventManager.java:340)
```

This is the relevant section of my MANIFEST.MF file:
```
Import-Package: javax.xml.bind,org.eclipse.milo.opcua.sdk.client;version="[0.3,1)",org.eclipse.milo.
 opcua.sdk.client.api;version="[0.3,1)",org.eclipse.milo.opcua.sdk.clien
 t.api.config;version="[0.3,1)",org.eclipse.milo.opcua.sdk.client.api.id
 entity;version="[0.3,1)",org.eclipse.milo.opcua.sdk.client.api.nodes;ve
 rsion="[0.3,1)",org.eclipse.milo.opcua.sdk.client.api.subscriptions;ver
 sion="[0.3,1)",org.eclipse.milo.opcua.sdk.client.subscriptions;version=
 "[0.3,1)",org.eclipse.milo.opcua.stack.client;version="[0.3,1)",org.ecl
 ipse.milo.opcua.stack.core;version="[0.3,1)",org.eclipse.milo.opcua.sta
 ck.core.security;version="[0.3,1)",org.eclipse.milo.opcua.stack.core.se
 rialization;version="[0.3,1)",org.eclipse.milo.opcua.stack.core.types.b
 uiltin;version="[0.3,1)",org.eclipse.milo.opcua.stack.core.types.builti
 n.unsigned;version="[0.3,1)",org.eclipse.milo.opcua.stack.core.types.en
 umerated;version="[0.3,1)",org.eclipse.milo.opcua.stack.core.types.stru
 ctured;version="[0.3,1)",...
```

This is the offending line in the generated MANIFEST.MF files of bsd-parser-core, sdk-client, stack-core:
Import-Package: ..., javax.xml.bind;version=**"[2.3,3)"**,

**Additional context**
Milo version: 0.3.8
java version: 1.8.0_231
OSGi version: 3.11.3.v20170209-1843 (tested with newer 3.15 version too)

I'm available for any help in testing this or provide any other information.

kevinherron, 2020-04-15:
Adding `java.xml.bind` to the `Import-Package` element still results in a jar file with the same import: `javax.xml.bind;version="[2.3,3)"`

I don't know enough (anything, really) about OSGi to fix this. Milo's support for OSGi has been done by volunteers who do need it.

I can say that depending on version 2.3+ of these JAXB libraries was intentional to support running on JDK 9+.

kevinherron, 2020-04-24:
Hey @ctron do you understand what's going on here enough to explain if it's something I can fix?

matteoporru, 2020-04-25:
@kevinherron, I have mitigated this on my side and was able to start the latest version of milo with OSGI+java 8. This is what I did, hoping it can help others too. 
I have identified what were the bundles started in my framework which needed the version 2.2. Actually they needed maximum 2.2.
It was a third party library with inside this Import hardcoded:
 javax.xml.bind;version="[0,2.2)"
Luckily, the supplier had developed a newer version without such restriction specified, so I could upgrade the library and move all dependencies to version 2.3.

Of course this may still be a problem for those who cannot upgrade all the dependencies to latest version of javax.xml.bind. But indeed it's lower priority than what I originally thought, since for many it should work.

Honestly I'm still convinced you can instruct the maven-bundle plugin in the main pom to force the Import being: java.xml.bind (without any version specified).
Alternatively you could try barcoding it like:
java.xml.bind;version="[0,3)"
That way you would be more permissive. Of course unless you need any functionality available only in 2.3+.

kevinherron, 2020-04-25:
Adding either of those results in an `Import-Package` statement that has `javax.xml.bind` listed twice:

```
Import-Package: com.google.common.base;version="[26.0,27)",com.google.co
 mmon.collect;version="[26.0,27)",com.google.common.io;version="[26.0,27
 )",com.google.common.primitives;version="[26.0,27)",io.netty.buffer;ver
 sion="[4.1,5)",io.netty.channel;version="[4.1,5)",io.netty.channel.nio;
 version="[4.1,5)",io.netty.util;version="[4.1,5)",io.netty.util.concurr
 ent;version="[4.1,5)",javax.crypto,javax.crypto.spec,javax.security.aut
 h.x500,javax.xml.bind;version="[2.3,3)",javax.xml.parsers,javax.xml.tra
 nsform,javax.xml.transform.dom,javax.xml.transform.stream,org.bouncycas
 tle.asn1;version="[1.61,2)",org.bouncycastle.asn1.pkcs;version="[1.61,2
 )",org.bouncycastle.asn1.x500;version="[1.61,2)",org.bouncycastle.asn1.
 x500.style;version="[1.61,2)",org.bouncycastle.asn1.x509;version="[1.61
 ,2)",org.bouncycastle.cert;version="[1.61,2)",org.bouncycastle.cert.jca
 jce;version="[1.61,2)",org.bouncycastle.jce.provider;version="[1.61,2)"
 ,org.bouncycastle.openssl;version="[1.61,2)",org.bouncycastle.operator;
 version="[1.61,2)",org.bouncycastle.operator.jcajce;version="[1.61,2)",
 org.bouncycastle.pkcs;version="[1.61,2)",org.bouncycastle.pkcs.jcajce;v
 ersion="[1.61,2)",org.bouncycastle.util;version="[1.61,2)",org.bouncyca
 stle.util.io.pem;version="[1.61,2)",org.eclipse.milo.opcua.stack.core,o
 rg.eclipse.milo.opcua.stack.core.channel.headers,org.eclipse.milo.opcua
 .stack.core.channel.messages,org.eclipse.milo.opcua.stack.core.security
 ,org.eclipse.milo.opcua.stack.core.serialization,org.eclipse.milo.opcua
 .stack.core.serialization.codecs,org.eclipse.milo.opcua.stack.core.type
 s,org.eclipse.milo.opcua.stack.core.types.builtin,org.eclipse.milo.opcu
 a.stack.core.types.builtin.unsigned,org.eclipse.milo.opcua.stack.core.t
 ypes.enumerated,org.eclipse.milo.opcua.stack.core.types.structured,org.
 eclipse.milo.opcua.stack.core.util,org.eclipse.milo.opcua.stack.core.ut
 il.annotations,org.slf4j;version="[1.7,2)",org.w3c.dom,org.xml.sax,java
 .xml.bind
```

Once for version [2.3,3] and once at the end either without a version or with a fixed version depending on which statement I added to the pom file.

ctron, 2020-04-27:
I was away from OSGi quite a while :grin: …

My first suggestion would have been, if you can upgrade to 2.3, you should :grin: There are still a bunch of bundles out there, which don't have proper versioning, and I think, from an OSGi perspective, they should be fixed. But you could argue about that a lot I guess.

The second thing might be to check of the APIs of Milo actually use the JAX stuff on its API. If that is not the case, then it might be an idea to confine it to the "bsd" bundle. So users of Milo would not see this dependency, and thus your "mybundle" wouldn't have the problem of referencing two bundles, with different versions. Because OSGi can handle two versions of the same package in a classpath. As long as no bundle has access to both a the same time.

Personally I would not go down the route of using non-versioned imports/exports again. I think that only created a mess in the first place.

kevinherron, 2020-04-27:
`stack-core` uses stuff from `javax.bind` for XML encoding/decoding, so essentially every module in Milo needs it.

ctron, 2020-04-27:
Then I would go back to: upgrade to version 2.3 :grin:

kevinherron, 2020-04-27:
Yeah, I don't think there's really anything I can change here. It needs to depend on 2.3 so Milo runs on Java9+ where all the JAXB stuff was moved into external dependencies.

I've committed to not requiring higher than Java 8 as a courtesy for the eventual 1.0 release but I care a lot more about whether Milo works on JDK9+ (really 11+).

closed 2020-05-20 by kevinherron