eclipse-milo/milo #629: Whether offline reconnection is supported

xiannvdezhu, 2020-04-17:
Hello! If I disconnect the server from PLC during operation, can I monitor dynamically and reconnect

kevinherron, 2020-04-17:
I'm not sure I understand your question.

If you're using the client SDK and connecting to an OPC UA server, it will automatically reconnect.

If you're using the server SDK then it's entirely up to you if you're connected to a PLC and whether or not you reconnect to the PLC if it is disconnected. This library has no code for dealing with PLCs, that's just simply one thing OPC UA servers are implemented to model/communicate with.

xiannvdezhu, 2020-04-17:
okay, thank you.
SDK, do you have any help documents

kevinherron, 2020-04-17:
> do you have any help documents

Nothing right now beyond the Javadoc on the APIs and the [example projects](https://github.com/eclipse/milo/tree/master/milo-examples) :disappointed:

xiannvdezhu, 2020-04-17:
strive,

closed 2020-04-17 by kevinherron