eclipse-milo/milo #635: Bulk subscription

xiannvdezhu, 2020-04-28:
hi:
How to implement batch subscription variables,
I can only subscribe to a single variable now.
If I want to batch subscribe,
I need to make a circular call to the subscription method in the method.
`public void subscription() { 
    try { 
      for (int i=1;i<=200;i++) { 
        opcUaOperationSupport.createSubscription(i,"\"Pin01\".\"SinaPos\"["+i+"]"+"."+"\"Vel\""); 
      } 
   } catch (Exception e) { 
      log.error("error" + e.getMessage());
   } 
}`
` public void createSubscription(Integer index, String param) {
      try {
         OpcUaClient client = ClientGen.opcUaClient;
         client.connect().get();
        UaSubscription subscription = client.getSubscriptionManager().createSubscription(1000.0).get();
        MonitoringParameters parameters = new MonitoringParameters(uint(1),1000.0,null,uint(10),true);
        List items = new ArrayList<>();
        NodeId nodeId = new NodeId(index, param);
        ReadValueId readValueId = new ReadValueId(nodeId, AttributeId.Value.uid(), null, null);
        MonitoredItemCreateRequest request = new MonitoredItemCreateRequest(readValueId, 
        MonitoringMode.Reporting, parameters);
        List requests = new ArrayList<>();
        requests.add(request);
        BiConsumer<UaMonitoredItem, Integer> onItemCreated =
                            (item, id) -> {
                                    item.setValueConsumer((item1, value) -> {
                                           log.info("Subscribe params: item={}, value={}", item1.getReadValueId().getNodeId(), value.getValue());
                      });
        };
       items = subscription.createMonitoredItems(TimestampsToReturn.Both,newArrayList(request),onItemCreated).get();
        for (UaMonitoredItem item : items) {
              if (item.getStatusCode().isGood()) {
                     log.info("success nodeId={}", item.getReadValueId().getNodeId());
              } else {
                     log.warn("errornodeId={} (status={})", item.getReadValueId().getNodeId(),item.getStatusCode());
              }
       }
    } catch (Exception e) {
        log.error("error"+e.getMessage());
    }
}`

xiannvdezhu, 2020-04-29:
UaServiceFaultException: status=Bad_TooManySubscriptions, message=The server has reached its maximum number of subscriptions.

xiannvdezhu, 2020-04-29:
Is the number of subscriptions fixed?

xiannvdezhu, 2020-04-29:
hi:
0.3.2->0.4.0
error : java.lang.NoSuchMethodError: org.bouncycastle.util.Arrays.areAllZeroes([BII)Z

xiannvdezhu, 2020-04-29:
help!Thank you!

kevinherron, 2020-04-29:
The number of subscriptions allowed depends on the server you're connected to.

I can't read the code you posted because you didn't format it correctly. Are you creating 1 subscription for each item instead of adding all the items to 1 subscription?

xiannvdezhu, 2020-04-29:
Yes, I created a subscription for each variable, but when the number of connections reached 20, the creation failed!

kevinherron, 2020-04-29:
You don't need to create a subscription for each variable.

Create 1 subscription, then add multiple monitored items to that subscription.

xiannvdezhu, 2020-04-29:
I see. I'll try! Thank you!

xiannvdezhu, 2020-04-29:
Thank you! It's a success!

closed 2020-04-29 by kevinherron

Ralfidze, 2020-08-05:
> You don't need to create a subscription for each variable.
> 
> Create 1 subscription, then add multiple monitored items to that subscription.
@xiannvdezhu @kevinherron 
Hey guys! 
Could you please help me how to add multiple monitored items to the subsctiption?

xiannvdezhu, 2020-08-06:
Create a data loading class! Inherit applicationrunner to implement the run method, which will be loaded automatically when the project is started! Query the existing items in the run method, put them into the set collection, create a main subscription method, and call the items in the set set set!

xiannvdezhu, 2020-08-06:
@Ralfidze

Ralfidze, 2020-08-06:
> Create a data loading class! Inherit applicationrunner to implement the run method, which will be loaded automatically when the project is started! Query the existing items in the run method, put them into the set collection, create a main subscription method, and call the items in the set set set!

Thank you @xiannvdezhu My question was related to events actually. Is it the same logic for the events ? Do I need to create another subscription for events ?

xiannvdezhu, 2020-08-07:
No, the connection method is a long connection, only a public method is needed to subscribe,

[test.txt](https://github.com/eclipse/milo/files/5038665/test.txt)
@Ralfidze

Ralfidze, 2020-08-07:
> No, the connection method is a long connection, only a public method is needed to subscribe,
> 
> [test.txt](https://github.com/eclipse/milo/files/5038665/test.txt)
> @Ralfidze

Thanks a lot @xiannvdezhu very useful )