eclipse-milo/milo #639: Simulate opc ua server deployment

YW767486516, 2020-05-07:
Hello,
I want to deploy an opcua server on the Internet. I opened port 12686 for the opcua server on the Internet. The address is ‘opc.tcp: //114.242.10.242: 12686’, but I can’t connect through the ip address. I can connect via LAN. Can you tell us what are the requirements for deploying opc ua server on the Internet? Best Regards.

Caused by: org.eclipse.milo.opcua.stack.core.UaException: io.netty.channel.ConnectTimeoutException: connection timed out: /192.168.1.5:12686
	at org.eclipse.milo.opcua.stack.client.transport.uasc.ClientChannelFsm$ClientChannelActions.lambda$connect$0(ClientChannelFsm.java:127) ~[stack-client-0.3.2.jar:0.3.2]
	at io.netty.util.concurrent.DefaultPromise.notifyListener0(DefaultPromise.java:502) ~[netty-common-4.1.34.Final.jar:4.1.34.Final]
	at io.netty.util.concurrent.DefaultPromise.notifyListeners0(DefaultPromise.java:495) ~[netty-common-4.1.34.Final.jar:4.1.34.Final]
	at io.netty.util.concurrent.DefaultPromise.notifyListenersNow(DefaultPromise.java:474) ~[netty-common-4.1.34.Final.jar:4.1.34.Final]
	at io.netty.util.concurrent.DefaultPromise.notifyListeners(DefaultPromise.java:415) ~[netty-common-4.1.34.Final.jar:4.1.34.Final]
	at io.netty.util.concurrent.DefaultPromise.setValue0(DefaultPromise.java:540) ~[netty-common-4.1.34.Final.jar:4.1.34.Final]
	at io.netty.util.concurrent.DefaultPromise.setFailure0(DefaultPromise.java:533) ~[netty-common-4.1.34.Final.jar:4.1.34.Final]
	at io.netty.util.concurrent.DefaultPromise.tryFailure(DefaultPromise.java:114) ~[netty-common-4.1.34.Final.jar:4.1.34.Final]
	at io.netty.channel.nio.AbstractNioChannel$AbstractNioUnsafe$1.run(AbstractNioChannel.java:269) ~[netty-transport-4.1.34.Final.jar:4.1.34.Final]
	at io.netty.util.concurrent.PromiseTask$RunnableAdapter.call(PromiseTask.java:38) ~[netty-common-4.1.34.Final.jar:4.1.34.Final]
	at io.netty.util.concurrent.ScheduledFutureTask.run(ScheduledFutureTask.java:127) ~[netty-common-4.1.34.Final.jar:4.1.34.Final]
	at io.netty.util.concurrent.AbstractEventExecutor.safeExecute(AbstractEventExecutor.java:163) ~[netty-common-4.1.34.Final.jar:4.1.34.Final]
	at io.netty.util.concurrent.SingleThreadEventExecutor.runAllTasks(SingleThreadEventExecutor.java:404) ~[netty-common-4.1.34.Final.jar:4.1.34.Final]
	at io.netty.channel.nio.NioEventLoop.run(NioEventLoop.java:495) ~[netty-transport-4.1.34.Final.jar:4.1.34.Final]
	at io.netty.util.concurrent.SingleThreadEventExecutor$5.run(SingleThreadEventExecutor.java:905) ~[netty-common-4.1.34.Final.jar:4.1.34.Final]
	at java.base/java.lang.Thread.run(Thread.java:834) ~[na:na]

kevinherron, 2020-05-07:
A couple things to check:
1) the you're binding to the network adapter or IP address that is reachable over the internet.
2) if the adapter from 1 is internal (or 0.0.0.0) and your server is reached via forwarding from an external IP, you're creating extra endpoint configurations that explicitly contain the external IP/hostname. See https://github.com/digitalpetri/opc-ua-demo-server/blob/62bb65dc4d666614700e45f0dc30ab1f84451ce7/src/main/kotlin/com/digitalpetri/opcua/server/DemoServer.kt#L291-L380 for an example of this.

YW767486516, 2020-05-08:
It is good! The default for hostnames is "HostnameUtil.getHostname ()". After adding the internet ip, I can access the opcua server.Best Regards.

closed 2020-05-08 by kevinherron

zty1010, 2020-07-23:
Hello, I seem to have the same problem as you.I put kepwear software in a cloud server as the server, I input the IP is opc.tcp: / / 114.55.210.92:49320, the error information is message=io.netty.channel. ConnectTimeoutException: connection timed out:/172.16.98.112:49320.So could you tell me your solution? Thank you very much
> 
> 
> It is good! The default for hostnames is "HostnameUtil.getHostname ()". After adding the internet ip, I can access the opcua server.Best Regards.

kevinherron, 2020-07-23:
@zty1010 you have a different problem, which is that Kepware is not configured to return the correct IP address.

If you can't configure Kepware correctly then you can use something like this approach in your client: https://stackoverflow.com/a/40555243/85927

zty1010, 2020-07-23:
> Since that post was written there is now an `EndpointUtil#updateUrl` method you can use

Hello, as you said the problem is different, So I want to describe my problem in detail.114.55.210.92 is WAN IP and 172.16.98.112 is LAN IP. I can connect to kepwear of cloud server through KEpwear, but I need to input public IP.
I use milo as the client, when I enter the opc tcp: / / 114.55.210.92:49320, the Error message is ：the Error running client example: UaException: status = Bad_Timeout, message=io.net ty. Channel. ConnectTimeoutException: connection timed out:/ 172.16.98.112:49320；when I enter the opc. tcp: / / 172.16.98.112:49320, the Error message is： the Error getting client: io.net ty. Channel. ConnectTimeoutException: connection timed out: / 172.16.98.11:49320.So how do I use the code in the link?
```
String endpointUrl = String.format(
        "%s://%s:%s%s",
        uri.getScheme(),
        hostname,
        uri.getPort(),
        uri.getPath()
    );
```
Replace %s://%s:%s%s with Intranet IP？
Thank  you very much for your reply and help

zty1010, 2020-07-27:
@kevinherron Hello, I still can't solve this problem. Can you help me

kevinherron, 2020-07-27:
With whatever `EndpointDescription` you've chosen, do something like the:

```java
EndpointDescription newEndpoint = EndpointUtil.updateUrl(endpoint, "114.55.210.92");
```

Then when you build your `OpcUaClientConfig` make sure to provide `newEndpoint`.

This assumes your WAN address is correct and the ports are open, etc...

zty1010, 2020-07-28:
> With whatever `EndpointDescription` you've chosen, do something like the:
>
> ```java
> EndpointDescription newEndpoint = EndpointUtil.updateUrl(endpoint, "114.55.210.92");
> ```
>
> Then when you build your [...]

The problem has been solved. Thank you very much for your help