eclipse-milo/milo #640: Is it supported opc AE？

mumugong, 2020-05-20:
I want  to read a alarms & events , Does milo support it?

kevinherron, 2020-05-20:
Yes, as a client it's just an event subscription:

https://github.com/eclipse/milo/blob/master/milo-examples/client-examples/src/main/java/org/eclipse/milo/examples/client/EventSubscriptionExample.java

https://github.com/eclipse/milo/blob/master/milo-examples/client-examples/src/main/java/org/eclipse/milo/examples/client/EventSubscriptionExampleProsys.java

closed 2020-05-20 by kevinherron

mumugong, 2020-05-21:
I want to connect to the local kepware AE ,
,What needs to be done? 
I modify the ClientExample 
 `default String getEndpointUrl() {
        return "opc.tcp://127.0.0.1:49320";
    }`
and receive nothing?

mumugong, 2020-05-21:
> Yes, as a client it's just an event subscription: [...]

> Yes, as a client it's just an event subscription: [...]

I want to connect to the local kepware AE ,
,What needs to be done?
I modify the ClientExample
default String getEndpointUrl() { return "opc.tcp://127.0.0.1:49320"; }
and receive nothing?

mumugong, 2020-05-21:
> Yes, as a client it's just an event subscription: [...]

I want to connect to the local kepware AE ,
,What needs to be done?
I modify the ClientExample
default String getEndpointUrl() { return "opc.tcp://127.0.0.1:49320"; }
and receive nothing?

kevinherron, 2020-05-21:
Kepware’s AE support is OPC Classic only as far as I know. It is not OPC UA Alarms and Conditions.