eclipse-milo/milo #641: HostnameUtil.getHostnames is slow

arxx-opus, 2020-05-20:
On my PC with several network cards, HostnameUtil.getHostnames () takes 37" 

It hangs on isLoopbackAddress()

```
public static Set<String> getHostnames(String address, boolean includeLoopback) {
.....
    for (NetworkInterface ni : Collections.list(nis)) {
        Collections.list(ni.getInetAddresses()).forEach(ia -> {
            if (ia instanceof Inet4Address) {
                boolean loopback = ia.isLoopbackAddress();  // hangs here
                if (!loopback || includeLoopback) {
                    hostnames.add(ia.getHostName());
                    hostnames.add(ia.getHostAddress());
                    hostnames.add(ia.getCanonicalHostName());
                }
            }
        });
.....
```

This statement is called even if `includeLoopback` is `true`.
Improvement
```
        if (ia instanceof Inet4Address) {
                boolean add = includeLoopback ? true : ! ia.isLoopbackAddress();
                if ( add ) {
                    hostnames.add(ia.getHostName());
                    hostnames.add(ia.getHostAddress());
                    hostnames.add(ia.getCanonicalHostName());
                }
            }

```

kevinherron, 2020-05-20:
Okay, I can short-circuit the call to `isLoopbackAddress()` when `includeLoopback`, but I can't do anything about that operation being slow on your system for some reason.

arxx-opus, 2020-05-20:
Thank you. 
By calling  HostnameUtil.getHostnames () with  includeLoopback = true  we avoid long waits like those on my troubled PC.

commit 8ff49516 (2020-05-20): Short-circuit call to InetAddress.isLoopbackAddress()

Prevents unnecessary call to isLoopbackAddress() when includeLoopback
parameter is true because it can be a slow operation on some systems.

fixes #641
PR #643 (2020-05-20): Short-circuit call to InetAddress.isLoopbackAddress()
Prevents unnecessary call to isLoopbackAddress() when includeLoopback
parameter is true because it can be a slow operation on some systems.

fixes #641
commit 484dbc1e (2020-05-20): Short-circuit call to InetAddress.isLoopbackAddress() (#643)

Prevents unnecessary call to isLoopbackAddress() when includeLoopback
parameter is true because it can be a slow operation on some systems.

fixes #641
closed 2020-05-20 by kevinherron