eclipse-milo/milo #642: Does milo support histroy reading now?

0DennisLee0, 2020-05-20:
Hi, kevin, I was using the history read of the ua client. Below is my code based on the **HistoryReadExampleProsys**.

```java
@Override
    public void run(OpcUaClient client, CompletableFuture<OpcUaClient> future) throws Exception {
        client.connect().get();

        HistoryReadDetails historyReadDetails = new ReadRawModifiedDetails(
            false,
            DateTime.MIN_VALUE,
            DateTime.now(),
            uint(0),
            true
        );

        List<HistoryReadValueId> historyReadValueIds = IntStream.rangeClosed(2004, 2006).mapToObj(value -> new HistoryReadValueId(
                new NodeId(5, value),
                null,
                QualifiedName.NULL_VALUE,
                ByteString.NULL_VALUE
        )).collect(Collectors.toList());

        HistoryReadResponse historyReadResponse = client.historyRead(
            historyReadDetails,
            TimestampsToReturn.Server,
            false,
            historyReadValueIds
        ).get();

        HistoryReadResult[] historyReadResults = historyReadResponse.getResults();

        if (historyReadResults != null) {
            HistoryReadResult historyReadResult = historyReadResults[0];
            HistoryData historyData = (HistoryData) historyReadResult.getHistoryData().decode(
                client.getSerializationContext()
            );

            List<DataValue> dataValues = l(historyData.getDataValues());

            dataValues.forEach(v -> System.out.println("value=" + v));
        }

        future.complete(client);
    }
```

I have set up 3 nodes support the history R&W and the historizing bit is true, as well.
However, I got this exception:
```
org.eclipse.milo.opcua.stack.core.UaSerializationException: no codec registered for encodingId=NodeId{ns=0, id=0}
	at org.eclipse.milo.opcua.stack.core.types.OpcUaDefaultBinaryEncoding.decode(OpcUaDefaultBinaryEncoding.java:103)
	at org.eclipse.milo.opcua.stack.core.types.builtin.ExtensionObject.lambda$decode$0(ExtensionObject.java:127)
	at org.eclipse.milo.opcua.stack.core.util.Lazy.maybeCompute(Lazy.java:49)
	at org.eclipse.milo.opcua.stack.core.util.Lazy.getOrCompute(Lazy.java:35)
	at org.eclipse.milo.opcua.stack.core.types.builtin.ExtensionObject.decode(ExtensionObject.java:127)
	at org.eclipse.milo.opcua.stack.core.types.builtin.ExtensionObject.decode(ExtensionObject.java:118)
	at com.dennis.client.HistoryReadExample.run(HistoryReadExample.java:67)
	at com.dennis.client.ClientExampleRunner.run(ClientExampleRunner.java:119)
	at com.dennis.client.HistoryReadExample.main(HistoryReadExample.java:33)
22:31:09.606 [ForkJoinPool.commonPool-worker-1] ERROR c.dennis.client.ClientExampleRunner - Error running example: no codec registered for encodingId=NodeId{ns=0, id=0}
org.eclipse.milo.opcua.stack.core.UaSerializationException: no codec registered for encodingId=NodeId{ns=0, id=0}
	at org.eclipse.milo.opcua.stack.core.types.OpcUaDefaultBinaryEncoding.decode(OpcUaDefaultBinaryEncoding.java:103)
	at org.eclipse.milo.opcua.stack.core.types.builtin.ExtensionObject.lambda$decode$0(ExtensionObject.java:127)
	at org.eclipse.milo.opcua.stack.core.util.Lazy.maybeCompute(Lazy.java:49)
	at org.eclipse.milo.opcua.stack.core.util.Lazy.getOrCompute(Lazy.java:35)
	at org.eclipse.milo.opcua.stack.core.types.builtin.ExtensionObject.decode(ExtensionObject.java:127)
	at org.eclipse.milo.opcua.stack.core.types.builtin.ExtensionObject.decode(ExtensionObject.java:118)
	at com.dennis.client.HistoryReadExample.run(HistoryReadExample.java:67)
	at com.dennis.client.ClientExampleRunner.run(ClientExampleRunner.java:119)
	at com.dennis.client.HistoryReadExample.main(HistoryReadExample.java:33)
```
Then I read some issues submitted before and got to know milo used to not support the HDA in server. And I want to know if it supported now?
The nodes are parsed by the milo node parser.

kevinherron, 2020-05-20:
Milo's example server does not implement history. It's technically possible if you implement all of the services yourself but the SDK does not do anything for you.

The client has always supported it as long as you're connected to a server that actually implements it.

What server are you connected to when you get that error? It's strange that the encodingId for the data you are trying to decode is null (`NodeId{ns=0, id=0}`), I'm not sure if that's a bug in the server or what.

kevinherron, 2020-05-20:
Ah, I get the same error against the latest Prosys OPC UA Simulation Server.

It looks like the NodeId of the simulated counter has changed. If you update the NodeId to `ns=3;s=Counter` (in my server, anyway), the example works again.

 I'll look closer at what these errors mean and why they are happening.

kevinherron, 2020-05-20:
Okay, the example has been updated: https://github.com/eclipse/milo/commit/f06b3751cb9f9cdf57c5985ed9136ca2fadeb368

closed 2020-05-20 by kevinherron

0DennisLee0, 2020-05-21:
Thank you for your kindly reply