eclipse-milo/milo #644: Implement Reverse Connect Client Facet

GoetzGoerisch, 2020-05-29:
**Implement [Reverse Connect Client Facet](https://reference.opcfoundation.org/v104/Core/docs/Part7/6.6.75/)**

Addititional information:
[https://reference.opcfoundation.org/v104/Core/docs/Part6/7.1.3/](https://reference.opcfoundation.org/v104/Core/docs/Part6/7.1.3/)

Benefit:
For a larger installation of an aggregating OPC UA server on Azure, this feature would be quite helpful.
OPC UA servers from various locations and networks connect to this central installtion based on milo.

kevinherron, 2020-05-29:
Reverse connect is on my radar but it’s part of OPC UA 1.04, so it’s going to have to wait a while.

The priority right now is getting Milo to version 1.0 by September or sooner, implementing OPC UA 1.03.

Once that is done I will start a `dev/2.0` branch and begin working on OPC UA 1.04 support.

GoetzGoerisch, 2020-05-29:
@kevinherron anything I can help with?

kevinherron, 2020-05-29:
As far as helping with reverse connect or 1.04 - if you have some idea what the API might look like for that I’m open to suggestions. Other than that 1.04 work really needs to wait until the code generators I’m building are finished and new nodes and structures can be generated before anything else happens.

If you want to help the push towards the 1.0 release what I will need very soon is feedback on the new client API going into the 0.5.0 release, as well as feedback on any existing or missing APIs (using 0.5.0 as a starting point).

I’m working on client SDK documentation as I build the new API but there is none yet for the server, so suggestions or contributions there would be useful too.

What existing functionality Milo has is of sufficient quality for a 1.0 release. The big issue is I want to commit to a stable API for 1.0, so that’s where I need feedback from users other than myself.

closed 2020-06-16 by kevinherron

ctron, 2020-07-20:
I just wanted to check if that is still on your radar?

kevinherron, 2020-07-20:
@ctron sure, but it's blocked until Milo 1.0 is released and work on 2.0 implementing OPC UA 1.04 begins.

ctron, 2020-07-20:
Let me know if I can help!

mb34890, 2022-07-27:
Is reverse connect implemented in the mean time?