eclipse-milo/milo #651: Java11 Jaxb problems with 0.3.8: com.sun.xml.internal.bind.v2.ContextFactory

heywiki, 2020-06-26:
I've added all necessary dependencies including the new jaxb implementations, but i still get this error: The class seems not to be present in any dependency

```
Caused by: javax.xml.bind.JAXBException: Implementation of JAXB-API has not been found on module path or classpath.
 - with linked exception:
[java.lang.ClassNotFoundException: com.sun.xml.internal.bind.v2.ContextFactory]
	at javax.xml.bind.ContextFinder.newInstance(ContextFinder.java:278)
	at javax.xml.bind.ContextFinder.find(ContextFinder.java:421)
	at javax.xml.bind.JAXBContext.newInstance(JAXBContext.java:721)
	at javax.xml.bind.JAXBContext.newInstance(JAXBContext.java:662)
	at org.eclipse.milo.opcua.binaryschema.parser.BsdParser.parse(BsdParser.java:40)
	at org.eclipse.milo.opcua.sdk.client.DataTypeDictionaryReader.createDataTypeDictionary(DataTypeDictionaryReader.java:251)
	at org.eclipse.milo.opcua.sdk.client.DataTypeDictionaryReader.lambda$readDataTypeDictionary$6(DataTypeDictionaryReader.java:134)
	at java.base/java.util.concurrent.CompletableFuture$UniCompose.tryFire(CompletableFuture.java:1072)
	... 9 more
```

I found this post that tells me to provide a jaxb.properties file to indicate a different factory method.

https://stackoverflow.com/questions/58009038/eclipse-4-12-java-lang-classnotfoundexception-com-sun-xml-internal-bind-v2-co

```
javax.xml.bind.context.factory=com.sun.xml.bind.v2.JAXBContextFactory
```

I've located it under resources/jaxb.properties in my calling jar but without success. Where does this property file need to reside? Does it have to reside in the milo packages itself? Or can this be resolved differently?

kevinherron, 2020-06-26:
Sorry, I don't know what this error means.

What build system are you using? What version of Java are you trying to run this on?

heywiki, 2020-06-26:
```
java -version
openjdk version "11.0.6" 2020-01-14
OpenJDK Runtime Environment AdoptOpenJDK (build 11.0.6+10)
OpenJDK 64-Bit Server VM AdoptOpenJDK (build 11.0.6+10, mixed mode)
```

Its a jsp web application running in a docker container: tomcat:9.0.31-jdk11-adoptopenjdk-hotspot
The jar files are all there and if i incorporate the code from the BsdParser in my code (so in the same claspath) 
it works fine

```
JAXBContext context = JAXBContext.newInstance(ObjectFactory.class);
```

Question is how can i force newInstance to use a certain factory to work around this: could Milow make shure that the right implementation is used?

kevinherron, 2020-06-26:
I don't know why you're having an issue. We run Milo on JDK 11 without doing anything extra and there aren't any problems like this.

heywiki, 2020-06-26:
all jars are in one lib (not managed by maven) i already removed jaxb-impl-2.3.3.jar, no effect

heywiki, 2020-06-29:
nope, just tested it: all dependencies seem right (i downloaded them with maven, then manually checked and copied them) bsdparser 0.3.8: no change, still get this exception

kevinherron, 2020-06-29:
How did you end up with version 2.3.3 of these jars? Milo 0.3.8 uses version 2.3.2.

I still don’t know what this error could mean other than that you have multiple JAXB runtimes on your classpath.

heywiki, 2020-06-29:
Theres another jar that specifies 2.3.3 as its dependency, so i keep the larger number. How does Maven solve this?

kevinherron, 2020-06-29:
Using the later version is okay, you just don't want to end up with multiple copies of the same dependency with differing versions.

heywiki, 2020-06-29:
I removed all libraries with double entries. No change.

But i found a solution. I cloned the repo, switched to branch 0.3.8 and added the following line to BsdParser that sets the classloader to the one of the "com.sun.xml.bind.v2.ContextFactory".
```
    public DictionaryDescription parse(InputStream inputStream) throws JAXBException {
    	Thread.currentThread().setContextClassLoader(ContextFactory.class.getClassLoader());
    	JAXBContext context = JAXBContext.newInstance(ObjectFactory.class);

```
After building bsd-parser-core-0.3.8.jar and using them in the lib the error was gone. So something is switchting the classloader...

kevinherron, 2020-06-29:
> Null indicates the bootstrap classloader, so the normal one. So something sets a different classloader. Does "app" ring any bell?

You mentioned this is a JSP application, so probably something to do with the application server your app is running in.

heywiki, 2020-06-29:
Yea, Dockerfile "FROM tomcat:9.0.31-jdk11-adoptopenjdk-hotspot". I am just reading this: https://stackoverflow.com/questions/19559210/how-tomcat-classloader-separates-different-webapps-object-scope-in-same-jvm

So i might not be the only one that will face this issue, because if a webapp gets shipped the libs need to be compiled into the war anyway. Question is how milo can make shure that the right classloader is used.

kevinherron, 2020-07-01:
@heywiki instead of the change you made, can you try modifying BsdParser line 40 so creating the JAXBContext looks like this:
```java
JAXBContext context = JAXBContext.newInstance(
    "org.opcfoundation.opcua.binaryschema", 
    getClass().getClassLoader()
);
```

kevinherron, 2020-07-07:
@heywiki if you want to pursue this further let me know. If the above option works I can introduce an overload somewhere that allows the ClassLoader used by BsdParser to be set or passed in. I'm going to close this issue in the meantime.

closed 2020-07-07 by kevinherron