eclipse-milo/milo #652: How to extract registered servers details from a LDS server in Milo

jdhanasekar, 2020-06-26:
Hi Milo,
        I have registered a Milo server with openLDS server. When I call FindServerOnNetwork() API and I am getting output like this,
```
 Servers=[ServerOnNetwork{RecordId=0, ServerName=LDS-quanta, DiscoveryUrl=opc.tcp://quanta:4840, ServerCapabilities=[LDS]}]}

```

But I am not seeing a list of registered opc ua servers with LDS-quanta.  
I do not know how to extract those using the Milo Project. 
I am seeing the ```RegisteredServer``` class file.
 Will this help to extract registered server details from LDS ?

Please help me on this.

kevinherron, 2020-06-26:
If LDS-quanta is not the server you're already getting this response from, then you would connect to it and call FindServers or FindServersOnNetwork.

If you don't get any additional servers in response to that call then there probably aren't any registered.

closed 2020-06-26 by kevinherron

jdhanasekar, 2020-06-26:
Sorry, I did not understand. 
Please help me to understand. 

I have registered my Milo server with LDS-quanta (Both are running in same Linux host). 
I am calling FindServerOnNetwork() in Windows host like this,
```
FindServersOnNetworkRequest myServOnNetReq = new FindServersOnNetworkRequest(reqHead,startingRecordId,maxRecordsToReturn,serverCapabilityFilter);
FindServersOnNetworkResponse myServOnNetRes= (FindServersOnNetworkResponse) stackClient.sendRequest(myServOnNetReq).get();
```
and I am getting output like this,

```
 FindServersOnNetworkResponse{
ResponseHeader=ResponseHeader
{
Timestamp=DateTime { utcTime=132376456641219860, javaDate=Fri Jun 26 17:17:44 IST 2020}, 
RequestHandle=4, 
ServiceResult=StatusCode {name=Good, value=0x00000000, quality=good}, 
ServiceDiagnostics=null, 
StringTable=null, 
AdditionalHeader=ExtensionObject {encoded=ByteString{bytes=null}, 
encodingId=NodeId{ns=0, id=0}}}, 
LastCounterResetTime=DateTime{utcTime=132376456103686840, javaDate=Fri Jun 26 17:16:50 IST 2020}, 
Servers=[
ServerOnNetwork{
        RecordId=0, ServerName=LDS-quanta, DiscoveryUrl=opc.tcp://quanta:4840, ServerCapabilities=[LDS]}]}
```
**You want me to connect to LDS server (which enumerated by FindServerOnNetwork) like ReadExample.java  and extract details from it.** 

In open62541 project, ```discovery_client_find_servers``` is a feature which is extracting registered servers from open62541 LDS and providing details like this,

```
Server[0]: urn:open62541.example.local_discovery_server
	Name: open62541-based OPC UA Application
	Application URI: urn:open62541.example.local_discovery_server
	Product URI: http://open62541.org
	Type: Discovery Server
	Discovery URLs:
		[0]: opc.tcp://quanta:4840/

Server[1]: urn:eclipse:milo:examples:server
	Name: eclipse milo opc-ua server
	Application URI: urn:eclipse:milo:examples:server
	Product URI: https://opcfoundation.org/
	Type: Server
	Discovery URLs:
		[0]: opc.tcp://localhost.localdomain:12686/milo/discovery/

```
I want to display details like this.

kevinherron, 2020-06-26:
Running open62541 LDS and the Milo demo server in a VM, registered with the LDS.

```java
import java.util.Arrays;
import java.util.List;
import java.util.concurrent.ExecutionException;

import org.eclipse.milo.opcua.stack.client.DiscoveryClient;
import org.eclipse.milo.opcua.stack.core.types.structured.ApplicationDescription;

class Scratch {
    
    public static void main(String[] args) throws ExecutionException, InterruptedException {
        List<ApplicationDescription> applications =
            DiscoveryClient.findServers("opc.tcp://192.168.86.21:4840").get();

        for (ApplicationDescription application : applications) {
            System.out.println("Found server: " + application.getApplicationName().getText());
            System.out.println("\tapplicationType: " + application.getApplicationType());
            System.out.println("\tapplicationUri: " + application.getApplicationUri());
            System.out.println("\tproductUri: " + application.getProductUri());
            System.out.println("\tdiscoveryUrls: " + Arrays.toString(application.getDiscoveryUrls()));
        }
    }
    
}
```

Results:
```
Found server: open62541-based OPC UA Application
	applicationType: DiscoveryServer
	applicationUri: urn:open62541.example.local_discovery_server
	productUri: http://open62541.org
	discoveryUrls: [opc.tcp://ubuntu:4840/]
Found server: Eclipse Milo OPC UA Demo Server
	applicationType: Server
	applicationUri: urn:eclipse:milo:opcua:server:ad2afe0f-9e81-4efa-9e18-7fa1ccc23c7f
	productUri: https://github.com/digitalpetri/opc-ua-demo-server
	discoveryUrls: [opc.tcp://ubuntu:62541/milo/discovery, opc.tcp://127.0.1.1:62541/milo/discovery, opc.tcp://localhost:62541/milo/discovery, opc.tcp://127.0.0.1:62541/milo/discovery]
```

jdhanasekar, 2020-06-26:
Thank you so much...
Now it is working..

jdhanasekar, 2020-06-30:
Hi Kevin,

Once I found out each registered servers, is it possible to connect to those servers and browse through their nodes. ?.

kevinherron, 2020-06-30:
Each of those results you printed out has a discovery URL you can use to create a client, connect, and browse. You will have to make a client and connect to each server individually.

jdhanasekar, 2020-06-30:
You mean I have to create a single Milo client and this Milo client will connect to each server (Registered server of LDS). I should not create clients for every server..

kevinherron, 2020-06-30:
One `OpcUaClient` instance per server you want to connect to.

jdhanasekar, 2020-07-01:
Once Creating a ```OpcUaClient``` instance per server. After establishing connection, all registered server's ```nodes```  will be stored in my Milo server?. 
 Actually, I want to copy all the nodes from registered servers of LDS.  I want to hide all server details from other stakeholders.  A Stakeholder only has access to a Milo server (I/O), they will read or write a node(copied from registered servers) based on their permission provided. 

Please guide me. 
Please correct me, if my objectives are incorrect.

kevinherron, 2020-07-01:
No, just connecting doesn’t do anything for you in regards to a Milo server. 

What you’re talking about building is called an aggregating server and it’s much more involved than I’ll be able to help you with. You’re going to have to learn a lot more and figure this out on your own.

jdhanasekar, 2020-07-01:
Ok.. Where can I get initial help or documentation or example projects for building an Aggregate server. As of now , I am going to aggregate one or two opc ua servers only.  

I will put all my effort into learning, Please guide me. First, I will start with a small no. of servers. Once I get the knowledge, I will progress further.

jdhanasekar, 2020-07-02:
Instead of Aggregating servers, Can I do subscription of nodes from a registered server?. will this be almost equal to aggregating servers ?