eclipse-milo/milo #654: Kepserver software cannot connect to Milo-OpcUa-Server

YW767486516, 2020-06-30:
Hi,
My milo-opcua-server only uses'USER_TOKEN_POLICY_X509'. I can connect with milo-opcua-client, but I cannot connect with opcua-client of kepservers software.
I found that milo uses java.security.KeyPair, but opcfoundation uses org.opcfoundation.ua.transport.security, will it be for this reason?

Best regards
`
                // milo-opcua-server
                EndpointConfiguration.Builder builder = EndpointConfiguration.newBuilder()
                    .setBindAddress(bindAddress)
                    .setHostname(hostname)
                    .setPath("/milo")
                    .setCertificate(certificate)
                    .addTokenPolicies(
                        //USER_TOKEN_POLICY_ANONYMOUS,
                        //USER_TOKEN_POLICY_USERNAME,
                        USER_TOKEN_POLICY_X509);
                endpointConfigurations.add(buildTcpEndpoint(builder));
                endpointConfigurations.add(buildHttpsEndpoint(builder));

               // milo-opcua-client
               OpcUaClientConfigBuilder cfg = new OpcUaClientConfigBuilder();
               cfg.setEndpoint(findBest(endpoints))
                    .setRequestTimeout(uint(5000))
                    .setCertificate(cer)
                    .setKeyPair(keyPair)
                    .setIdentityProvider(new X509IdentityProvider(cer,keyPair.getPrivate()));

`

kevinherron, 2020-06-30:
Does Kepware’s OPC UA client support authentication via X509?

kevinherron, 2020-06-30:
Kepware’s client driver manual only mentions username/password support: https://www.kepware.com/getattachment/3c0687fe-fa1d-4833-8967-69cc63d291b8/opc-ua-client-manual.pdf#page14

YW767486516, 2020-07-01:
Thanks for your help!

closed 2020-07-01 by YW767486516