eclipse-milo/milo #655: The user identity token is valid but the server has rejected it with X509 certificate authentication

Abhilash1368, 2020-07-02:
Hello Kevin,
I currently running into following issue, trying from past two days but no luck, any help or reference documention is highly appreciated.
I am using eclipse-milo v0.4 SDK in client  and Prosys OPC UA server with certificate authentication enabled.
I am using KeystoreLoader class  to create  certificate and key pair and I have added this certificate to the trusted ones on the Prosys server and tried connecting using client but I am unable to authenticate successfully and it throws "status=Bad_IdentityTokenRejected, description=The user identity token is valid but the server has rejected it."

**Client configuration**
```
OpcUaClientConfigBuilder configBuilder = new OpcUaClientConfigBuilder();
            configBuilder.setApplicationName(LocalizedText.english("appName"));
            configBuilder.setIdentityProvider(new X509IdentityProvider(keyStoreLoader.getClientCertificate(), keyStoreLoader.getClientKeyPair().getPrivate()));
            configBuilder.setEndpoint(endpointDescriptions.get(0));
            configBuilder.setCertificate(keyStoreLoader.getClientCertificate());
            configBuilder.setKeyPair(keyStoreLoader.getClientKeyPair());
            configBuilder.setApplicationUri("urn:localhost:appName");
            opcUaClient = OpcUaClient.create(configBuilder.build());
            opcUaClient.connect().get();
```
**Prosys server configuration**
[image]

[image]

**Exception in client:**
java.util.concurrent.ExecutionException: UaServiceFaultException: status=Bad_IdentityTokenRejected, message=The user identity token is valid but the server has rejected it.
	at java.base/java.util.concurrent.CompletableFuture.reportGet(CompletableFuture.java:395)
	at java.base/java.util.concurrent.CompletableFuture.get(CompletableFuture.java:2070)
	at com.cat.ai.runtime.common.connection.daq.OpcUAConnection.connect(OpcUAConnection.java:200)
.......................	org.springframework.beans.factory.annotation.InitDestroyAnnotationBeanPostProcessor$LifecycleElement.invoke(InitDestroyAnnotationBeanPostProcessor.java:363)
	at org.springframework.beans.factory.annotation.InitDestroyAnnotationBeanPostProcessor$LifecycleMetadata.invokeInitMethods(InitDestroyAnnotationBeanPostProcessor.java:307)
	at org.springframework.beans.factory.annotation.InitDestroyAnnotationBeanPostProcessor.postProcessBeforeInitialization(InitDestroyAnnotationBeanPostProcessor.java:136)
	at org.springframework.beans.factory.support.AbstractAutowireCapableBeanFactory.applyBeanPostProcessorsBeforeInitialization(AbstractAutowireCapableBeanFactory.java:414)
	at org.springframework.beans.factory.support.AbstractAutowireCapableBeanFactory.initializeBean(AbstractAutowireCapableBeanFactory.java:1770)
	at org.springframework.beans.factory.support.AbstractAutowireCapableBeanFactory.doCreateBean(AbstractAutowireCapableBeanFactory.java:593)
	at org.springframework.beans.factory.support.AbstractAutowireCapableBeanFactory.createBean(AbstractAutowireCapableBeanFactory.java:515)
	at org.springframework.beans.factory.support.AbstractBeanFactory.lambda$doGetBean$0(AbstractBeanFactory.java:320)
	at org.springframework.beans.factory.support.DefaultSingletonBeanRegistry.getSingleton(DefaultSingletonBeanRegistry.java:222)
	at org.springframework.beans.factory.support.AbstractBeanFactory.doGetBean(AbstractBeanFactory.java:318)
	at org.springframework.beans.factory.support.AbstractBeanFactory.getBean(AbstractBeanFactory.java:199)
	at org.springframework.beans.factory.support.DefaultListableBeanFactory.preInstantiateSingletons(DefaultListableBeanFactory.java:847)
	at org.springframework.context.support.AbstractApplicationContext.finishBeanFactoryInitialization(AbstractApplicationContext.java:877)
	at org.springframework.context.support.AbstractApplicationContext.refresh(AbstractApplicationContext.java:549)
	at org.springframework.boot.web.servlet.context.ServletWebServerApplicationContext.refresh(ServletWebServerApplicationContext.java:141)
	at org.springframework.boot.SpringApplication.refresh(SpringApplication.java:744)
	at org.springframework.boot.SpringApplication.refreshContext(SpringApplication.java:391)
	at org.springframework.boot.SpringApplication.run(SpringApplication.java:312)
	at org.springframework.boot.SpringApplication.run(SpringApplication.java:1215)
	at org.springframework.boot.SpringApplication.run(SpringApplication.java:1204)
	at com.cat.ai.services.edge.EdgeApplication.main(EdgeApplication.java:48)
Caused by: org.eclipse.milo.opcua.stack.core.UaServiceFaultException: status=Bad_IdentityTokenRejected, description=The user identity token is valid but the server has rejected it.
	at org.eclipse.milo.opcua.stack.client.UaStackClient.lambda$deliverResponse$5(UaStackClient.java:275)
	at org.eclipse.milo.opcua.stack.core.util.ExecutionQueue$Task.run(ExecutionQueue.java:119)
	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128)
	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628)
	at java.base/java.lang.Thread.run(Thread.java:835)

Thanks

kevinherron, 2020-07-02:
Well, the first thing that stands out is that you're re-using the
application instance certificate as an X509 identity certificate and that's
not okay to do, even if it's not the problem here.

Second, I can't tell from your screenshots whether you've actually added
your X509 identity to the Prosys server's list of trusted identities. I'm
not even sure where you do that in Prosys server.

Abhilash1368, 2020-07-02:
Thanks for suggestion,
I made sure that cert is trusted in Prosys server 
[image]
Req/Res from server side:
[image]

kevinherron, 2020-07-02:
That UI is for trusting application instance certificates, not identity certificates. 

I don't see a UI for managing these. On my system there is a separate directory at `~/.prosysopc/prosys-opc-ua-simulation-server/USERS_PKI/CA/certs` where the trusted identity certificates would need to be placed.

Abhilash1368, 2020-07-02:
Yep you are correct, I have added cert  to `~/.prosysopc/prosys-opc-ua-simulation-server/USERS_PKI/CA/certs ` and  It works! Thank a lot.

closed 2020-07-02 by kevinherron

guojunchao0201, 2020-09-07:
how resolve it , can you help me

KishanKishore, 2021-03-09:
> That UI is for trusting application instance certificates, not identity certificates.
>
> I don't see a UI for managing these. On my system there is a separate directory at [...]

Could you share a link which explains the difference betweens the types of certificates ?

KishanKishore, 2021-03-09:
> No, sorry, I'm not aware of any kind of documentation I can point you towards that explains the difference.
> 
> The OPC UA spec itself does on okay job of explaining the purpose of an Application Instance Certificate, which is used to provide application level security.
> 
> It also does not tell you that an Application Instance Certificate is to be used with X509 authentication, but rather a certificate that has been issued to the user authenticating. There are no requirements I'm aware of for the certificate issued to the user - that would be an implementation decision your applications made.

 But if we read here it is actually directing to X509 certificates: https://reference.opcfoundation.org/v104/Core/docs/Part2/8/

kevinherron, 2021-03-09:
Yes, what's your point? The Application Instance Certificate is an X509 certificate.

kevinherron, 2021-03-09:
I am a bit confused about what you are doing. Are you using Milo to write a client or a server?

KishanKishore, 2021-03-09:
> I am a bit confused about what you are doing. Are you using Milo to write a client or a server?

I am not using milo but another python based library to read from a OPCUA server but the error that I faced was exactly what was described here.

kevinherron, 2021-03-09:
You can't programmatically force a server to accept or trust your client's certificates, that would defeat the purpose.

You _can_ programmatically accept the server's in your client, but how you would do that depends on the library you are using.

KishanKishore, 2021-03-09:
> You can't programmatically force a server to accept or trust your client's certificates, that would defeat the purpose.
> 
> You _can_ programmatically accept the server's certificate in your client, but how you would do that depends on the library you are using.

- I understand that. 
- What I don't get is that when the client generates a new certificates and presents it to the server (ProSys Simulation Server) and from the server we accept that certificate even after that why would I need to copy it to the ```USERS_PKI``` folder manually. 
- I want the connection to work once the certificate has been accepted from the UI. 
- Perhaps I need to read more about this. Thanks for your replies though. Really helpful.