eclipse-milo/milo #660: Expection received UaException: status=Bad_ConnectionClosed, message=connection closed running over Debian 10

javierMolinero, 2020-07-16:
**Describe the bug**
I have an OPC client implemented using sdk-client library version 0.4.1. I use certificates to connect to the OPC Server. 
I compile and run my java applications over Windows 10 with jdk-11.0.7.10-hotspot, and over Debian 10 with java-1.11.0-openjdk-amd64. 
The problem is when I execute the client over Linux (in this case a Debian distro), I receive sometimes the exception written below ( UaException: status=Bad_ConnectionClosed, message=connection closed). This exception is received when I call to the client method connect().get(). This behaviour is random (sometimes I do connect properly), but other times fails, when starts to fail, it fails consistently until reboot my Debian device. 
Over Windows 10 always works.

**Expected behavior**
The expected behavior would be working in both OS, Windows 10 and Debian.

**Logs and Packet Captures**
java.util.concurrent.ExecutionException: UaException: status=Bad_ConnectionClosed, message=connection closed
	at java.util.concurrent.CompletableFuture.reportGet(CompletableFuture.java:395) ~[?:?]
	at java.util.concurrent.CompletableFuture.get(CompletableFuture.java:1999) ~[?:?]
	at com.siemens.coreshield.upload_client.opc.OpcUtilities.clientConnect(OpcUtilities.java:201) [upload-client-CLI-0.0.2-SNAPSHOT-jar-with-dependencies.jar:?]
	at com.siemens.coreshield.upload_client.main.ArgumentManager.runUploadClient(ArgumentManager.java:48) [upload-client-CLI-0.0.2-SNAPSHOT-jar-with-dependencies.jar:?]
	at com.siemens.coreshield.upload_client.main.MainClassCLI.run(MainClassCLI.java:87) [upload-client-CLI-0.0.2-SNAPSHOT-jar-with-dependencies.jar:?]
	at picocli.CommandLine.executeUserObject(CommandLine.java:1769) [upload-client-CLI-0.0.2-SNAPSHOT-jar-with-dependencies.jar:?]
	at picocli.CommandLine.access$900(CommandLine.java:145) [upload-client-CLI-0.0.2-SNAPSHOT-jar-with-dependencies.jar:?]
	at picocli.CommandLine$RunLast.executeUserObjectOfLastSubcommandWithSameParent(CommandLine.java:2150) [upload-client-CLI-0.0.2-SNAPSHOT-jar-with-dependencies.jar:?]
	at picocli.CommandLine$RunLast.handle(CommandLine.java:2144) [upload-client-CLI-0.0.2-SNAPSHOT-jar-with-dependencies.jar:?]
	at picocli.CommandLine$RunLast.handle(CommandLine.java:2108) [upload-client-CLI-0.0.2-SNAPSHOT-jar-with-dependencies.jar:?]
	at picocli.CommandLine$AbstractParseResultHandler.execute(CommandLine.java:1975) [upload-client-CLI-0.0.2-SNAPSHOT-jar-with-dependencies.jar:?]
	at picocli.CommandLine.execute(CommandLine.java:1904) [upload-client-CLI-0.0.2-SNAPSHOT-jar-with-dependencies.jar:?]
	at com.siemens.coreshield.upload_client.main.MainClassCLI.main(MainClassCLI.java:79) [upload-client-CLI-0.0.2-SNAPSHOT-jar-with-dependencies.jar:?]
Caused by: org.eclipse.milo.opcua.stack.core.UaException: connection closed
	at org.eclipse.milo.opcua.stack.client.transport.uasc.UascClientMessageHandler.channelInactive(UascClientMessageHandler.java:153) ~[upload-client-CLI-0.0.2-SNAPSHOT-jar-with-dependencies.jar:?]
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelInactive(AbstractChannelHandlerContext.java:260) ~[upload-client-CLI-0.0.2-SNAPSHOT-jar-with-dependencies.jar:?]
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelInactive(AbstractChannelHandlerContext.java:246) ~[upload-client-CLI-0.0.2-SNAPSHOT-jar-with-dependencies.jar:?]
	at io.netty.channel.AbstractChannelHandlerContext.fireChannelInactive(AbstractChannelHandlerContext.java:239) ~[upload-client-CLI-0.0.2-SNAPSHOT-jar-with-dependencies.jar:?]
	at io.netty.handler.codec.ByteToMessageDecoder.channelInputClosed(ByteToMessageDecoder.java:386) ~[upload-client-CLI-0.0.2-SNAPSHOT-jar-with-dependencies.jar:?]
	at io.netty.handler.codec.ByteToMessageDecoder.channelInactive(ByteToMessageDecoder.java:351) ~[upload-client-CLI-0.0.2-SNAPSHOT-jar-with-dependencies.jar:?]
	at io.netty.handler.codec.ByteToMessageCodec.channelInactive(ByteToMessageCodec.java:118) ~[upload-client-CLI-0.0.2-SNAPSHOT-jar-with-dependencies.jar:?]
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelInactive(AbstractChannelHandlerContext.java:260) ~[upload-client-CLI-0.0.2-SNAPSHOT-jar-with-dependencies.jar:?]
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelInactive(AbstractChannelHandlerContext.java:246) ~[upload-client-CLI-0.0.2-SNAPSHOT-jar-with-dependencies.jar:?]
	at io.netty.channel.AbstractChannelHandlerContext.fireChannelInactive(AbstractChannelHandlerContext.java:239) ~[upload-client-CLI-0.0.2-SNAPSHOT-jar-with-dependencies.jar:?]
	at io.netty.channel.DefaultChannelPipeline$HeadContext.channelInactive(DefaultChannelPipeline.java:1405) ~[upload-client-CLI-0.0.2-SNAPSHOT-jar-with-dependencies.jar:?]
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelInactive(AbstractChannelHandlerContext.java:260) ~[upload-client-CLI-0.0.2-SNAPSHOT-jar-with-dependencies.jar:?]
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelInactive(AbstractChannelHandlerContext.java:246) ~[upload-client-CLI-0.0.2-SNAPSHOT-jar-with-dependencies.jar:?]
	at io.netty.channel.DefaultChannelPipeline.fireChannelInactive(DefaultChannelPipeline.java:901) ~[upload-client-CLI-0.0.2-SNAPSHOT-jar-with-dependencies.jar:?]
	at io.netty.channel.AbstractChannel$AbstractUnsafe$8.run(AbstractChannel.java:818) ~[upload-client-CLI-0.0.2-SNAPSHOT-jar-with-dependencies.jar:?]
	at io.netty.util.concurrent.AbstractEventExecutor.safeExecute(AbstractEventExecutor.java:164) ~[upload-client-CLI-0.0.2-SNAPSHOT-jar-with-dependencies.jar:?]
	at io.netty.util.concurrent.SingleThreadEventExecutor.runAllTasks(SingleThreadEventExecutor.java:472) ~[upload-client-CLI-0.0.2-SNAPSHOT-jar-with-dependencies.jar:?]
	at io.netty.channel.nio.NioEventLoop.run(NioEventLoop.java:497) ~[upload-client-CLI-0.0.2-SNAPSHOT-jar-with-dependencies.jar:?]
	at io.netty.util.concurrent.SingleThreadEventExecutor$4.run(SingleThreadEventExecutor.java:989) ~[upload-client-CLI-0.0.2-SNAPSHOT-jar-with-dependencies.jar:?]
	at io.netty.util.internal.ThreadExecutorMap$2.run(ThreadExecutorMap.java:74) ~[upload-client-CLI-0.0.2-SNAPSHOT-jar-with-dependencies.jar:?]
	at java.lang.Thread.run(Thread.java:834) ~[?:?]
com.siemens.coreshield.upload_client.exceptions.ClientConnectException: Failed connecting to the OPC UA server.
	at com.siemens.coreshield.upload_client.opc.OpcUtilities.clientConnect(OpcUtilities.java:204)
	at com.siemens.coreshield.upload_client.main.ArgumentManager.runUploadClient(ArgumentManager.java:48)
	at com.siemens.coreshield.upload_client.main.MainClassCLI.run(MainClassCLI.java:87)
	at picocli.CommandLine.executeUserObject(CommandLine.java:1769)
	at picocli.CommandLine.access$900(CommandLine.java:145)
	at picocli.CommandLine$RunLast.executeUserObjectOfLastSubcommandWithSameParent(CommandLine.java:2150)
	at picocli.CommandLine$RunLast.handle(CommandLine.java:2144)
	at picocli.CommandLine$RunLast.handle(CommandLine.java:2108)
	at picocli.CommandLine$AbstractParseResultHandler.execute(CommandLine.java:1975)
	at picocli.CommandLine.execute(CommandLine.java:1904)
	at com.siemens.coreshield.upload_client.main.MainClassCLI.main(MainClassCLI.java:79)
Caused by: java.util.concurrent.ExecutionException: UaException: status=Bad_ConnectionClosed, message=connection closed
	at java.base/java.util.concurrent.CompletableFuture.reportGet(CompletableFuture.java:395)
	at java.base/java.util.concurrent.CompletableFuture.get(CompletableFuture.java:1999)
	at com.siemens.coreshield.upload_client.opc.OpcUtilities.clientConnect(OpcUtilities.java:201)
	... 10 more
Caused by: UaException: status=Bad_ConnectionClosed, message=connection closed
	at org.eclipse.milo.opcua.stack.client.transport.uasc.UascClientMessageHandler.channelInactive(UascClientMessageHandler.java:153)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelInactive(AbstractChannelHandlerContext.java:260)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelInactive(AbstractChannelHandlerContext.java:246)
	at io.netty.channel.AbstractChannelHandlerContext.fireChannelInactive(AbstractChannelHandlerContext.java:239)
	at io.netty.handler.codec.ByteToMessageDecoder.channelInputClosed(ByteToMessageDecoder.java:386)
	at io.netty.handler.codec.ByteToMessageDecoder.channelInactive(ByteToMessageDecoder.java:351)
	at io.netty.handler.codec.ByteToMessageCodec.channelInactive(ByteToMessageCodec.java:118)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelInactive(AbstractChannelHandlerContext.java:260)
[... cut ...]
	at io.netty.channel.nio.NioEventLoop.run(NioEventLoop.java:497)
	at io.netty.util.concurrent.SingleThreadEventExecutor$4.run(SingleThreadEventExecutor.java:989)
	at io.netty.util.internal.ThreadExecutorMap$2.run(ThreadExecutorMap.java:74)
	at java.base/java.lang.Thread.run(Thread.java:834)

**Additional context**
Version of Eclipse Milo you are using, OPC UA client or software server you are interacting with, etc...
Eclipse Milo 0.4.1. A propietary OPC Server. (Not receiving logs from it when Client connection fails)

kevinherron, 2020-07-16:
Hmm, not sure why this would be happening on Debian but now Windows.

The server is closing the connection for some reason. How often are you calling `connect()`? Do you always make sure to call `disconnect()` after you’re done, even if the initial call to `connect()` fails? Maybe the server has some kind of rate limiting or limit to the number of connections.

It might help if you can get a Wireshark capture while this is happening and any logs there might be from the server.

javierMolinero, 2020-07-16:
Hi! first of all thanks so much for your quick response!

My client application is used to upload files with OPC, it works in CLI mode uploading files one by one. The procedure is very simple, first connect to OPC Server with a valid certificate, write the file in a OPC node and finally calls disconnect(). Following your comments, I don't disconnect after a failed connect() call, I will fix it.... I also going to check the server configuration to see if there is any limit in the number of connections. Finally I will send you a Wireshark capture with the logs when the connection fails, and some logs of server if exists (I remember check it and not receiving any log from OPC Server).

Thanks in advance.
Regards!!

javierMolinero, 2020-07-17:
Hi again! I have collected OPC Server logs from a correct procedure (connect, write file to node & disconnect) and a failed one (are consecutive, first ok and then fail). It seems that there is an "additional" check from the OPC Server -> "Checking Channels for lifetime expiration!" in the failed one. Since this part, the secure channel is deleted and the underlayer socket. Why? I don't know, I am studying the server configuration file to avoid limitation in the channel lifetime or number of connections. When I get it, I will also provide you the Wireshark logs. 

[OPCServerTransmissionKO.log](https://github.com/eclipse/milo/files/4936975/OPCServerTransmissionKO.log)
[OPCServerTransmissionOK.log](https://github.com/eclipse/milo/files/4936976/OPCServerTransmissionOK.log)

Thanks!

kevinherron, 2020-07-17:
The logs may be useful later but the Wireshark will also help. Do you have a support contract with whoever the vendor of the server is in case you need to get them involved?

kevinherron, 2020-08-07:
If this is still an issue feel free to comment or email me once you have the capture.

closed 2020-08-07 by kevinherron