eclipse-milo/milo #663: Subscriptions  - on data change Action Example

hbbend, 2020-07-21:
Hello,
The Example about subscription, only create the subscription.
How to create an on data change notification in order to do actions (For example: display or use the datavalue.)

Thank you,
Best Regards
Hamid

nacyolsa, 2020-07-21:
SubscriptionExample.java has `onSubscriptionValue()` method. Is it not what you are looking for?

kevinherron, 2020-07-21:
Yes, you hook up a callback using `UaMonitoredItem#setValueConsumer` as the example shows. 

You can also listen at the subscription level by adding a `NotificationListener` using `UaSubscription#addNotificationListener`.

hbbend, 2020-07-21:
this is used to create the UaMonitoredItem List, and it works fine.
I have an output indicating that the subscription was created:
17:22:34.103 [main] INFO  o.e.milo.examples.client.Test_Client - item created for nodeId=NodeId{ns=5, id=Counter1}
Now , I need to display the values of this counter as it changes.

closed 2020-07-21 by kevinherron

hbbend, 2020-07-21:
UaMonitoredItem#setValueConsumer is done.

I would like to add a notification listener, and an on data change notificaion.
I would like to display a counter as it change values.

is there a detailed example to do this. (the necessary code)
thank you