eclipse-milo/milo #665: Subscriptions - display continusely the values of the monitored item

hbbend, 2020-07-21:
Hello,
I am reading a counter values using opc ua milo client sdk.
I created a subscription to a monitored data item successfully.
I would like to display the values of the counter as it changes.

the ondatachangenotification wont work.
please find the code below:

//***************************
        NotificationListener listener = null;
        List<DataValue> dataValues = null;
        NotificationMessage notificationMessage = null;

listener.onDataChangeNotification(subscription,items,dataValues,notificationMessage.getPublishTime());
//*****************************		
           

Thank you,
Best Regards

elucash, 2020-07-23:
Are you just calling notification yourself instead of passing a listener to a subscription? It should work the other way around, you implement `NotificationListener` with `onDataChangeNotification` and register it (somewhere, I'm new to _milo_, will find out the details soon) so that `onDataChangeNotification` will be called when there is a change.

hbbend, 2020-07-26:
just a standard example for that...

closed 2020-08-07 by kevinherron