eclipse-milo/milo #670: Getting Error - After receiving  subscription message contentiously for 2 minutes,

jdhanasekar, 2020-07-31:
Hi Milo,

I have subscribed a node in ProsysSimulationServer. I am getting subscription message when ever changes occurs.. After 50 subscription message, I am getting error like this,

```
14:21:21.764 [NonceUtilSecureRandom] INFO  o.e.m.o.stack.core.util.NonceUtil - SecureRandom seeded in 0ms.
14:21:21.875 [main] INFO  o.e.m.e.c.ProsysSimulationServer2 - item created for nodeId=NodeId{ns=3, id=Random}
14:21:22.861 [milo-shared-thread-pool-0] INFO  o.e.m.e.c.ProsysSimulationServer2 - subscription value received: item=NodeId{ns=3, id=Random}, value=Variant{value=-0.9500780051285673}
14:21:22.862 [milo-shared-thread-pool-0] INFO  o.e.m.e.c.ProsysSimulationServer2 - subscription value received: item=NodeId{ns=3, id=Random}, value=Variant{value=0.17076364507272324}
14:21:23.853 [milo-shared-thread-pool-0] INFO  o.e.m.e.c.ProsysSimulationServer2 - subscription value received: item=NodeId{ns=3, id=Random}, value=Variant{value=-0.2792929731131992}
14:21:24.853 [milo-shared-thread-pool-0] INFO  o.e.m.e.c.ProsysSimulationServer2 - subscription value received: item=NodeId{ns=3, id=Random}, value=Variant{value=0.8186158626129081}
14:21:25.852 [milo-shared-thread-pool-0] INFO  o.e.m.e.c.ProsysSimulationServer2 - subscription value received: item=NodeId{ns=3, id=Random}, value=Variant{value=0.4602905023277182}
14:21:26.892 [milo-shared-thread-pool-2] INFO  o.e.m.e.c.ProsysSimulationServer2 - subscription value received: item=NodeId{ns=3, id=Random}, value=Variant{value=-0.28967014125725576}
14:21:27.853 [milo-shared-thread-pool-1] INFO  o.e.m.e.c.ProsysSimulationServer2 - subscription value received: item=NodeId{ns=3, id=Random}, value=Variant{value=-0.6479954419378005}
14:21:28.852 [milo-shared-thread-pool-1] INFO  o.e.m.e.c.ProsysSimulationServer2 - subscription value received: item=NodeId{ns=3, id=Random}, value=Variant{value=0.4499133341836621}
14:21:29.853 [milo-shared-thread-pool-1] INFO  o.e.m.e.c.ProsysSimulationServer2 - subscription value received: item=NodeId{ns=3, id=Random}, value=Variant{value=-1.4328400226037985E-4}
14:21:30.853 [milo-shared-thread-pool-1] INFO  o.e.m.e.c.ProsysSimulationServer2 - subscription value received: item=NodeId{ns=3, id=Random}, value=Variant{value=1.1206984258036745}
14:21:31.888 [milo-shared-thread-pool-1] INFO  o.e.m.e.c.ProsysSimulationServer2 - subscription value received: item=NodeId{ns=3, id=Random}, value=Variant{value=0.762373065518485}
14:21:32.853 [milo-shared-thread-pool-0] INFO  o.e.m.e.c.ProsysSimulationServer2 - subscription value received: item=NodeId{ns=3, id=Random}, value=Variant{value=1.8602819012445924}
14:21:33.852 [milo-shared-thread-pool-2] INFO  o.e.m.e.c.ProsysSimulationServer2 - subscription value received: item=NodeId{ns=3, id=Random}, value=Variant{value=1.501956540959403}
14:21:34.853 [milo-shared-thread-pool-1] INFO  o.e.m.e.c.ProsysSimulationServer2 - subscription value received: item=NodeId{ns=3, id=Random}, value=Variant{value=1.027189790285918}
14:21:35.853 [milo-shared-thread-pool-0] INFO  o.e.m.e.c.ProsysSimulationServer2 - subscription value received: item=NodeId{ns=3, id=Random}, value=Variant{value=0.6688643703960837}
14:21:36.889 [milo-shared-thread-pool-2] INFO  o.e.m.e.c.ProsysSimulationServer2 - subscription value received: item=NodeId{ns=3, id=Random}, value=Variant{value=1.7667732657268358}
14:21:37.852 [milo-shared-thread-pool-2] INFO  o.e.m.e.c.ProsysSimulationServer2 - subscription value received: item=NodeId{ns=3, id=Random}, value=Variant{value=1.4084479054416459}
14:21:38.853 [milo-shared-thread-pool-2] INFO  o.e.m.e.c.ProsysSimulationServer2 - subscription value received: item=NodeId{ns=3, id=Random}, value=Variant{value=-1.4707104443570636}
14:21:39.852 [milo-shared-thread-pool-0] INFO  o.e.m.e.c.ProsysSimulationServer2 - subscription value received: item=NodeId{ns=3, id=Random}, value=Variant{value=-1.920767062542986}
14:21:40.852 [milo-shared-thread-pool-1] INFO  o.e.m.e.c.ProsysSimulationServer2 - subscription value received: item=NodeId{ns=3, id=Random}, value=Variant{value=-0.8228582268168787}
14:21:41.889 [milo-shared-thread-pool-2] INFO  o.e.m.e.c.ProsysSimulationServer2 - subscription value received: item=NodeId{ns=3, id=Random}, value=Variant{value=-1.1811835274974238}
14:21:42.852 [milo-shared-thread-pool-2] INFO  o.e.m.e.c.ProsysSimulationServer2 - subscription value received: item=NodeId{ns=3, id=Random}, value=Variant{value=-1.9311442306870426}
14:21:43.852 [milo-shared-thread-pool-2] INFO  o.e.m.e.c.ProsysSimulationServer2 - subscription value received: item=NodeId{ns=3, id=Random}, value=Variant{value=1.710530409027768}
14:21:44.852 [milo-shared-thread-pool-2] INFO  o.e.m.e.c.ProsysSimulationServer2 - subscription value received: item=NodeId{ns=3, id=Random}, value=Variant{value=-1.19156069564148}
14:21:45.852 [milo-shared-thread-pool-2] INFO  o.e.m.e.c.ProsysSimulationServer2 - subscription value received: item=NodeId{ns=3, id=Random}, value=Variant{value=-1.6416173138274024}
14:21:46.890 [milo-shared-thread-pool-2] INFO  o.e.m.e.c.ProsysSimulationServer2 - subscription value received: item=NodeId{ns=3, id=Random}, value=Variant{value=-0.520775723230757}
14:21:47.853 [milo-shared-thread-pool-2] INFO  o.e.m.e.c.ProsysSimulationServer2 - subscription value received: item=NodeId{ns=3, id=Random}, value=Variant{value=-0.8791010239113017}
14:21:48.852 [milo-shared-thread-pool-0] INFO  o.e.m.e.c.ProsysSimulationServer2 - subscription value received: item=NodeId{ns=3, id=Random}, value=Variant{value=0.21880781181480558}
14:21:49.853 [milo-shared-thread-pool-0] INFO  o.e.m.e.c.ProsysSimulationServer2 - subscription value received: item=NodeId{ns=3, id=Random}, value=Variant{value=-0.13951748886573911}
14:21:50.852 [milo-shared-thread-pool-2] INFO  o.e.m.e.c.ProsysSimulationServer2 - subscription value received: item=NodeId{ns=3, id=Random}, value=Variant{value=1.9852090195433139}
14:21:51.891 [milo-shared-thread-pool-1] INFO  o.e.m.e.c.ProsysSimulationServer2 - subscription value received: item=NodeId{ns=3, id=Random}, value=Variant{value=1.6268836592581244}
14:21:52.852 [milo-shared-thread-pool-2] INFO  o.e.m.e.c.ProsysSimulationServer2 - subscription value received: item=NodeId{ns=3, id=Random}, value=Variant{value=-1.2752075050157683}
14:21:53.852 [milo-shared-thread-pool-0] INFO  o.e.m.e.c.ProsysSimulationServer2 - subscription value received: item=NodeId{ns=3, id=Random}, value=Variant{value=-1.6335328653009578}
14:21:54.852 [milo-shared-thread-pool-1] INFO  o.e.m.e.c.ProsysSimulationServer2 - subscription value received: item=NodeId{ns=3, id=Random}, value=Variant{value=-0.5126912150996672}
14:21:55.852 [milo-shared-thread-pool-2] INFO  o.e.m.e.c.ProsysSimulationServer2 - subscription value received: item=NodeId{ns=3, id=Random}, value=Variant{value=-0.5958226824733681}
14:21:56.893 [milo-shared-thread-pool-0] INFO  o.e.m.e.c.ProsysSimulationServer2 - subscription value received: item=NodeId{ns=3, id=Random}, value=Variant{value=-0.9656143903915044}
14:21:57.852 [milo-shared-thread-pool-0] INFO  o.e.m.e.c.ProsysSimulationServer2 - subscription value received: item=NodeId{ns=3, id=Random}, value=Variant{value=0.1437608525721945}
14:21:58.852 [milo-shared-thread-pool-2] INFO  o.e.m.e.c.ProsysSimulationServer2 - subscription value received: item=NodeId{ns=3, id=Random}, value=Variant{value=-0.20309810047540333}
14:21:59.852 [milo-shared-thread-pool-1] INFO  o.e.m.e.c.ProsysSimulationServer2 - subscription value received: item=NodeId{ns=3, id=Random}, value=Variant{value=-0.9645252109026137}
14:22:00.853 [milo-shared-thread-pool-0] INFO  o.e.m.e.c.ProsysSimulationServer2 - subscription value received: item=NodeId{ns=3, id=Random}, value=Variant{value=-1.3343169188207504}
14:22:01.852 [milo-shared-thread-pool-0] INFO  o.e.m.e.c.ProsysSimulationServer2 - subscription value received: item=NodeId{ns=3, id=Random}, value=Variant{value=-0.3166729337577845}
14:22:02.852 [milo-shared-thread-pool-1] INFO  o.e.m.e.c.ProsysSimulationServer2 - subscription value received: item=NodeId{ns=3, id=Random}, value=Variant{value=-0.6635318868053823}
14:22:03.852 [milo-shared-thread-pool-1] INFO  o.e.m.e.c.ProsysSimulationServer2 - subscription value received: item=NodeId{ns=3, id=Random}, value=Variant{value=0.4458433561583166}
14:22:04.851 [milo-shared-thread-pool-1] INFO  o.e.m.e.c.ProsysSimulationServer2 - subscription value received: item=NodeId{ns=3, id=Random}, value=Variant{value=0.07605164824017985}
[... cut ...]
14:22:26.887 [ForkJoinPool.commonPool-worker-3] ERROR o.e.m.e.client.ClientExampleRunner - Error running example: null
java.util.concurrent.TimeoutException: null
	at java.base/java.util.concurrent.CompletableFuture.timedGet(CompletableFuture.java:1886)
	at java.base/java.util.concurrent.CompletableFuture.get(CompletableFuture.java:2021)
	at org.eclipse.milo.examples.client.ClientExampleRunner.run(ClientExampleRunner.java:147)
	at org.eclipse.milo.examples.client.ProsysSimulationServer2.main(ProsysSimulationServer2.java:36)

Process finished with exit code 0
```

kevinherron, 2020-07-31:
Take a look at whatever version of ClientExampleRunner you’ve copied and figure out where you’re waiting until a 2 minute timeout expires.

jdhanasekar, 2020-07-31:
I think, this line in ```  .setRequestTimeout(uint(5000))``` in ClientExampleRunner.java 

```
 OpcUaClientConfig config = OpcUaClientConfig.builder()
            .setApplicationName(LocalizedText.english("eclipse milo opc-ua client"))
            .setApplicationUri("urn:eclipse:milo:examples:client")
            .setCertificate(loader.getClientCertificate())
            .setKeyPair(loader.getClientKeyPair())
            .setEndpoint(endpoint)
            .setIdentityProvider(clientExample.getIdentityProvider())
            .setRequestTimeout(uint(5000))
            .build();
```

Am I correct ??

jdhanasekar, 2020-07-31:
After setting ReqestTimeOut is to 10000, like this, 
```
.setRequestTimeout(uint(10000))
```

I am not getting issue. But for 5000, I am getting error.

xiannvdezhu, 2020-08-06:
@jdhanasekar 
`NodeId nodeId = new NodeId(b.getNodeIndex(), b.getIdentifier());`
`ReadValueId readValueId = new ReadValueId(nodeId, AttributeId.Value.uid(), null, null);`
` **UInteger clientHandle = subscription.nextClientHandle();**`
` MonitoringParameters parameters = new MonitoringParameters(`
                       ` clientHandle,`
                     `   1000.0,`
                    `    null,`
                       ` uint(10),`
                      `  true`
 `);`

closed 2020-08-07 by kevinherron