eclipse-milo/milo #671: Subscribed values are not reflecting in UA Browser

jdhanasekar, 2020-08-06:
Hi Milo,

I have successfully subscribed to a multiple attribute from a server in a single call of ```createMonitoredItems``` and am able to see the values in IntellIJ IDE  when changes occur. 

But, I am not seeing the changed values (are not reflected) in UA Browser like Prosys Client , UaExpert. 

Please help me to solve this problem. 

**Program:**
```
package org.eclipse.milo.examples.client;

import org.eclipse.milo.opcua.sdk.client.OpcUaClient;
import org.eclipse.milo.opcua.sdk.client.api.subscriptions.UaMonitoredItem;
import org.eclipse.milo.opcua.sdk.client.api.subscriptions.UaSubscription;
import org.eclipse.milo.opcua.stack.core.AttributeId;
import org.eclipse.milo.opcua.stack.core.types.builtin.DataValue;
import org.eclipse.milo.opcua.stack.core.types.builtin.NodeId;
import org.eclipse.milo.opcua.stack.core.types.builtin.QualifiedName;
import org.eclipse.milo.opcua.stack.core.types.builtin.unsigned.UInteger;
import org.eclipse.milo.opcua.stack.core.types.builtin.unsigned.Unsigned;
import org.eclipse.milo.opcua.stack.core.types.enumerated.MonitoringMode;
import org.eclipse.milo.opcua.stack.core.types.enumerated.TimestampsToReturn;
import org.eclipse.milo.opcua.stack.core.types.structured.MonitoredItemCreateRequest;
import org.eclipse.milo.opcua.stack.core.types.structured.MonitoringParameters;
import org.eclipse.milo.opcua.stack.core.types.structured.ReadValueId;
import org.slf4j.Logger;
import org.slf4j.LoggerFactory;

import java.util.List;
import java.util.concurrent.CompletableFuture;
import java.util.concurrent.atomic.AtomicLong;
import java.util.function.BiConsumer;

import static com.google.common.collect.Lists.newArrayList;
import static org.eclipse.milo.opcua.stack.core.types.builtin.unsigned.Unsigned.uint;

public class ProsysSimulationServer2 implements ClientExample {
    private final Logger logger = LoggerFactory.getLogger(getClass());
    private final AtomicLong clientHandles = new AtomicLong(1L); // Used for subscriptions
    //private ClientExample clientExample;

    public static void main(String[] args) throws Exception {
        ProsysSimulationServer2 ProsysClient = new ProsysSimulationServer2();

        new ClientExampleRunner(ProsysClient, false).run();
    }

    @Override
    public void run(OpcUaClient client, CompletableFuture<OpcUaClient> future) throws Exception {
        NodeId Random = initString("Random");
        NodeId Counter = initString("Counter");
        NodeId Expression = initString("Expression");
        NodeId Sawtooth = initString("Sawtooth");

        

        // synchronous connect
        client.connect().get();

        // create a subscription @ 1000ms
        UaSubscription subscription = client.getSubscriptionManager().createSubscription(1000.0).get();

        // subscribe to the Value attribute of the server's CurrentTime node
        ReadValueId readValueId = new ReadValueId(
                Random,
                AttributeId.Value.uid(), null, QualifiedName.NULL_VALUE);

        // important: client handle must be unique per item
        UInteger clientHandle = uint(clientHandles.getAndIncrement());
[... 120 lines cut ...]
16:00:08.466 [main] INFO  o.e.m.e.c.ProsysSimulationServer2 - dhanasekar @Subscription.java NodeId-of-NegateValue =NodeId{ns=3, id=Counter}
16:00:08.467 [main] INFO  o.e.m.e.c.ProsysSimulationServer2 - dhanasekar @Subscription.java NodeId-of-NegateValue =NodeId{ns=3, id=Expression}
16:00:08.467 [main] INFO  o.e.m.e.c.ProsysSimulationServer2 - dhanasekar @Subscription.java NodeId-of-NegateValue =NodeId{ns=3, id=Sawtooth}
16:00:08.494 [NonceUtilSecureRandom] INFO  o.e.m.o.stack.core.util.NonceUtil - SecureRandom seeded in 0ms.
16:00:08.604 [main] INFO  o.e.m.e.c.ProsysSimulationServer2 - item created for nodeId=NodeId{ns=3, id=Random}
16:00:08.604 [main] INFO  o.e.m.e.c.ProsysSimulationServer2 - item created for nodeId=NodeId{ns=3, id=Counter}
16:00:08.604 [main] INFO  o.e.m.e.c.ProsysSimulationServer2 - item created for nodeId=NodeId{ns=3, id=Expression}
16:00:08.604 [main] INFO  o.e.m.e.c.ProsysSimulationServer2 - item created for nodeId=NodeId{ns=3, id=Sawtooth}
16:00:09.589 [milo-shared-thread-pool-2] INFO  o.e.m.e.c.ProsysSimulationServer2 - subscription value received: item=NodeId{ns=3, id=Random}, value=Variant{value=1.2722789303387945}
16:00:09.589 [milo-shared-thread-pool-2] INFO  o.e.m.e.c.ProsysSimulationServer2 - subscription value received: item=NodeId{ns=3, id=Counter}, value=Variant{value=56}
16:00:09.590 [milo-shared-thread-pool-2] INFO  o.e.m.e.c.ProsysSimulationServer2 - subscription value received: item=NodeId{ns=3, id=Expression}, value=Variant{value=3.1890438489363624}
16:00:09.590 [milo-shared-thread-pool-2] INFO  o.e.m.e.c.ProsysSimulationServer2 - subscription value received: item=NodeId{ns=3, id=Sawtooth}, value=Variant{value=-0.7999999523162842}
16:00:09.590 [milo-shared-thread-pool-2] INFO  o.e.m.e.c.ProsysSimulationServer2 - subscription value received: item=NodeId{ns=3, id=Random}, value=Variant{value=0.8222223121528716}
16:00:09.590 [milo-shared-thread-pool-2] INFO  o.e.m.e.c.ProsysSimulationServer2 - subscription value received: item=NodeId{ns=3, id=Counter}, value=Variant{value=57}
16:00:09.590 [milo-shared-thread-pool-2] INFO  o.e.m.e.c.ProsysSimulationServer2 - subscription value received: item=NodeId{ns=3, id=Expression}, value=Variant{value=3.5021130839188492}
16:00:09.590 [milo-shared-thread-pool-2] INFO  o.e.m.e.c.ProsysSimulationServer2 - subscription value received: item=NodeId{ns=3, id=Sawtooth}, value=Variant{value=-0.3999999761581421}
16:00:10.582 [milo-shared-thread-pool-2] INFO  o.e.m.e.c.ProsysSimulationServer2 - subscription value received: item=NodeId{ns=3, id=Random}, value=Variant{value=1.9430639623541621}
16:00:10.582 [milo-shared-thread-pool-2] INFO  o.e.m.e.c.ProsysSimulationServer2 - subscription value received: item=NodeId{ns=3, id=Counter}, value=Variant{value=58}
16:00:10.582 [milo-shared-thread-pool-2] INFO  o.e.m.e.c.ProsysSimulationServer2 - subscription value received: item=NodeId{ns=3, id=Expression}, value=Variant{value=3.7320508462150834}
16:00:10.582 [milo-shared-thread-pool-2] INFO  o.e.m.e.c.ProsysSimulationServer2 - subscription value received: item=NodeId{ns=3, id=Sawtooth}, value=Variant{value=0.0}
16:00:11.581 [milo-shared-thread-pool-2] INFO  o.e.m.e.c.ProsysSimulationServer2 - subscription value received: item=NodeId{ns=3, id=Random}, value=Variant{value=1.5847386020689727}
16:00:11.582 [milo-shared-thread-pool-2] INFO  o.e.m.e.c.ProsysSimulationServer2 - subscription value received: item=NodeId{ns=3, id=Counter}, value=Variant{value=59}
16:00:11.582 [milo-shared-thread-pool-2] INFO  o.e.m.e.c.ProsysSimulationServer2 - subscription value received: item=NodeId{ns=3, id=Expression}, value=Variant{value=3.886289671032016}
16:00:11.582 [milo-shared-thread-pool-2] INFO  o.e.m.e.c.ProsysSimulationServer2 - subscription value received: item=NodeId{ns=3, id=Sawtooth}, value=Variant{value=0.3999999761581421}
```

kevinherron, 2020-08-07:
This is hardly an appropriate question to ask here. Try refreshing the browser view or contact support for one of those products. I can't see how this has anything to do with Milo.

closed 2020-08-07 by kevinherron

jdhanasekar, 2020-08-08:
When I do a subscribe request for a single attribute, Values are reflected in dummy nodes properly. I am not seeing issue, 
```
 List<UaMonitoredItem> items = subscription.createMonitoredItems(
                TimestampsToReturn.Both,
                newArrayList(request),
                onItemCreated
        ).get();
```
I am getting output in UA Broswer..
[image]

When I do a subscribe request for multiple attributes, Values are not reflected in dummy nodes properly. I am seeing issue, 
```
 List<UaMonitoredItem> items = subscription.createMonitoredItems(
                TimestampsToReturn.Both,
                newArrayList(request,SquareRequest),
                onItemCreated
        ).get();
```

[image]

jdhanasekar, 2020-08-10:
Instead multiple attribute subscription request , can I do single request in ```createMontioredItems``` ??..
Because When I do multiple request in createMontioredItems, values are not reflected in dummy nodes.

Does it issue with UA Browser or my code ?

kevinherron, 2020-08-10:
Seems like a problem with the simulation server.

jdhanasekar, 2020-08-10:
Can you please suggest some other simulation server which will work well with multiple subscriptions ?