eclipse-milo/milo #676: ModelChangeEvent does not trigger a BrowseRequest for the affected NodeId

achintya-kumar, 2020-08-10:
Hi @kevinherron :)

I'm posting it here so that others may find it useful in the future.
After reading the specification, I found that the GeneralModelChangeEvent must be emitted on the Server node as I've done here in the ExampleNamespace. I'm adding a variable-nodes every 2 seconds, emitting a GeneralModelChangeEvent on the Server node and updating the NodeVersion property-node of the object where I'm adding the variables-nodes.
[image]
[image]

With the code above, I can see that UaExpert is logging these events as follows:
[image]

However, in a **different server** I'm working with, upon receipt of such events by UaExpert, an automatic rebrowse operation occurs as follows:
[image]

The question is: How can I implement similar behavior using Milo so that an automatic rebrowse operation occurs on the affected node?

I'm attaching the wireshark-capture from the **different server** where it's working fine. 
[wireshark-dump.zip](https://github.com/eclipse/milo/files/5053778/wireshark-dump.zip)

FYI @Ralfidze

kevinherron, 2020-08-10:
Can you include a capture of the Milo traffic as well?

kevinherron, 2020-08-10:
Also - I don't see any Browse requests in the capture from the other server.

achintya-kumar, 2020-08-10:
@kevinherron 
I performed another capture for both the servers. Here are they:
Milo: [milo.zip](https://github.com/eclipse/milo/files/5053906/milo.zip)
Other server: [other-server.zip](https://github.com/eclipse/milo/files/5053907/other-server.zip)

Thanks! :)

kevinherron, 2020-08-10:
It looks like you uploaded the same capture for the other server, there's still no browse request present.

kevinherron, 2020-08-10:
Might not need it though, looks like the SourceNode event field is being reported as `null` in the Milo capture.

kevinherron, 2020-08-10:
Can you paste your Milo test code into a gist so I can debug it?

achintya-kumar, 2020-08-10:
I made these changes :)
[image]

Here are the captures:
[captures.zip](https://github.com/eclipse/milo/files/5053961/captures.zip)

kevinherron, 2020-08-10:
What's the Type Definition for the ObjectNode you're adding properties to? `Identifiers.ObjectNode` is not the correct value to use in the `ModelChangeStructureDataType` - that's the only difference I can see from the other server right now.

If it's just a generic `UaObjectNode` you created with no special type definition try `Identifiers.BaseObjectType` instead.

kevinherron, 2020-08-10:
Oh, also the Verb seems to be different. Try setting the "Reference Added" bit instead.

achintya-kumar, 2020-08-11:
I made the following change:
[image]

UaExpert still doesn't seem to be reacting :(
[image]

Milo capture: 
[milo.zip](https://github.com/eclipse/milo/files/5054035/milo.zip)

kevinherron, 2020-08-11:
Settings the Reference Added bit would be a `ubyte(4)`, not `ubyte(2)`.

achintya-kumar, 2020-08-11:
OMG! It works 💯%

Apologies for the error. I was inspired by UnifiedAutomation's documentation of the **ModelChangeStructureDataType**.
[image]

As usual, thanks again! :D

closed 2020-08-11 by achintya-kumar

kevinherron, 2020-08-11:
Glad it works! I've never attempted this before. 👍