eclipse-milo/milo #677: Set target node for events

achintya-kumar, 2020-08-13:
Hey Kevin :)

I'm working on an example where I can post events on particular nodes of the address-space. 
For example, in **ExampleNamespace** in Milo, I'm trying to post 2 different events to 2 different Object-nodes.
[image]

However, upon checking in UaExpert for just one of the 2 Object nodes, I see both the events. Could you please point me to what I'm doing incorrect here?
[image]

kevinherron, 2020-08-13:
Hmm. Doesn’t look like anything is wrong... any chance those 2 Nodes have the same NodeId for some reason?

kevinherron, 2020-08-13:
I think what you're seeing in the screenshot is that objectAK1 is the EventNotifier, which is different than setting the SourceNode.

If you modified UaExpert to include the SourceNode in the event subscription you'd see that it is correctly sending whatever you set as source.

[image]
[image]

achintya-kumar, 2020-08-13:
Thank you for the response :) Please take a look at the following:

I wanted to show you how the 'objectAK' nodes are arranged in the address-space.
As you can see, they are present inside the **HelloWorld** folder. I'm setting the source node as follows:

- _eventNode1_ (with message _AK1 event message!_) to **objectAK1**
- _eventNode2_ (with message _AK2 event message!_) to **objectAK2**

**Goal:** When I subscribe to **objectAK1** in EventView, only the first event (with message _AK1 event message!_) should show up and not the second one.

[image]

[image]

[image]

[image]

kevinherron, 2020-08-13:
The problem here likely has to do with your usage of only a single EventBus and your implementation of the `onEventItems...` methods.

To do this correctly, one approach would be to create a separate EventBus associated with each ObjectNode that will act as an event notifier (i.e. it can be subscribed to by a client to receive events). Then, in the `onEventItemsCreated` method, you need to actually pay attention to which ObjectNode is being subscribed to so you can hook up the EventBus associated with that ObjectNode to the EventItems.

achintya-kumar, 2020-08-13:
Thanks, Kevin! That fixes my problem! :D

closed 2020-08-13 by achintya-kumar