eclipse-milo/milo #678: Browse to Information Model (Tree Hierarchy) Takes too much time

SavanTripathi, 2020-08-14:
Hello Kevin,

I am facing an issue while fetching (browsing) OPCUA Server of schneider plc modicon m580. There are 5000 Tags structured in Tree hierarchy( Parent - Child) format. While connecting and fetching information using BrowseNodeExample.java it is getting more and more time (Lazy Loading) and takes 1-2 hours. Any suggestions or solutions for this problem?

Thank You.

kevinherron, 2020-08-14:
This sounds like an issue with the server being slow and not anything that you can fix on the client side.

Perhaps you'll have to consider not doing a fully recursive browse from your application and instead browsing on demand. You could also verify that the server contains no "loops" in the node graph and the browse isn't stuck forever, though it sounds like you're saying it does finish eventually.

closed 2020-08-17 by kevinherron

SavanTripathi, 2020-08-17:
Thank you for your reply and  Sorry for late because doing some R&D with other clients...But I had tried different simulators...in UA Expert Same OPCUA Server is responding within 5-10 Seconds... So as you have suggested about browsing on demand, I am trying with it... because UA Expert is also client by Unified Automation..and Same server is responding with 5-20 seconds..

kevinherron, 2020-08-17:
UA Expert browses one node at a time as you expanded the node in the GUI. This Milo example is doing a fully recursive browse of the entire address space. You're talking about the difference between browsing _one_ Node and _all_ the Nodes.

SavanTripathi, 2020-08-18:
Thank you kevin. I got the point. I want to browse entire address space. So is there any solution or file in Milo for browsing nodes on click event? So that lazy loading can be reduced?

kevinherron, 2020-08-18:
There are no examples that show a GUI and let you browse one at a time.

From a programming perspective it's very easy, though: just don't continually browse using the results of the previous one. Ask the user what Node to browse or start from Root, then show them the immediate results, and ask which of those to browse before doing it.

SavanTripathi, 2020-08-19:
OK Thanks. Will Let you know after completing it.