eclipse-milo/milo #689: Write value return true for "isGood()",but it's not real write success!

somnuscq, 2020-09-03:
**Describe the bug**
A clear and concise description of behavior being reported and, if applicable, why it's incorrect.

**Expected behavior**
A clear and concise description of what you expected to happen instead.

**Logs and Packet Captures**
If applicable, add application logs or Wireshark captures to help explain your problem.

**Additional context**
Version of Eclipse Milo you are using, OPC UA client or software server you are interacting with, etc...

kevinherron, 2020-09-03:
If the server reports the write succeeded but in reality it didn't then there's nothing you can do from the client side. You should contact the server vendor.

somnuscq, 2020-09-03:
StatusCode status = statusCodes.get(0);
Do I need to ensure all statusCodes are good when batch writeValues?
Will it happens that  someone of the status is Bad?

kevinherron, 2020-09-03:
Yes, you should check all of them if you care about the results of the writes. It could happen that some are good and some aren’t.

closed 2020-09-04 by kevinherron

somnuscq, 2020-09-04:
Thanks a lot!

            
            
                
                    
                            
                                
                            
                            
                                283451661
                            
                    
                        
                            
                                    邮箱：283451661@qq.com
                            
                        
                
            
        Signature is customized by Netease Mail Master

On 09/04/2020 20:44, Kevin Herron wrote: 
Closed #689.

[
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/eclipse/milo/issues/689#event-3729574676",
"url": "https://github.com/eclipse/milo/issues/689#event-3729574676",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

somnuscq, 2020-10-09:
Unfortunately, there is still this problem! KEPServerEX 6 Administration