eclipse-milo/milo #690: Caused by: java.net.UnknownHostException，Upgrade sdk-client version from 0.2.4 to 0.4.3

liuxiaojianalibaba, 2020-09-04:
**Describe the bug**
Upgrade sdk-client version from 0.2.4 to 0.4.3, 0.2.4 version it works, but 0.4.3 version it does not work。

java.util.concurrent.ExecutionException: java.net.UnknownHostException: DESKTOP-K0PS3S5: nodename nor servname provided, or not known
	at java.util.concurrent.CompletableFuture.reportGet(CompletableFuture.java:357)
	at java.util.concurrent.CompletableFuture.get(CompletableFuture.java:1895)
	at com.aliyun.et.industry.igate.datamanager.OpcUaTest.main(OpcUaTest.java:31)
Caused by: java.net.UnknownHostException: DESKTOP-K0PS3S5: nodename nor servname provided, or not known
	at java.net.Inet6AddressImpl.lookupAllHostAddr(Native Method)
	at java.net.InetAddress$2.lookupAllHostAddr(InetAddress.java:928)
	at java.net.InetAddress.getAddressesFromNameService(InetAddress.java:1323)
	at java.net.InetAddress.getAllByName0(InetAddress.java:1276)
	at java.net.InetAddress.getAllByName(InetAddress.java:1192)
	at java.net.InetAddress.getAllByName(InetAddress.java:1126)
	at java.net.InetAddress.getByName(InetAddress.java:1076)
	at io.netty.util.internal.SocketUtils$8.run(SocketUtils.java:148)
	at io.netty.util.internal.SocketUtils$8.run(SocketUtils.java:145)
	at java.security.AccessController.doPrivileged(Native Method)
	at io.netty.util.internal.SocketUtils.addressByName(SocketUtils.java:145)
	at io.netty.resolver.DefaultNameResolver.doResolve(DefaultNameResolver.java:43)
	at io.netty.resolver.SimpleNameResolver.resolve(SimpleNameResolver.java:63)
	at io.netty.resolver.SimpleNameResolver.resolve(SimpleNameResolver.java:55)
	at io.netty.resolver.InetSocketAddressResolver.doResolve(InetSocketAddressResolver.java:57)
	at io.netty.resolver.InetSocketAddressResolver.doResolve(InetSocketAddressResolver.java:32)
	at io.netty.resolver.AbstractAddressResolver.resolve(AbstractAddressResolver.java:108)
	at io.netty.bootstrap.Bootstrap.doResolveAndConnect0(Bootstrap.java:200)
	at io.netty.bootstrap.Bootstrap.access$000(Bootstrap.java:46)
	at io.netty.bootstrap.Bootstrap$1.operationComplete(Bootstrap.java:180)
	at io.netty.bootstrap.Bootstrap$1.operationComplete(Bootstrap.java:166)
	at io.netty.util.concurrent.DefaultPromise.notifyListener0(DefaultPromise.java:577)
	at io.netty.util.concurrent.DefaultPromise.notifyListenersNow(DefaultPromise.java:551)
	at io.netty.util.concurrent.DefaultPromise.notifyListeners(DefaultPromise.java:490)
	at io.netty.util.concurrent.DefaultPromise.setValue0(DefaultPromise.java:615)
	at io.netty.util.concurrent.DefaultPromise.setSuccess0(DefaultPromise.java:604)
	at io.netty.util.concurrent.DefaultPromise.trySuccess(DefaultPromise.java:104)
	at io.netty.channel.DefaultChannelPromise.trySuccess(DefaultChannelPromise.java:84)
	at io.netty.channel.AbstractChannel$AbstractUnsafe.safeSetSuccess(AbstractChannel.java:984)
	at io.netty.channel.AbstractChannel$AbstractUnsafe.register0(AbstractChannel.java:504)
	at io.netty.channel.AbstractChannel$AbstractUnsafe.access$200(AbstractChannel.java:417)
	at io.netty.channel.AbstractChannel$AbstractUnsafe$1.run(AbstractChannel.java:474)
	at io.netty.util.concurrent.AbstractEventExecutor.safeExecute(AbstractEventExecutor.java:164)
	at io.netty.util.concurrent.SingleThreadEventExecutor.runAllTasks(SingleThreadEventExecutor.java:472)
	at io.netty.channel.nio.NioEventLoop.run(NioEventLoop.java:500)
	at io.netty.util.concurrent.SingleThreadEventExecutor$4.run(SingleThreadEventExecutor.java:989)
	at io.netty.util.internal.ThreadExecutorMap$2.run(ThreadExecutorMap.java:74)
	at java.lang.Thread.run(Thread.java:745)

**Expected behavior**
Is there any way that I do not need to config the host to avoid this.  Because it is already used, and I 
configure the endpointUrl in web page. I want users do not config the host.

kevinherron, 2020-09-04:
I can't imagine why an upgrade had any effect on this. Your client can't resolve the hostname "DESKTOP-K0PS3S5".

The workaround is well documented: https://stackoverflow.com/questions/63345326/connecting-to-an-opc-ua-server-using-eclipse-milo-fails-on-badhostunknown/63345707#63345707

liuxiaojianalibaba, 2020-09-04:
> I can't imagine why an upgrade had any effect on this. Your client can't resolve the hostname "DESKTOP-K0PS3S5".
> 
> The workaround is well documented: https://stackoverflow.com/questions/63345326/connecting-to-an-opc-ua-server-using-eclipse-milo-fails-on-badhostunknown/63345707#63345707

Thanks a lot for your reply, I can config the client's host to avoid this issue, but I don't want to solve it this way.  Before you replay, I have used a similar way to resolve it.

closed 2020-09-05 by kevinherron

tietnhathung, 2023-09-20:
I have problem when trying to connect to OPC using IP.
DiscoveryClient.getEndpoints auto replace my ip to  hostname that makes me unable to connect.
opc.tcp://192.168.31.230:49320 -> opc.tcp://AMI-SERVER-IOT:49320

kevinherron, 2023-09-20:
@tietnhathung see https://stackoverflow.com/questions/63345326/connecting-to-an-opc-ua-server-using-eclipse-milo-fails-on-badhostunknown/63345707#63345707