eclipse-milo/milo #691: a little bug in SubscriptionExample

RexCN, 2020-09-14:
**Describe the bug**
In org.eclipse.milo.examples.client.SubscriptionExample ,
I created multi MonitoredItemCreateRequest like        
List<UaMonitoredItem> items = subscription.createMonitoredItems(
                TimestampsToReturn.Both,
                newArrayList(**request1, request2, request3**),
                onItemCreated
).get(); 
It‘s never changed that UaMonitoredItem in callback and always were the last in MonitoredItemCreateRequests.
but value always worked.
private void onSubscriptionValue(UaMonitoredItem item, DataValue value) {
        logger.info(
                "subscription value received: item={}, value={}",
                **item.getReadValueId().getNodeId()**, value.getValue());
    }

**Expected behavior**
UaMonitoredItem ' s value  can be rightly in consumer callback like DataValue

kevinherron, 2020-09-14:
I'm not sure what you're trying to describe. Can you attach the full code with all modifications you've made?

RexCN, 2020-09-15:
I ignored that remark
// IMPORTANT: client handle must be unique per item within the context of a subscription.
now it works

        MonitoredItemCreateRequest request = new MonitoredItemCreateRequest(
                readValueId,
                MonitoringMode.Reporting,
                new MonitoringParameters(
                        subscription.nextClientHandle(),
                        1000.0,     // sampling interval
                        null,       // filter, null means use default
                        uint(10),   // queue size
                        true        // discard oldest
                )
        );
        MonitoredItemCreateRequest request1 = new MonitoredItemCreateRequest(
                readValueId1,
                MonitoringMode.Reporting,
                new MonitoringParameters(
                        subscription.nextClientHandle(),
                        1000.0,     // sampling interval
                        null,       // filter, null means use default
                        uint(10),   // queue size
                        true        // discard oldest
                )
        );

closed 2020-09-15 by RexCN

kevinherron, 2020-09-15:
Check out the easier to use subscription API in 0.5.0 as well: https://github.com/eclipse/milo/blob/master/milo-examples/client-examples/src/main/java/org/eclipse/milo/examples/client/ManagedSubscriptionDataExample.java

RexCN, 2020-09-16:
thank you so much