eclipse-milo/milo #692: BrowseAsyncExample fails to complete due to java.util.concurrent.ExecutionException: java.lang.NullPointerException

MattiasBjorkman, 2020-09-14:
**Describe the bug**
I'm using the code from browse example (https://github.com/eclipse/milo/blob/master/milo-examples/client-examples/src/main/java/org/eclipse/milo/examples/client/BrowseAsyncExample.java) on a test copy of a production OPC-UA server.

The problem is when try to run it i get the following error:

> java.util.concurrent.ExecutionException: java.lang.NullPointerException
	at java.base/java.util.concurrent.CompletableFuture.reportGet(CompletableFuture.java:395)
	at java.base/java.util.concurrent.CompletableFuture.get(CompletableFuture.java:1999)
        ....
Caused by: java.lang.NullPointerException
	at java.base/java.util.Collections.addAll(Collections.java:5451)
	at org.eclipse.milo.opcua.sdk.client.BrowseHelper.maybeBrowseNext(BrowseHelper.java:90)
	at org.eclipse.milo.opcua.sdk.client.BrowseHelper.lambda$browse$1(BrowseHelper.java:78)
	at java.base/java.util.concurrent.CompletableFuture$UniCompose.tryFire(CompletableFuture.java:1072)
	at java.base/java.util.concurrent.CompletableFuture.postComplete(CompletableFuture.java:506)
	at java.base/java.util.concurrent.CompletableFuture.complete(CompletableFuture.java:2073)
	at org.eclipse.milo.opcua.stack.client.UaStackClient.lambda$deliverResponse$5(UaStackClient.java:256)
	at org.eclipse.milo.opcua.stack.core.util.ExecutionQueue$Task.run(ExecutionQueue.java:119)
	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128)
	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628)
	... 1 more

The exception happens when I try to call `browseRecursive(client, tree).get()`.
```
public void run(OpcUaClient client, CompletableFuture<OpcUaClient> future) throws Exception {
        // synchronous connect
        client.connect().get();

        // start browsing at root folder
        UaNode rootNode = client.getAddressSpace().getNode(Identifiers.RootFolder);

        Tree<UaNode> tree = new Tree<>(rootNode);

        long startTime = System.nanoTime();
        browseRecursive(client, tree).get();    <--- This line result in the Exception. 
```

This is a result from the BrowseHelper. When you reach a "leaf" in the browsing of the nodes.
The `result.getReferences()` is `null` in my case when `nextContinuationPoint.isNull() == true`.
The `null` is added to the `references` list which is passed to the completableFuture.
```
  private static CompletableFuture<List<ReferenceDescription>> maybeBrowseNext(
        UaStackClient client,
        OpcUaSession session,
        List<ReferenceDescription> references,
        BrowseResult result
    ) {

        if (result.getStatusCode().isGood()) {
            Collections.addAll(references, result.getReferences());   < --  result.getReferences() == null when nextContinuationPoint.isNull() == true. 

            ByteString nextContinuationPoint = result.getContinuationPoint();

            if (nextContinuationPoint == null || nextContinuationPoint.isNull()) {
                return CompletableFuture.completedFuture(references);  <-- The null is passed to the completableFuture
            } else {
                return browseNext(client, session, nextContinuationPoint, references);
            }
        } else {
            return CompletableFuture.completedFuture(references);
        }
    }
```

**Expected behavior**
I have tested some and made a test code with a null check. This seem to work in my case. 
But I OPC-UA noob and don't have the whole picture.

```
private static CompletableFuture<List<ReferenceDescription>> maybeBrowseNext(
        UaStackClient client,
        OpcUaSession session,
        List<ReferenceDescription> references,
        BrowseResult result
    ) {

        if (result.getStatusCode().isGood()) {
            if (result.getReferences() != null) {
                Collections.addAll(references, result.getReferences());
            }
            
            ByteString nextContinuationPoint = result.getContinuationPoint();

            if (nextContinuationPoint == null || nextContinuationPoint.isNull()) {
                return CompletableFuture.completedFuture(references);
            } else {
                return browseNext(client, session, nextContinuationPoint, references);
            }
        } else {
            return CompletableFuture.completedFuture(references);
        }
    }
```

**Logs and Packet Captures**
This is a printout of the response.
BrowseResult(statusCode=StatusCode{name=Good, value=0x00000000, quality=good}, continuationPoint=ByteString{bytes=null}, references=null)], diagnosticInfos=[])

**Additional context**
I'm using the 5.0.0 version of Milo.

kevinherron, 2020-09-14:
Thank you. Do you know what stack/SDK this server was built with?

MattiasBjorkman, 2020-09-14:
Hi.
Not sure. But I see this if I browse the Server node:

> [http://opcfoundation.org/UA/,http://www.atvise.com/atServer/UA/]

So I guess its delivered by: http://www.atvise.com
I seems reasonable when if checked their website. But I dont really know anything more.

kevinherron, 2020-09-14:
Ok, thanks. I will fix this in Milo regardless, but this is technically a bug in the server.

OPC UA requires decoders to preserve the distinction between `null` (length -1) and empty (length=0 when decoding an array field. This server is returning `null` value for the `References[]` rather than an empty array, but the spec states it should be an empty array.

MattiasBjorkman, 2020-09-14:
Super thanks and thank you for all your effort in this project.

I will try to pass on the information to the server owner, that it not comply with standard.
But I'm pretty sure that it a commercial product. 
Thanks again.

MattiasBjorkman, 2020-09-14:
I tested your branch with the fix. It solved my problem.
Thanks again.

kevinherron, 2020-09-25:
@MattiasBjorkman 0.5.1 has been released, containing this fix.

PR #693 (2020-09-14): Guard against BrowseResult#getReferences() returning null
The spec says this should be an empty array when there are no references
but it has been observed that some out-of-spec servers will send a null
value instead.

fixes #692
commit 47ee5161 (2020-09-14): Guard against BrowseResult#getReferences() returning null (#693)

The spec says this should be an empty array when there are no references
but it has been observed that some out-of-spec servers will send a null
value instead.

fixes #692