eclipse-milo/milo #694: OPC-UA client fails for getting variable node, connecting to kepware server

sushantkale90, 2020-09-16:
**Describe the bug**
I'm trying to connect to kepware server simulation node using the OPC UA client 0.5.0 SDK
the connection succeeds and I'm able to write value to it.
when I try to use this connection to get the variable node that I wrote to it give me an error saying that
`UaException: status=Bad_UnexpectedError, message=java.util.concurrent.ExecutionException: java.lang.ClassCastException: org.eclipse.milo.opcua.stack.core.types.builtin.unsigned.UInteger cannot be cast to java.lang.Double`
up on inspecting it further it is seen that the sampling interval is by default set to 10 and it fails when ittried to get the variable node attributes for 
`        Double minimumSamplingInterval = (Double) attributeValues.get(13).getValue().getValue();`
as this is 10 and it gets a value of 10 as Uint.

**Expected behavior**
Since I was able to get the node and read the value using the previous SDK I should be able to get the node without any exception as nothing has changed

**Logs and Packet Captures**
```
Caused by: java.lang.ClassCastException: org.eclipse.milo.opcua.stack.core.types.builtin.unsigned.UInteger cannot be cast to java.lang.Double
	at org.eclipse.milo.opcua.sdk.client.AddressSpace.newVariableNode(AddressSpace.java:1221)
	at org.eclipse.milo.opcua.sdk.client.AddressSpace.lambda$getVariableNodeAsync$4(AddressSpace.java:310)
	at java.util.concurrent.CompletableFuture.uniApply(CompletableFuture.java:602)
	at java.util.concurrent.CompletableFuture$UniApply.tryFire(CompletableFuture.java:577)
	at java.util.concurrent.CompletableFuture.postComplete(CompletableFuture.java:474)
	at java.util.concurrent.CompletableFuture.complete(CompletableFuture.java:1962)
	at org.eclipse.milo.opcua.stack.client.UaStackClient.lambda$deliverResponse$5(UaStackClient.java:256)
	at org.eclipse.milo.opcua.stack.core.util.ExecutionQueue$Task.run(ExecutionQueue.java:119)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
	at java.lang.Thread.run(Thread.java:748)
```

**Additional context**
below is the capture from ua client tool
[image]

kevinherron, 2020-09-16:
I think you would have seen an error in the previous implementation had you actually tried to read the MinimumSamplingInterval attribute.

This is clearly a bug in Kepware. They appear to be returning a `UInt` value instead of `Double` when reading this attribute.

I'll see if there's something I can do on the Milo side to catch this error when the cast fails, but this should definitely be reported to Kepware and fixed there as well.

sushantkale90, 2020-09-16:
im not trying to read the sampling interval I'm just trying to read the variable node.

is there any other alternative to get the node without the properties, as I am blocked by this one.

the kepware version I'm using is `kepware V6`

kevinherron, 2020-09-16:
All of a Node's attribute are read when the instance is created.

You'll have to modify the code and build against a SNAPSHOT version for now. This was not a cast I expected could fail because the datatype of each attribute is such a fundamental part of the OPC UA spec.

This patch is a quick hack:
```
Index: opc-ua-sdk/sdk-client/src/main/java/org/eclipse/milo/opcua/sdk/client/AddressSpace.java
IDEA additional info:
Subsystem: com.intellij.openapi.diff.impl.patch.CharsetEP
<+>UTF-8
===================================================================
--- opc-ua-sdk/sdk-client/src/main/java/org/eclipse/milo/opcua/sdk/client/AddressSpace.java	(revision c8679e01d80f3686453aa820f0484c380282adb9)
+++ opc-ua-sdk/sdk-client/src/main/java/org/eclipse/milo/opcua/sdk/client/AddressSpace.java	(date 1600252693761)
@@ -1218,7 +1218,13 @@
         UInteger[] arrayDimensions = (UInteger[]) attributeValues.get(10).getValue().getValue();
         UByte accessLevel = (UByte) attributeValues.get(11).getValue().getValue();
         UByte userAccessLevel = (UByte) attributeValues.get(12).getValue().getValue();
-        Double minimumSamplingInterval = (Double) attributeValues.get(13).getValue().getValue();
+
+        Double minimumSamplingInterval = null;
+        Object minimumSamplingIntervalObject = attributeValues.get(13).getValue().getValue();
+        if (minimumSamplingIntervalObject instanceof Double) {
+            minimumSamplingInterval = (Double) minimumSamplingIntervalObject;
+        }
+
         Boolean historizing = (Boolean) attributeValues.get(14).getValue().getValue();
 
         VariableTypeManager.VariableNodeConstructor constructor = client.getVariableTypeManager()
```

I don't know what the actual fix will look like. This server so so fundamentally broken that I'm at a bit of a loss.

kevinherron, 2020-09-16:
Fix for this is pending along with a fix for related #696

closed 2020-09-16 by kevinherron

sushantkale90, 2020-09-19:
@kevinherron  do you know the approx timeline for 0.5.1 release

kevinherron, 2020-09-19:
@sushantkale90 I was hoping a few more issues would shake out first, but if you're unable to use a SNAPSHOT version then I can prepare a release soon.

kevinherron, 2020-09-25:
@sushantkale90 0.5.1 has been released.

PR #697 (2020-09-16): Handle null optional attributes and unexpected types
fixes #694 and #696