eclipse-milo/milo #695: How to set a proxy for client part

lloubaud, 2020-09-16:
I want to know if it is possible to set a proxy for the client part of the library.
I tried in java param:
-DsocksProxyHost=proxy
-DsocksProxyPort=8080
-Dhttp.proxyHost=proxy
-Dhttp.proxyPort=8080
-Dhttps.proxyHost=proxy
-Dhttps.proxyPort=8080
-Dhttp.proxy=proxy
-Dhttps.proxy=proxy

and in env variable:
HTTPS_PROXY=proxy:8080
HTTP_PROXY=proxy:8080

but none is able to pass trough my proxy.

kevinherron, 2020-09-16:
There is no proxy support. I'm not sure an HTTP proxy would even work. Maybe SOCKS though.

lloubaud, 2020-09-16:
I tried to set socks proxy also but still not working.

Do you plan to add this feature ?

THKS

kevinherron, 2020-09-16:
> Do you plan to add this feature ?

You're the first to ask, so I hadn't planned on it, but we'll see. I'd rather not implement and maintain features that will not be used often. I've converted this issue into a feature request.

lloubaud, 2020-09-16:
Thanks @kevinherron for your quick support

closed 2020-09-16 by lloubaud