eclipse-milo/milo #705: How to implement addNodes() / onAddNodes()

phillip-h, 2020-10-06:
We're trying to allow our client to add nodes to the server. So far we've implemented `addNodes()` in our namespace (extending `ManagedNamespace`), but we still get a `Bad_ServiceUnsupported` from `onAddNodes()` in `DefaultNodeManagementServiceSet`. I'm guessing what we need to do is implement our own `NodeManagementServiceSet` that passes through the ServiceRequests to our Namespace, but how do we get the server to use the custom implementation?

kevinherron, 2020-10-06:
It looks like a real implementation for `DefaultNodeManagementServiceSet` is just something that needs to be implemented in Milo itself. I thought this had already been done but clearly not.

Making the session manager and service sets properly extensible is something I've got on my radar for a future (after 1.0) version.

If you'd like to PR an implementation that would be fine, otherwise I can try to get to it soon.

kevinherron, 2020-10-07:
Tracking this in #706.

kevinherron, 2020-10-07:
Began work in https://github.com/eclipse/milo/pull/707.

kevinherron, 2020-10-07:
@phillip-h can you build this branch and verify that your `AddressSpace` implementation now receives node management calls when you override the default implementation?

phillip-h, 2020-10-07:
I built from commit `ff6c2c6a` and can confirm that our overridden addNodes() is now called as expected.

closed 2020-10-07 by phillip-h