eclipse-milo/milo #708: Proper client subscription error handling

jadamroth, 2020-10-07:
Hi,

From client side, what is the proper way to handle errors for subscriptions to server nodes that have lost connection?

I see on the wiki that a disconnected client will try to automatically reconnect after you initially call client.connect().get(); which seems to attempt to reconnect to the server indefinitely until we explicitly call client.disconnect()

1. How often does this attempted reconnect occur? 
2. Is there a callback when the disconnect happens or a method like isDisconnected()? I'd like to be notified, so I can know to resubscribe to the nodes again
3. Should error handling (after successfully subscribing to the server) for opcua clients be treated the same as the individual nodes?

kevinherron, 2020-10-07:
Reconnects occur immediately and then follow an exponential backoff to 16 seconds apart.

You can get notified when the _Session_ goes active or inactive by adding a `SessionActivityListener` via `OpcUaClient::addSessionActivityListener`.

You can also add a `SubscriptionListener` to the `OpcUaSubscriptionManager` via `UaSubscriptionManager::addSubscriptionListener`. The reconnection will automatically resume your session or transfer subscriptions if possible. If neither of these can be done then `SubscriptionListener::onSubscriptionTransferFailed` is invoked. This is where you would re-create your subscription and re-subscribe your nodes. Take care not to block inside this callback, though.

jadamroth, 2020-10-07:
Thank you for the detailed response. 

I will close for now and reopen if I have any issues

jadamroth, 2020-11-10:
Is it okay to have a blocking mysql update statement inside of the subscription manager listener onKeepAlive()? I just update the timestamp which I use as a healthcheck for my connections.

kevinherron, 2020-11-10:
@jadamroth it's not ideal. Both `SubscriptionListener::onKeepAlive` and `NotificationListener::onKeepAliveNotification` are invoked on the delivery queue while holding the notification semaphore, so you will be blocking further delivery of data change notifications while you run your query. 

Further, these subscription keep alive notifications probably aren't the health check metric you want. They only occur in the _absence_ of other activity on the subscription it is delivered to. So if the items you subscribe to are generally active enough to avoid this your health check timestamp will not be updated.

jadamroth, 2020-11-10:
Got it. Thank you

closed 2020-11-10 by jadamroth