eclipse-milo/milo #711: AccessControlException when creating JAXBContext in BsdParser

WimVriend, 2020-10-10:
**Describe the bug**
Line 40 of BsdParser throws a security error, which results in failure to create decoders for ExtensionObjects.

`        JAXBContext context = JAXBContext.newInstance(ObjectFactory.class);`

It reads:

> 
> java.security.AccessControlException: access denied ("java.lang.RuntimePermission" "accessDeclaredMembers")
> 	at java.security.AccessControlContext.checkPermission(AccessControlContext.java:472)
> 	at java.security.AccessController.checkPermission(AccessController.java:884)
> 	at java.lang.SecurityManager.checkPermission(SecurityManager.java:549)
> 	at java.lang.Class.checkMemberAccess(Class.java:2348)
> 	at java.lang.Class.getDeclaredField(Class.java:2067)
> 	at com.sun.xml.internal.bind.v2.model.nav.ReflectionNavigator.getDeclaredField(ReflectionNavigator.java:273)
> 	at com.sun.xml.internal.bind.v2.model.nav.ReflectionNavigator.getDeclaredField(ReflectionNavigator.java:47)
> 	at com.sun.xml.internal.bind.v2.model.impl.RegistryInfoImpl.<init>(RegistryInfoImpl.java:70)
> 	at com.sun.xml.internal.bind.v2.model.impl.ModelBuilder.addRegistry(ModelBuilder.java:402)
> 	at com.sun.xml.internal.bind.v2.model.impl.ModelBuilder.getTypeInfo(ModelBuilder.java:372)
> 	at com.sun.xml.internal.bind.v2.runtime.JAXBContextImpl.getTypeInfoSet(JAXBContextImpl.java:439)
> 	at com.sun.xml.internal.bind.v2.runtime.JAXBContextImpl.<init>(JAXBContextImpl.java:277)
> 	at com.sun.xml.internal.bind.v2.runtime.JAXBContextImpl.<init>(JAXBContextImpl.java:124)
> 	at com.sun.xml.internal.bind.v2.runtime.JAXBContextImpl$JAXBContextBuilder.build(JAXBContextImpl.java:1123)
> 	at com.sun.xml.internal.bind.v2.ContextFactory.createContext(ContextFactory.java:147)
> 	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
> 	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
> 	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
> 	at java.lang.reflect.Method.invoke(Method.java:498)
> 	at javax.xml.bind.ContextFinder.newInstance(ContextFinder.java:247)
> 	at javax.xml.bind.ContextFinder.newInstance(ContextFinder.java:234)
> 	at javax.xml.bind.ContextFinder.find(ContextFinder.java:462)
> 	at javax.xml.bind.JAXBContext.newInstance(JAXBContext.java:641)
> 	at javax.xml.bind.JAXBContext.newInstance(JAXBContext.java:584)
> 	at gti.plc.server.miloopcua.sdk.client.DataTypeDictionaryReader.createDataTypeDictionary(DataTypeDictionaryReader.java:326)

**Expected behavior**
The line should create a JAXBContext without error.

**Additional context**
The project is built using Eclipse 2019-06 and uses Java version 1.8.0.172. Version of Eclipse Milo is 0.4.3 and we are successfully connecting with a Siemens S7-1512 PLC.

The problem is clearly related to issue [651](https://github.com/eclipse/milo/issues/651). If we execute the statement of line 40 elsewhere in the code (for example, before calling '_createDataTypeDictionary_' in _DataTypeDictionaryReader_ the context is successfully created. It seems that the problem is caused by the fact that it runs in a different thread.

We have tried the solutions mentioned in [651](https://github.com/eclipse/milo/issues/651), but to no avail. _setContextClassLoader_ also throws a security error, using _AccessController.doPrivileged_ does not change the problem and calling _JAXBContext.newInstance("org.opcfoundation.opcua.binaryschema", getClass().getClassLoader());_ also seems to do nothing.

We would really appreciate some help, because this is a showstopper at this moment...

Maybe this is a silly question to end with: why is the CompletableFuture stuff used to create the decoder(s)? Can it be converted to sequential  code? (I am new to lambda's and such, and do not entirely understand the code of the DictionaryReader...)

kevinherron, 2020-10-10:
This sounds like the same thing you described in https://github.com/eclipse/milo/issues/570 and I can only imagine it's the same problem: your Java application environment has a SecurityManager configured that is preventing this from working.

kevinherron, 2020-10-10:
> Maybe this is a silly question to end with: why is the CompletableFuture stuff used to create the decoder(s)? Can it be converted to sequential code? (I am new to lambda's and such, and do not entirely understand the code of the DictionaryReader...)

Because it involves many rounds of back-and-forth communication with the server to get all the information needed. It can take 10s of seconds to finish reading everything from some servers with larger dictionaries. If it was written in a synchronous/blocking manner this would tie up a shared thread for all that time, which might be fine when you only have a few client connections but doesn't scale as well when there are thousands. Much of Milo's implementation details are written in an async/nonblocking manner in the name of performance and scalability.

WimVriend, 2020-10-11:
Thanks again for your quick reply. 

As I also mentioned in the other thread, the problem seems to be more subtle. On my original development PC, the same project (with the same security settings) works fine with milo . After I moved the branch to another PC, this problem surfaced. It took very long to pinpoint the actual culprit, because the error(s) never showed until I changed the logging settings. Only after moving all source code to the project and debugging the actual code, could I discover that the JAXBContext could not be created. Additionally I also tried the milo ExtensionObject example on the same PC: that works fine (so it's not the network, the PLC, Windows or Java-version that cause the problem).

Now that the culprit is found: how to solve it? The thread calling the CompletableFuture is able create the context, so the security issue arises when a new thread is created. I would expect that it inherits the Sceurity context from its parent, but it apparently does not?

WimVriend, 2020-10-11:
> Because it involves many rounds of back-and-forth communication with the server to get all the information needed. It can take 10s of seconds to finish reading everything from some servers with [...]

When debugging the DataTypeDictionary creation process, I see that the StructureType definitions are retrieved from the PLC very quickly (less than 1 second). They look like this:
```

<opc:StructuredType BaseType="ua:ExtensionObject" Name="BUTTON_STATUS">
<opc:Field TypeName="opc:Boolean" Name="fInLogical"/>
<opc:Field TypeName="opc:Boolean" Name="fAck"/>
<opc:Field TypeName="opc:Boolean" Name="fBlink"/>
<opc:Field TypeName="opc:Boolean" Name="fSpare3"/>
<opc:Field TypeName="opc:Boolean" Name="fInDelRiseEdge"/>
<opc:Field TypeName="opc:Boolean" Name="fInDelFallEdge"/>
<opc:Field TypeName="opc:Boolean" Name="fSpare6"/>
<opc:Field TypeName="opc:Boolean" Name="fSpare7"/>
<opc:Field TypeName="opc:Boolean" Name="fSpare10"/>
<opc:Field TypeName="opc:Boolean" Name="fSpare11"/>
<opc:Field TypeName="opc:Boolean" Name="fSpare12"/>
<opc:Field TypeName="opc:Boolean" Name="fSpare13"/>
<opc:Field TypeName="opc:Boolean" Name="fSpare14"/>
<opc:Field TypeName="opc:Boolean" Name="fSpare15"/>
<opc:Field TypeName="opc:Boolean" Name="fSpare16"/>
<opc:Field TypeName="opc:Boolean" Name="fSpare17"/>
</opc:StructuredType>

```
It looks like all information required is right there: am I mistaken?

kevinherron, 2020-10-11:
I assume your application is running inside of some kind of application server? Tomcat, GlassFish, etc...?

It probably has some kind of policy file that can be configured.

I missed this originally but you also said:
> setContextClassLoader also throws a security error
Which is more evidence that this is some kind of problem with the environment you’re trying to run this in.

I don’t know _anything_ about your application, so I’m not sure how much more advice I can offer. Maybe if you post to the mailing list someone else will have had a similar experience.

kevinherron, 2020-10-13:
Interesting.

It seems this line is using `thenComposeAsync` without there being much reason: https://github.com/eclipse/milo/blob/937139b1db84faf9e8f960e3aa0ec3fc77479fb9/opc-ua-sdk/dictionary-reader/src/main/java/org/eclipse/milo/opcua/sdk/client/DataTypeDictionaryReader.java#L199

I think if I use `thenCompose` instead of `thenComposeAsync` then there would be no reason this code jumps onto the default ForkJoinPool rather than staying on one of Milo’s own thread pools.

WimVriend, 2020-10-14:
I decided you try the DictionaryReader without 'Async' but I think it was there for a reason: the process seems to 'hang' without it.
For now I will stick with the ForkJoinPool ...

kevinherron, 2020-10-14:
It was async for a reason; removing it lead to a stack overflow. 

Instead I modified it to use the configured executor from the client config: https://github.com/eclipse/milo/commit/fc6202763f55994784b053ef5cd760630b5c1a76

WimVriend, 2020-10-14:
Great! Thanks as always :)

kevinherron, 2020-10-14:
I am going to close this now.

Between the strange behavior you've uncovered when using a SecurityManager and a ForkJoinPool and this change in Milo to prevent the DataTypeDictionaryReader from using the FJP I think it's as close to resolved as we'll get.

closed 2020-10-14 by kevinherron