eclipse-milo/milo #713: Loading official NodeSets from https://github.com/OPCFoundation/UA-Nodeset

ebrithiljonas, 2020-10-14:
Is it possible to load NodeSets from the official OPC Foundation [repository ](https://github.com/OPCFoundation/UA-Nodeset) into Milo? And if not, is that feature planned? BEcause it would make it much easier to define types for special cases like the umati standart.

kevinherron, 2020-10-14:
This is not officially supported yet, but there’s a WIP node set parser and a comment explaining how it can be used here: https://github.com/digitalpetri/opc-ua-nodeset-parser/issues/7#issuecomment-609451277

I do plan to have better support for this in the future.

GoetzGoerisch, 2020-10-14:
@achintya-kumar could you elaborate one how you did this for us?

kevinherron, 2020-10-14:
I should clarify: the comment and code I linked to will load a NodeSet XML into an address space, but it does not have anything to do with generating ObjectType or VariableType nodes or struct instances for a given NodeSet. That is a separate concern.

ebrithiljonas, 2020-10-14:
So basically it is not possible to generate Object- and VariableTypes from an xml file with the current version?

kevinherron, 2020-10-14:
> So basically it is not possible to generate Object- and VariableTypes from an xml file with the current version?

It's not, and the code generators I am building will not be part of the Milo project. 

They _may_ be open source but I won't decide that until they are done.

ebrithiljonas, 2020-10-14:
Okey, thank you for the quick replies and the clarification.

closed 2020-10-14 by ebrithiljonas