eclipse-milo/milo #725: OPC UA TSN

sarthak-k, 2020-11-03:
I was reading about OPC UA TSN but couldn't find much about implementation. Does the Eclipse Milo have any plans in the future to support Pub/Sub over TSN?

kevinherron, 2020-11-03:
Sarthak,

I don't know enough about TSN to tell you if that's even technically
possible from a high level language like Java. Either way I wouldn't count
on it happening any time soon.

sarthak-k, 2020-11-03:
Kevin,

I guess this is a very primary thing right now, may be in the future when standards come.

Closing for now.

closed 2020-11-03 by sarthak-k

GoetzGoerisch, 2020-11-04:
To get an update on the state of FLC please view the OPC Day Recap: https://opcconnect.opcfoundation.org/2020/09/watch-download-opc-day-international-2020/

[TSN Sub Group Update](https://www.youtube.com/watch?v=dXNoFIfsEEM&list=PLROM1mLWekVBXVlqgAuyIZih-KA_wcX_I&index=4&t=0s)
Alexander Ziegler, Siemens (Chairperson of Group)

[Field Level Communications Initiative – Overview & Update](https://www.youtube.com/watch?v=iCoI13F29FQ&list=PLROM1mLWekVBXVlqgAuyIZih-KA_wcX_I&index=7&t=0s)
Peter Lutz, Director, OPCF FLC

[Field Level Communications Initiative – Deep Dive C2C](https://www.youtube.com/watch?v=iCoI13F29FQ&list=PLROM1mLWekVBXVlqgAuyIZih-KA_wcX_I&index=7&t=0s)
Paul Brooks, Rockwell (Member of FLC Steering Committee)