eclipse-milo/milo #727: Get Engineering Unit Information While Browsing OPC Server

sarthak-k, 2020-11-12:
The current scenrio recursively get all the nodes form OPC Server, there are certain nodes which of the type Analog and have additional information like Engineering Units but using Eclipse Milo that information can not be retrieved because processing get blocked after calling `getEngineeringUnits`. Is this the way to get or I'm doing something wrong?

```java
if (uaNode.getClass().equals(AnalogItemTypeNode.class)) {
            try {
                AnalogItemTypeNode analogItemTypeNode = (AnalogItemTypeNode) uaNode;
                EUInformation euInformation = analogItemTypeNode.getEngineeringUnits();
                message.append("Eu Information: ").append(euInformation.getDisplayName());
            } catch (Exception e) {
                System.out.println(e.getMessage());
            }
        }
```

kevinherron, 2020-11-12:
You should be able to call `getEngineeringUnits` but if the AnalogItemTypeNode in the address space doesn't have an EngineeringUnits property then it will throw UaException.

You can run this modified version of BrowseExample to see:

```java
package org.eclipse.milo.examples.client;

import java.util.List;
import java.util.concurrent.CompletableFuture;

import org.eclipse.milo.opcua.sdk.client.OpcUaClient;
import org.eclipse.milo.opcua.sdk.client.model.types.variables.AnalogItemType;
import org.eclipse.milo.opcua.sdk.client.nodes.UaNode;
import org.eclipse.milo.opcua.stack.core.UaException;
import org.eclipse.milo.opcua.stack.core.types.builtin.NodeId;
import org.eclipse.milo.opcua.stack.core.types.structured.EUInformation;
import org.eclipse.milo.opcua.stack.core.types.structured.Range;
import org.slf4j.Logger;
import org.slf4j.LoggerFactory;

public class BrowseNodeExample implements ClientExample {

    public static void main(String[] args) throws Exception {
        BrowseNodeExample example = new BrowseNodeExample();

        new ClientExampleRunner(example, false).run();
    }

    private final Logger logger = LoggerFactory.getLogger(getClass());

    @Override
    public void run(OpcUaClient client, CompletableFuture<OpcUaClient> future) throws Exception {
        // synchronous connect
        client.connect().get();

        NodeId root = NodeId.parse("ns=2;s=CTT/Static/DA Profile/Analog Type");

        // start browsing at root folder
        browseNode("", client, root);

        future.complete(client);
    }

    @Override
    public String getEndpointUrl() {
        return "opc.tcp://milo.digitalpetri.com:62541/milo";
    }

    private void browseNode(String indent, OpcUaClient client, NodeId browseRoot) {
        try {
            List<? extends UaNode> nodes = client.getAddressSpace().browseNodes(browseRoot);

            for (UaNode node : nodes) {
                logger.info("{} Node={}", indent, node.getBrowseName().getName());

                if (node instanceof AnalogItemType) {
                    try {
                        Range euRange = ((AnalogItemType) node).getEuRange();
                        logger.info("{} euRange={}", indent, euRange);
                    } catch (UaException e) {
                        logger.warn("Error getting EURange value", e);
                    }
                    try {
                        EUInformation engineeringUnits = ((AnalogItemType) node).getEngineeringUnits();
                        logger.info("{} engineeringUnits={}", indent, engineeringUnits);
                    } catch (UaException e) {
                        logger.warn("Error getting EngineeringUnits value", e);
                    }
                }

                // recursively browse to children
                browseNode(indent + "  ", client, node.getNodeId());
            }
        } catch (UaException e) {
            logger.error("Browsing nodeId={} failed: {}", browseRoot, e.getMessage(), e);
        }
    }

}
```

sarthak-k, 2020-11-12:
I can confirm that engineering units are available and trying out the above code does not get engineering units or range. It's just gets blocked after calling either `getEuRange` or `getEngineeringUnits'. For the reference, I'm attaching the screenshot of OPC Server Analog DataType:

[image]

kevinherron, 2020-11-12:
What version are you using? What server are you connected to?

You may just have to get in there and debug this yourself if it's not something I can reproduce.

sarthak-k, 2020-11-12:
I'm running simulated server on my own machine with random analog type of nodes being added. Both the server and client are built using latest version i.e 0.5.3.

kevinherron, 2020-11-12:
Maybe an uncaught exception or some other problem in your server implementation is leading to no response ever being sent?

I modified the demo server to add an EngineeringUnit value to each of the analog nodes and then pointed the modified example at the modified demo server and it read them all without issue:
```
13:12:45.901 [main] INFO  o.e.m.e.client.BrowseNodeExample -  Node=Byte
13:12:45.920 [main] INFO  o.e.m.e.client.BrowseNodeExample -  euRange=Range(low=0.0, high=100.0)
13:12:45.926 [main] INFO  o.e.m.e.client.BrowseNodeExample -  engineeringUnits=EUInformation(namespaceUri=http://www.opcfoundation.org/UA/units/un/cefact, unitId=4408652, displayName=LocalizedText{text=°C, locale=en}, description=LocalizedText{text=degree Celsius, locale=en})
13:12:45.929 [main] INFO  o.e.m.e.client.BrowseNodeExample -    Node=EURange
13:12:45.932 [main] INFO  o.e.m.e.client.BrowseNodeExample -    Node=EngineeringUnits
13:12:45.940 [main] INFO  o.e.m.e.client.BrowseNodeExample -  Node=Double
13:12:45.952 [main] INFO  o.e.m.e.client.BrowseNodeExample -  euRange=Range(low=0.0, high=100.0)
13:12:45.958 [main] INFO  o.e.m.e.client.BrowseNodeExample -  engineeringUnits=EUInformation(namespaceUri=http://www.opcfoundation.org/UA/units/un/cefact, unitId=4408652, displayName=LocalizedText{text=°C, locale=en}, description=LocalizedText{text=degree Celsius, locale=en})
13:12:45.960 [main] INFO  o.e.m.e.client.BrowseNodeExample -    Node=EURange
13:12:45.962 [main] INFO  o.e.m.e.client.BrowseNodeExample -    Node=EngineeringUnits
13:12:45.963 [main] INFO  o.e.m.e.client.BrowseNodeExample -  Node=Float
13:12:45.969 [main] INFO  o.e.m.e.client.BrowseNodeExample -  euRange=Range(low=0.0, high=100.0)
13:12:45.974 [main] INFO  o.e.m.e.client.BrowseNodeExample -  engineeringUnits=EUInformation(namespaceUri=http://www.opcfoundation.org/UA/units/un/cefact, unitId=4408652, displayName=LocalizedText{text=°C, locale=en}, description=LocalizedText{text=degree Celsius, locale=en})
13:12:45.976 [main] INFO  o.e.m.e.client.BrowseNodeExample -    Node=EURange
13:12:45.977 [main] INFO  o.e.m.e.client.BrowseNodeExample -    Node=EngineeringUnits
13:12:45.979 [main] INFO  o.e.m.e.client.BrowseNodeExample -  Node=Int16
13:12:45.984 [main] INFO  o.e.m.e.client.BrowseNodeExample -  euRange=Range(low=0.0, high=100.0)
13:12:45.988 [main] INFO  o.e.m.e.client.BrowseNodeExample -  engineeringUnits=EUInformation(namespaceUri=http://www.opcfoundation.org/UA/units/un/cefact, unitId=4408652, displayName=LocalizedText{text=°C, locale=en}, description=LocalizedText{text=degree Celsius, locale=en})
13:12:45.990 [main] INFO  o.e.m.e.client.BrowseNodeExample -    Node=EURange
13:12:45.991 [main] INFO  o.e.m.e.client.BrowseNodeExample -    Node=EngineeringUnits
13:12:45.992 [main] INFO  o.e.m.e.client.BrowseNodeExample -  Node=Int32
13:12:45.997 [main] INFO  o.e.m.e.client.BrowseNodeExample -  euRange=Range(low=0.0, high=100.0)
13:12:46.003 [main] INFO  o.e.m.e.client.BrowseNodeExample -  engineeringUnits=EUInformation(namespaceUri=http://www.opcfoundation.org/UA/units/un/cefact, unitId=4408652, displayName=LocalizedText{text=°C, locale=en}, description=LocalizedText{text=degree Celsius, locale=en})
13:12:46.005 [main] INFO  o.e.m.e.client.BrowseNodeExample -    Node=EURange
13:12:46.007 [main] INFO  o.e.m.e.client.BrowseNodeExample -    Node=EngineeringUnits
13:12:46.008 [main] INFO  o.e.m.e.client.BrowseNodeExample -  Node=Int64
13:12:46.012 [main] INFO  o.e.m.e.client.BrowseNodeExample -  euRange=Range(low=0.0, high=100.0)
13:12:46.015 [main] INFO  o.e.m.e.client.BrowseNodeExample -  engineeringUnits=EUInformation(namespaceUri=http://www.opcfoundation.org/UA/units/un/cefact, unitId=4408652, displayName=LocalizedText{text=°C, locale=en}, description=LocalizedText{text=degree Celsius, locale=en})
13:12:46.017 [main] INFO  o.e.m.e.client.BrowseNodeExample -    Node=EURange
13:12:46.018 [main] INFO  o.e.m.e.client.BrowseNodeExample -    Node=EngineeringUnits
13:12:46.019 [main] INFO  o.e.m.e.client.BrowseNodeExample -  Node=SByte
13:12:46.025 [main] INFO  o.e.m.e.client.BrowseNodeExample -  euRange=Range(low=0.0, high=100.0)
13:12:46.030 [main] INFO  o.e.m.e.client.BrowseNodeExample -  engineeringUnits=EUInformation(namespaceUri=http://www.opcfoundation.org/UA/units/un/cefact, unitId=4408652, displayName=LocalizedText{text=°C, locale=en}, description=LocalizedText{text=degree Celsius, locale=en})
13:12:46.031 [main] INFO  o.e.m.e.client.BrowseNodeExample -    Node=EURange
13:12:46.033 [main] INFO  o.e.m.e.client.BrowseNodeExample -    Node=EngineeringUnits
13:12:46.034 [main] INFO  o.e.m.e.client.BrowseNodeExample -  Node=UInt16
13:12:46.041 [main] INFO  o.e.m.e.client.BrowseNodeExample -  euRange=Range(low=0.0, high=100.0)
13:12:46.044 [main] INFO  o.e.m.e.client.BrowseNodeExample -  engineeringUnits=EUInformation(namespaceUri=http://www.opcfoundation.org/UA/units/un/cefact, unitId=4408652, displayName=LocalizedText{text=°C, locale=en}, description=LocalizedText{text=degree Celsius, locale=en})
13:12:46.046 [main] INFO  o.e.m.e.client.BrowseNodeExample -    Node=EURange
13:12:46.047 [main] INFO  o.e.m.e.client.BrowseNodeExample -    Node=EngineeringUnits
13:12:46.048 [main] INFO  o.e.m.e.client.BrowseNodeExample -  Node=UInt32
13:12:46.052 [main] INFO  o.e.m.e.client.BrowseNodeExample -  euRange=Range(low=0.0, high=100.0)
[... cut ...]
13:12:46.063 [main] INFO  o.e.m.e.client.BrowseNodeExample -  euRange=Range(low=0.0, high=100.0)
13:12:46.066 [main] INFO  o.e.m.e.client.BrowseNodeExample -  engineeringUnits=EUInformation(namespaceUri=http://www.opcfoundation.org/UA/units/un/cefact, unitId=4408652, displayName=LocalizedText{text=°C, locale=en}, description=LocalizedText{text=degree Celsius, locale=en})
13:12:46.067 [main] INFO  o.e.m.e.client.BrowseNodeExample -    Node=EURange
13:12:46.069 [main] INFO  o.e.m.e.client.BrowseNodeExample -    Node=EngineeringUnits
```

sarthak-k, 2020-11-12:
```java
@Override
    public CompletableFuture<TranslateBrowsePathsToNodeIdsResponse> translateBrowsePaths(List<BrowsePath> browsePaths) {
        return getSession().thenCompose(session -> {
            TranslateBrowsePathsToNodeIdsRequest request = new TranslateBrowsePathsToNodeIdsRequest(
                newRequestHeader(session.getAuthenticationToken()),
                a(browsePaths, BrowsePath.class)
            );

            return sendRequest(request);
        });
    }
```

I can debug the client till this point. After the sendRequest nothing could be debug or any fault was registered (I added a service fault listener but of no help).

kevinherron, 2020-11-12:
So maybe there's an error on the server side when it does the TranslateBrowsePathsToNodeIds service.

sarthak-k, 2020-11-12:
I also tried using iba Systems OPC UA Server and the similar problems occured, after calling `getEngineeringUnits` no further information is loaded or error is reported.

kevinherron, 2020-11-12:
If you capture this on Wireshark do you see a response to the TranslateBrowsePathsToNodeIdsRequest? Can you capture the browse traffic against this other server and upload it?

sarthak-k, 2020-11-12:
I can get the following information if I stop calling `getEngineeringUnits` but is this something valid or it's a structural issue:

          Class : class org.eclipse.milo.opcua.sdk.client.model.nodes.variables.AnalogItemTypeNode NodeId : NodeId{ns=3, id=V:0.3.38.0.14} Name: ************************
             Class : class org.eclipse.milo.opcua.sdk.client.model.nodes.variables.PropertyTypeNode NodeId : NodeId{ns=3, id=P:0.3.38.0.14.0} Name: EURange
             Class : class org.eclipse.milo.opcua.sdk.client.model.nodes.variables.PropertyTypeNode NodeId : NodeId{ns=3, id=P:0.3.38.0.14.1} Name: EngineeringUnits

sarthak-k, 2020-11-12:
I can get the EU Information by doing so:

```java
 PropertyTypeNode node = (PropertyTypeNode) uaNode;
 ExtensionObject xo = (ExtensionObject) node.getValue().getValue().getValue();
 Object decoded = xo.decode(opcBrowser.client.getSerializationContext());
 EUInformation euInformation = (EUInformation) decoded;
 System.out.println(euInformation);
```

kevinherron, 2020-11-12:
Did you get a Wireshark capture against the other server and see if the translate request comes back?

sarthak-k, 2020-11-12:
Have the WireShark Messages attached herewith.

[OPC_UA.zip](https://github.com/eclipse/milo/files/5533627/OPC_UA.zip)

kevinherron, 2020-11-12:
What IPs/ports are being used? I don't see any OPC UA traffic on the default ports.

sarthak-k, 2020-11-12:
OPC Server running on 5000 and IP's are local intranet one.

kevinherron, 2020-11-12:
There's no traffic on port 5000 in the capture you uploaded.

sarthak-k, 2020-11-12:
@kevinherron Hey! So sorry for all this. It was more of concurrency issue rather than OPC UA issue (although, iba Systems does not implement TranslateNodePathToNodeId I guess, I'm receiving UaException: status=Bad_NoMatch, message=The requested operation has no match to return.). I just had to put this job on different threads rather than the same as recursive thread.

closed 2020-11-12 by sarthak-k

kevinherron, 2020-11-12:
> although, iba Systems does not implement TranslateNodePathToNodeId

Strange, the service isn't optional. Are you sure there is an EngineeringUnits node in that server with the correct QualifiedName for its BrowseName? The QualifiedName must have the correct namespace index and name.

sarthak-k, 2020-11-13:
Qualified is not valid for few signals, there are signals which have valid namespace URI and valid EU Information. Trying to talk with respective owner to put the correct units so it can be reflected correctly.