eclipse-milo/milo #731: LDS,LDS-ME, GDS implementation

RakshanPremsagar, 2020-11-20:
Hello,
just a beginner and doing some research into the stack. I am quite curious to know if the discovery services have been implemented in the stack.
Also, would it be possible to use the LDS provided by the OPC foundation and use the RegisterServer  as specified in part 12 of the foundation. 
Also, GDS is not being implemented yet right?

Regards,
Rakshan

kevinherron, 2020-11-20:
> I am quite curious to know if the discovery services have been implemented in the stack.

Discovery services needed to run a standard server are implemented: https://github.com/eclipse/milo/blob/6e10014c4fb2c7aae6ddcc7eea3f35ff4400d6e6/opc-ua-stack/stack-server/src/main/java/org/eclipse/milo/opcua/stack/server/UaStackServer.java#L530-L707

Discover services as would be needed by an LDS or GDS implementation are not. I think any LDS or GDS implementation should be done as a standalone project, not part of Milo itself.

> Also, would it be possible to use the LDS provided by the OPC foundation and use the RegisterServer as specified in part 12 of the foundation.

Yes, that should work fine. The Milo demo server can register itself with an LDS: https://github.com/digitalpetri/opc-ua-demo-server/blob/53964a54298429f573af73274eb690a51c61aad3/src/main/kotlin/com/digitalpetri/opcua/server/DemoServer.kt#L266-L312

RakshanPremsagar, 2020-11-23:
Ah sorry, I see that there are implementations for getEnpoints and FindServers on the UaStackServer.java file. I also see, that you have added FindServersOnNetwork(with no definition). How about RegisterServer and RegisterServer2? Are they yet to be done?
I saw an ongoing conversation where it said, that the discovery services have been done here? Sorry if i missed something here?

kevinherron, 2020-11-27:
I’m saying these implementations don’t belong as part of this project. They are only required by an LDS or GDS server and Milo is not aiming to provide implementations of those right now.

closed 2020-12-04 by kevinherron