eclipse-milo/milo #732: Errors when building the project

noble-man, 2020-11-28:
Hello dear contributors,
I cloned the milo project and executed "mvn clean install" at the root, but it doesn't build.
Could you help me understand the error?
(see maven output attached)
[image]
[image]
[image]
[image]

kevinherron, 2020-11-30:
It seems the tests are taking extraordinarily long to run your computer for some reason.

Can you build successfully if you skip tests?

Try`mvn clean install -DskipTests`

closed 2020-12-04 by kevinherron

noble-man, 2020-12-15:
Hi Kevin,
I changed the version in the _pom.xml_ file to _0.5.3_ and it worked.
Thanks