eclipse-milo/milo #733: Eclipse Milo: Reading and decoding a custom data type node from a Simatic S7-1500

DickonForKing, 2020-12-01:
**The problem and what I suspect goes wrong:**
I am trying to read a custom data type node from a Siemens Simatic S7-1500 and decoding fails with:
```
Error running client example: max string length exceeded (length=1819042152, max=2097152)
org.eclipse.milo.opcua.stack.core.UaSerializationException: max string length exceeded (length=1819042152, max=2097152)
```

**What I did so far:**
First of all, I've read and tried out the ReadWriteCustomDataTypeNodeExample from the client examples module. I've basically copied this example and replaced CustomStructType for a new custom type called Status containing 10 booleans. Like in the CustomStructType class, I've added a static codec to Status that reads the boolean values from the ByteString by using decoder.readBoolean(). This works perfectly. I have managed to read nodes, modify them and write them back to the OPC UA server. So far so good.

```
    NodeId binaryEncodingId = new NodeId(3, "TE_\"DB_FUNCTION_STATUS\".\"TEST\"");
    // Register codec with the client DataTypeManager instance
    client.getDataTypeManager().registerCodec(
            binaryEncodingId,
            new TestNode.Codec().asBinaryCodec()
    );

    // synchronous read request via VariableNode
    NodeId nodeId = new NodeId(3, "\"DB_FUNCTION_STATUS\".\"TEST\"");
    UaVariableNode node = client.getAddressSpace().getVariableNode(nodeId);
    DataValue value = node.readValue();

    logger.info("====== Reading value from OPC UA Server =======");
    logger.info("Value={}", value);

    Variant variant = value.getValue();
    ExtensionObject xo = (ExtensionObject) variant.getValue();

    TestNode decoded = (TestNode) xo.decode(
            client.getSerializationContext()
    );
    logger.info("Decoded={}", decoded);
```

Next up, I want to do the same as above, but with another custom type TestNode containing a String and an integer. I created the class with a codec that uses decoder.readString() and decoder.readInt16(). When I try to read and decode the ByteString for this node, two issues appear:
- readInt16() reads a different value than the value I see when I read this node with UaExpert.
- readString() throws the exception stated above.

At this moment, I started doing research and found the following:
- I assume readString() uses the OPC UA String type node ns0;i=12, while, according to https://support.industry.siemens.com/cs/document/109780313/under-which-node-ids-do-you-find-the-type-descriptions-of-the-simatic-data-types-in-the-opc-ua-server-of-a-simatic-s7-1200-s7-1500-?dti=0&lc=en-GB, the Siemens OPC UA server uses ns=3;i=3014. This is probably where it goes wrong, but I'm not sure.
- Also, I don't understand why readInt16() reads a different value than the actual value. I have added a screenshot that shows the node in UaExpert with the value and data types. 

[image]

What are my options in this situation?

kevinherron, 2020-12-01:
Typically when connecting to a server like this you don't try to hand-write custom codecs.

Instead, you'd do what the `UnifiedAutomationReadCustomDataTypeExample` does: add a `DataTypeDictionarySessionInitializer` so the datatype dictionary is read when the client connects. Optionally you can use a `JsonBsdParser` instead of `GenericBsdParser` if you'd rather deal with JSON than the generic `Struct` object.

As for why your custom codec isn't working - I'm not sure. I'd probably need to see your codec as well as the actual structure definition from the datatype dictionary in the server. It's possible the server is exposing this as a struct with optional fields and you're missing a hidden field or something.

DickonForKing, 2020-12-02:
Thank you for your quick response.

Before posting my issue I also tried the `UnifiedAutomationReadCustomDataTypeExample` however it wasn't working either. After reading your reply, I tried it again, but I enabled the debug logging. I found out that things went wrong with parsing the OPC UA Dictionary in xml, because of JAXB errors. This in turn was related to compiling the project in Java 11 instead of Java 1.8 and after changing the JDK it works now.

I will probably need JDK 11 though, so I'm going to try to solve this. Have you tried this already?

kevinherron, 2020-12-02:
There shouldn't be any issue compiling or running with JDK 11. The artifacts you get from Maven are compiled with JDK 8, but we run everything on JDK 11 at work.

kevinherron, 2020-12-02:
Are you running inside an application server that may be interfering like [in this issue](https://github.com/eclipse/milo/issues/711)?

DickonForKing, 2020-12-03:
I tried again compiling and running the application with JDK 11 and it still seems to get an exception related to the JAXB-API.

```
08:32:03.176 [ForkJoinPool.commonPool-worker-3] WARN  o.e.m.o.s.c.DataTypeDictionaryReader - Failed to create DataTypeDictionary nodeId=NodeId{ns=2, id=6435}
java.util.concurrent.CompletionException: javax.xml.bind.JAXBException: Implementation of JAXB-API has not been found on module path or classpath.
 - with linked exception:
[java.lang.ClassNotFoundException: com.sun.xml.bind.v2.ContextFactory]
	at java.base/java.util.concurrent.CompletableFuture.encodeRelay(CompletableFuture.java:367)
	at java.base/java.util.concurrent.CompletableFuture.completeRelay(CompletableFuture.java:376)
	at java.base/java.util.concurrent.CompletableFuture$UniCompose.tryFire(CompletableFuture.java:1074)
	at java.base/java.util.concurrent.CompletableFuture.postComplete(CompletableFuture.java:506)
	at java.base/java.util.concurrent.CompletableFuture.postFire(CompletableFuture.java:610)
	at java.base/java.util.concurrent.CompletableFuture$UniCompose.tryFire(CompletableFuture.java:1085)
	at java.base/java.util.concurrent.CompletableFuture$Completion.exec(CompletableFuture.java:479)
	at java.base/java.util.concurrent.ForkJoinTask.doExec(ForkJoinTask.java:290)
	at java.base/java.util.concurrent.ForkJoinPool$WorkQueue.topLevelExec(ForkJoinPool.java:1020)
	at java.base/java.util.concurrent.ForkJoinPool.scan(ForkJoinPool.java:1656)
	at java.base/java.util.concurrent.ForkJoinPool.runWorker(ForkJoinPool.java:1594)
	at java.base/java.util.concurrent.ForkJoinWorkerThread.run(ForkJoinWorkerThread.java:183)
Caused by: javax.xml.bind.JAXBException: Implementation of JAXB-API has not been found on module path or classpath.
	at javax.xml.bind.ContextFinder.newInstance(ContextFinder.java:232)
	at javax.xml.bind.ContextFinder.find(ContextFinder.java:375)
	at javax.xml.bind.JAXBContext.newInstance(JAXBContext.java:691)
	at javax.xml.bind.JAXBContext.newInstance(JAXBContext.java:632)
	at org.eclipse.milo.opcua.binaryschema.parser.BsdParser.parse(BsdParser.java:40)
	at org.eclipse.milo.opcua.sdk.client.DataTypeDictionaryReader.createDataTypeDictionary(DataTypeDictionaryReader.java:251)
	at org.eclipse.milo.opcua.sdk.client.DataTypeDictionaryReader.lambda$readDataTypeDictionary$6(DataTypeDictionaryReader.java:134)
	at java.base/java.util.concurrent.CompletableFuture$UniCompose.tryFire(CompletableFuture.java:1072)
	... 9 common frames omitted
Caused by: java.lang.ClassNotFoundException: com.sun.xml.bind.v2.ContextFactory
	at java.base/jdk.internal.loader.BuiltinClassLoader.loadClass(BuiltinClassLoader.java:581)
	at java.base/jdk.internal.loader.ClassLoaders$AppClassLoader.loadClass(ClassLoaders.java:178)
	at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:522)
	at javax.xml.bind.ServiceLoaderUtil.nullSafeLoadClass(ServiceLoaderUtil.java:92)
	at javax.xml.bind.ServiceLoaderUtil.safeLoadClass(ServiceLoaderUtil.java:125)
	at javax.xml.bind.ContextFinder.newInstance(ContextFinder.java:230)
	... 16 common frames omitted
```

However, I have seen this issue before when it comes to switching from JDK8 to JDK11, because Java EE related technologies, which contained JAXB, were removed from JDK9 onwards, I believe. I used `mvn clean install -U`, so that I force to update the artifact from Maven, in case I had jar files from a local compiled version with JDK11. Or is there something else I can try?

Is it possible that the JAXB dependencies have been manually added to your work project and that it works for that reason?

Regarding your second comment about an application server, that's not the case here. I'm running a stand alone application with no code in there yet aside from Milo.

kevinherron, 2020-12-03:
I can run `UnifiedAutomationReadCustomDataTypeExample` with JDK 11 with no issues.

The `bsd-core` module should have the required dependencies:
```
[INFO] +- com.sun.activation:jakarta.activation:jar:1.2.1:compile
[INFO] \- org.glassfish.jaxb:jaxb-runtime:jar:2.3.2:compile
[INFO]    +- jakarta.xml.bind:jakarta.xml.bind-api:jar:2.3.2:compile
[INFO]    +- org.glassfish.jaxb:txw2:jar:2.3.2:compile
[INFO]    +- com.sun.istack:istack-commons-runtime:jar:3.0.8:compile
[INFO]    +- org.jvnet.staxex:stax-ex:jar:1.8.1:compile
[INFO]    +- com.sun.xml.fastinfoset:FastInfoset:jar:1.2.16:compile
[INFO]    \- jakarta.activation:jakarta.activation-api:jar:1.2.1:compile
```

closed 2020-12-09 by kevinherron

DickonForKing, 2020-12-10:
Strange, I'm still getting exceptions when trying it with JDK 11. 
I created a Maven project with the following dependencies:

```
  <dependencies>
    <dependency>
      <groupId>org.eclipse.milo</groupId>
      <artifactId>sdk-client</artifactId>
      <version>0.5.2</version>
    </dependency>
    <dependency>
      <groupId>org.eclipse.milo</groupId>
      <artifactId>dictionary-reader</artifactId>
      <version>0.5.2</version>
    </dependency>
    <dependency>
      <groupId>org.eclipse.milo</groupId>
      <artifactId>server-examples</artifactId>
      <version>0.5.2</version>
    </dependency>
    <dependency>
      <groupId>ch.qos.logback</groupId>
      <artifactId>logback-classic</artifactId>
      <version>1.1.7</version>
      <exclusions>
        <exclusion>
          <groupId>org.slf4j</groupId>
          <artifactId>slf4j-api</artifactId>
        </exclusion>
      </exclusions>
    </dependency>
  </dependencies>
```

The `sdk-client` should contain the library in this case. Also according to Intellij the libraries are actually present in the External Libraries folder. Any idea on how to fix this?

kevinherron, 2020-12-10:
https://github.com/kevinherron/milo-jdk11-test

```
kevin@kevins-mbp ~/D/T/milo-jdk11-test> java -version
openjdk version "11.0.8" 2020-07-14
OpenJDK Runtime Environment AdoptOpenJDK (build 11.0.8+10)
OpenJDK 64-Bit Server VM AdoptOpenJDK (build 11.0.8+10, mixed mode)
```

```
kevin@kevins-mbp ~/D/T/milo-jdk11-test (master)> mvn clean package exec:exec
[INFO] Scanning for projects...
[INFO] 
[INFO] ------------------< com.kevinherron:milo-jdk11-test >-------------------
[INFO] Building milo-jdk11-test 1.0-SNAPSHOT
[INFO] --------------------------------[ jar ]---------------------------------
[INFO] 
[INFO] --- maven-clean-plugin:2.5:clean (default-clean) @ milo-jdk11-test ---
[INFO] Deleting /Users/kevin/Development/ThirdParty/milo-jdk11-test/target
[INFO] 
[INFO] --- maven-resources-plugin:2.6:resources (default-resources) @ milo-jdk11-test ---
[WARNING] Using platform encoding (UTF-8 actually) to copy filtered resources, i.e. build is platform dependent!
[INFO] Copying 0 resource
[INFO] 
[INFO] --- maven-compiler-plugin:3.1:compile (default-compile) @ milo-jdk11-test ---
[INFO] Changes detected - recompiling the module!
[WARNING] File encoding has not been set, using platform encoding UTF-8, i.e. build is platform dependent!
[INFO] Compiling 1 source file to /Users/kevin/Development/ThirdParty/milo-jdk11-test/target/classes
[INFO] 
[INFO] --- maven-resources-plugin:2.6:testResources (default-testResources) @ milo-jdk11-test ---
[WARNING] Using platform encoding (UTF-8 actually) to copy filtered resources, i.e. build is platform dependent!
[INFO] skip non existing resourceDirectory /Users/kevin/Development/ThirdParty/milo-jdk11-test/src/test/resources
[INFO] 
[INFO] --- maven-compiler-plugin:3.1:testCompile (default-testCompile) @ milo-jdk11-test ---
[INFO] Nothing to compile - all classes are up to date
[INFO] 
[INFO] --- maven-surefire-plugin:2.12.4:test (default-test) @ milo-jdk11-test ---
[INFO] No tests to run.
[INFO] 
[INFO] --- maven-jar-plugin:2.4:jar (default-jar) @ milo-jdk11-test ---
[INFO] Building jar: /Users/kevin/Development/ThirdParty/milo-jdk11-test/target/milo-jdk11-test-1.0-SNAPSHOT.jar
[INFO] 
[INFO] --- exec-maven-plugin:3.0.0:exec (default-cli) @ milo-jdk11-test ---
[main] INFO org.eclipse.milo.opcua.sdk.client.OpcUaClient - Eclipse Milo OPC UA Stack version: 0.5.3
[main] INFO org.eclipse.milo.opcua.sdk.client.OpcUaClient - Eclipse Milo OPC UA Client SDK version: 0.5.3
[NonceUtilSecureRandom] INFO org.eclipse.milo.opcua.stack.core.util.NonceUtil - SecureRandom seeded in 1ms.
[main] INFO com.kevinherron.UnifiedAutomationExample - value: Struct{name=WorkOrderType, members={ID=Member{name=ID, value=1aea372c-3f0c-45a7-a2ad-1bf4783b797b}, AssetID=Member{name=AssetID, value=123-X-Y-Z}, StartTime=Member{name=StartTime, value=DateTime{utcTime=132520782218128349, javaDate=Thu Dec 10 04:50:21 PST 2020}}, StatusComments=Member{name=StatusComments, value=[Struct{name=WorkOrderStatusType, members={Actor=Member{name=Actor, value=Wendy Terry}, Timestamp=Member{name=Timestamp, value=DateTime{utcTime=132520782218128349, javaDate=Thu Dec 10 04:50:21 PST 2020}}, Comment=Member{name=Comment, value=LocalizedText{text=Mission accomplished!, locale=en-US}}}}, Struct{name=WorkOrderStatusType, members={Actor=Member{name=Actor, value=Gavin Mackenzie}, Timestamp=Member{name=Timestamp, value=DateTime{utcTime=132520782218128349, javaDate=Thu Dec 10 04:50:21 PST 2020}}, Comment=Member{name=Comment, value=LocalizedText{text=I think clients would love this., locale=en-US}}}}, Struct{name=WorkOrderStatusType, members={Actor=Member{name=Actor, value=Phil Taylor}, Timestamp=Member{name=Timestamp, value=DateTime{utcTime=132520782218128349, javaDate=Thu Dec 10 04:50:21 PST 2020}}, Comment=Member{name=Comment, value=LocalizedText{text=And justice for all., locale=en-US}}}}]}}}
[INFO] ------------------------------------------------------------------------
[INFO] BUILD SUCCESS
[INFO] ------------------------------------------------------------------------
[INFO] Total time:  4.607 s
[INFO] Finished at: 2020-12-10T04:58:15-08:00
[INFO] ------------------------------------------------------------------------

```

DickonForKing, 2020-12-10:
Thank you for the test. 
I will dive deeper into what goes wrong in my application, thanks again for the help.

kevinherron, 2020-12-10:
Keep me updated if possible, I'm curious to know what you find.

adrianhoehne, 2021-08-12:
Hi, I discovered the same issue with the wrong values and the exception. 
It was because, the order was wrong of String and Integer in Codec.class decode() which I had to implement.
I changed the order, which solved the error.