eclipse-milo/milo #734: Upgrade from milo  stack-client version 0.1.2 to 0.5.4

joao-parana, 2020-12-08:
**Describe the bug**
When I try upgrade from version 0.1.2 to version 0.5.4 I get the compilation error: `cannot find symbol class UaTcpStackClient  location: package org.eclipse.milo.opcua.stack.client`

**Expected behavior**
I expect to compile without errors.

**Logs and Packet Captures**
Maven compilation error: `cannot find symbol class UaTcpStackClient  location: package org.eclipse.milo.opcua.stack.client`

**Additional context**
I'm trying upgrade Milo version from `0.1.2` which works for me.

kevinherron, 2020-12-08:
There have been a number of API changes since 0.1. That class is now `org.eclipse.milo.opcua.stack.client.UaStackClient`.

You will have to make updates to your code to accommodate the API changes.

joao-parana, 2020-12-08:
The `0.2.5` version works for me.

```xml
		<dependency>
			<groupId>org.eclipse.milo</groupId>
			<artifactId>sdk-client</artifactId>
			<version>0.2.5</version>
		</dependency>
		<dependency>
			<groupId>org.eclipse.milo</groupId>
			<artifactId>stack-client</artifactId>
			<version>0.2.5</version>
		</dependency>
```

@kevinherron is there a post, wiki or any kind of documentation about this change ? I appreciate any information.

kevinherron, 2020-12-08:
I'm sorry but there is not. 

There is now some documentation about using the client [in the wiki](https://github.com/eclipse/milo/wiki/Client) and there the [client examples](https://github.com/eclipse/milo/tree/master/milo-examples/client-examples/src/main/java/org/eclipse/milo/examples/client) that may server as an additional reference, but that's about it. 

You're welcome to ask questions as you run into issues updating your code here or preferably on the mailing list.

closed 2020-12-09 by kevinherron