eclipse-milo/milo #735: CEFACT Engineering Units Missing Display Name

sarthak-k, 2020-12-10:
**Describe the bug**
Few measurements units of CEFACT are missing display name.

**Expected behavior**
Correct Display Name for the measurement units.

For example,

```java
    public static final EUInformation StandardCubicMetrePerDay = new EUInformation(CEFACT_NAMESPACE_URI, 5321527, english(""), english("Standard cubic metre per day"));
    public static final EUInformation StandardCubicMetrePerHour = new EUInformation(CEFACT_NAMESPACE_URI, 5321528, english(""), english("Standard cubic metre per hour"));
    public static final EUInformation NormalizedCubicMetrePerDay = new EUInformation(CEFACT_NAMESPACE_URI, 5321529, english(""), english("Normalized cubic metre per day"));
    public static final EUInformation NormalizedCubicMetrePerHour = new EUInformation(CEFACT_NAMESPACE_URI, 5321776, english(""), english("Normalized cubic metre per hour"));
    public static final EUInformation KilowattHourPerNormalizedCubicMetre = new EUInformation(CEFACT_NAMESPACE_URI, 4937550, english(""), english("Kilowatt hour per normalized cubic metre"));
    public static final EUInformation KilowattHourPerStandardCubicMetre = new EUInformation(CEFACT_NAMESPACE_URI, 4937555, english(""), english("Kilowatt hour per standard cubic metre"));
    public static final EUInformation JoulePerNormalisedCubicMetre = new EUInformation(CEFACT_NAMESPACE_URI, 5321777, english(""), english("Joule per normalised cubic metre"));
    public static final EUInformation JoulePerStandardCubicMetre = new EUInformation(CEFACT_NAMESPACE_URI, 5321778, english(""), english("Joule per standard cubic metre"));
```

kevinherron, 2020-12-10:
Indeed. Going back to the original contribution it looks like they may have always been missing for some fields.

I'll have to whip up a quick code generator and generate a new one.

kevinherron, 2020-12-10:
I think this may have been generated from https://www.unece.org/fileadmin/DAM/cefact/recommendations/rec20/rec20.zip when it should probably get generated from https://opcfoundation.org/UA/EngineeringUnits/UNECE/UNECE_to_OPCUA.csv.

This means, though, that some of the field names are likely going to change. And they are going to be a PITA to sanitize.

sarthak-k, 2020-12-10:
I don't think so it's generated from https://www.unece.org/fileadmin/DAM/cefact/recommendations/rec20/rec20.zip as there are Unit Id mentioned anywhere in their data while https://opcfoundation.org/UA/EngineeringUnits/UNECE/UNECE_to_OPCUA.csv mentions `UnitId` explicitly. Though, the same CSV doesn't contain many of units which I mentioned above. Something to investigate.

kevinherron, 2020-12-10:
Maybe some combination, because of the of names it uses do not appear in the OPC UA CSV file but do appear in the file from unece.org.

kevinherron, 2020-12-10:
For example: "Standard cubic metre per day" is something that only appears in the UNECE Excel file, but is not part of the CSV, and indeed they have empty Symbol columns, which may be why they are empty in the generated code.

[image]

kevinherron, 2020-12-17:
Unless someone else is interested in trying to fix this without breaking the API I'm going to wait until 0.6.0 to generate new code, at which point I plan to derive new names for the static fields based on the value in the `UNECECode` column from the CSV mapping. I seriously doubt more than 1 or 2 people are even using this.

kevinherron, 2021-03-05:
@sarthak-k I don't know if you still care about this but take a look at #780 if you do.

As I mentioned, it's not the same set of static fields as before, but given that information the previous file was generated from is only partial, generating new code from the CSV file is as much work as I'm willing to put into this.

PR #780 (2021-03-09): Generate new CefactEngineeringUnits from UNECE_to_OPCUA.csv
https://opcfoundation.org/UA/EngineeringUnits/UNECE/UNECE_to_OPCUA.csv

fixes #735
commit f07af5a3 (2021-03-13): Generate new CefactEngineeringUnits from UNECE_to_OPCUA.csv (#780)

https://opcfoundation.org/UA/EngineeringUnits/UNECE/UNECE_to_OPCUA.csv

fixes #735
closed 2021-03-09 by kevinherron