eclipse-milo/milo #737: Unable to read OPCUA tags using browse Node Example

bigdatasunil, 2020-12-16:
**Describe the bug**
When ever  i run the BrowseExample.java class from  client-examples package to fetch all tags, I see that some of the tags are missing or not browsing, especially the tags is starting with Double Quotes ("").

**Example: "Channel"."Sensor1"** 

These Kind of tags unable to fetch by BrowseExample.java sample code. 
Having said that the same code works very well with other type OPCUA server which does not have any tags with Double Quotes. 
**Example: /Channel/Sensor2**

But I tried to run BrowseNodeExample. java, Which is actually fetching all tags including Double Quotes tags  but with lot of errors and Warnings. 
Below is some of logs. 
I have analysed the code see that there is  different logic.

Could you please help me to fix this problem? 

In my case i have to write logic to retrieve any type of tags  could be Slash or Quotes and store it some database.

**Question-2**: 
How can specify the End tag or Start and End Folder while browsing? Is there any possibility ??

Kindly let me know if you need more details on this. 

**I am using 0.5.3 version of Milo code.** 

**Expected behavior**
Code should be able to read any type tags irrespective of special characters. 

**Logs and Packet Captures**
 c..fetchtags.client.BrowseExample  : , node.getNodeId().....3
 [nio-7070-exec-9] c..fetchtags.client.BrowseExample  : getIdentifier..."DB_OPC_SCANIA_Messages"
2020-12-05 16:48:24.084  WARN 45456 --- [d-thread-pool-1] o.e.milo.opcua.sdk.client.AddressSpace   : Failed to create Node from Reference to ExpandedNodeId{ns=3, id="DB_OPC_SCANIA_Messages"."Message_26", serverIndex=0}

java.util.concurrent.CompletionException: UaException: status=Bad_Timeout, message=requestId=136 timed out after 5000ms
        at java.util.concurrent.CompletableFuture.encodeRelay(CompletableFuture.java:326) ~[na:1.8.0_201]
        at java.util.concurrent.CompletableFuture.completeRelay(CompletableFuture.java:338) ~[na:1.8.0_201]
        at java.util.concurrent.CompletableFuture.uniRelay(CompletableFuture.java:911) ~[na:1.8.0_201]
        at java.util.concurrent.CompletableFuture$UniRelay.tryFire(CompletableFuture.java:899) ~[na:1.8.0_201]
        at java.util.concurrent.CompletableFuture.postComplete(CompletableFuture.java:474) ~[na:1.8.0_201]
        at java.util.concurrent.CompletableFuture.completeExceptionally(CompletableFuture.java:1977) ~[na:1.8.0_201]
        at org.eclipse.milo.opcua.stack.client.transport.AbstractTransport.lambda$scheduleRequestTimeout$7(AbstractTransport.java:143) ~[stack-client-0.5.3.jar!/:0.5.3]
        at io.netty.util.HashedWheelTimer$HashedWheelTimeout.expire(HashedWheelTimer.java:672) ~[netty-common-4.1.52.Final.jar!/:4.1.52.Final]
        at io.netty.util.HashedWheelTimer$HashedWheelBucket.expireTimeouts(HashedWheelTimer.java:747) ~[netty-common-4.1.52.Final.jar!/:4.1.52.Final]
        at io.netty.util.HashedWheelTimer$Worker.run(HashedWheelTimer.java:472) ~[netty-common-4.1.52.Final.jar!/:4.1.52.Final]
        at java.lang.Thread.run(Thread.java:748) ~[na:1.8.0_201]
Caused by: org.eclipse.milo.opcua.stack.core.UaException: requestId=136 timed out after 5000ms
        at org.eclipse.milo.opcua.stack.client.transport.AbstractTransport.lambda$scheduleRequestTimeout$7(AbstractTransport.java:138) ~[stack-client-0.5.3.jar!/:0.5.3]
        ... 4 common frames omitted

2020-12-05 16:48:24.086  WARN 45456 --- [d-thread-pool-1] o.e.milo.opcua.sdk.client.AddressSpace   : Failed to create Node from Reference to ExpandedNodeId{ns=3, id="DB_OPC_SCANIA_Messages"."Message_27", serverIndex=0}

**Warning Logs:** 

   ... 1 common frames omitted

2020-12-05 16:48:38.522  WARN 45456 --- [d-thread-pool-1] o.e.m.o.s.c.t.u.UascClientMessageHandler : No pending request with requestId=382 for ReadResponse
2020-12-05 16:48:38.579  WARN 45456 --- [d-thread-pool-1] o.e.m.o.s.c.t.u.UascClientMessageHandler : No pending request with requestId=383 for ReadResponse
2020-12-05 16:48:38.647  WARN 45456 --- [d-thread-pool-1] o.e.m.o.s.c.t.u.UascClientMessageHandler : No pending request with requestId=372 for ReadResponse
2020-12-05 16:48:38.699  WARN 45456 --- [d-thread-pool-1] o.e.m.o.s.c.t.u.UascClientMessageHandler : No pending request with requestId=385 for ReadResponse
2020-12-05 16:48:38.763  WARN 45456 --- [d-thread-pool-1] o.e.m.o.s.c.t.u.UascClientMessageHandler : No pending request with requestId=386 for ReadResponse
2020-12-05 16:48:38.830  WARN 45456 --- [d-thread-pool-1] o.e.m.o.s.c.t.u.UascClientMessageHandler : No pending request with requestId=375 for ReadResponse
2020-12-05 16:48:38.883  WARN 45456 --- [d-thread-pool-1] o.e.m.o.s.c.t.u.UascClientMessageHandler : No pending request with requestId=388 for ReadResponse
2020-12-05 16:48:38.941  WARN 45456 --- [d-thread-pool-1] o.e.m.o.s.c.t.u.UascClientMessageHandler : No pending request with requestId=389 for ReadResponse
2020-12-05 16:48:39.050  WARN 45456 --- [d-thread-pool-1] o.e.m.o.s.c.t.u.UascClientMessageHandler : No pending request with requestId=391 for ReadResponse
2020-12-05 16:48:39.096  WARN 45456 --- [d-thread-pool-1] o.e.m.o.s.c.t.u.UascClientMessageHandler : No pending request with requestId=378 for ReadResponse
2020-12-05 16:48:39.156  WARN 45456 --- [d-thread-pool-1] o.e.m.o.s.c.t.u.UascClientMessageHandler : No pending request with requestId=392 for ReadResponse
2020-12-05 16:48:39.225  WARN 45456 --- [d-thread-pool-1] o.e.m.o.s.c.t.u.UascClientMessageHandler : No pending request with requestId=394 for ReadResponse
2020-12-05 16:48:39.274  WARN 45456 --- [d-thread-pool-1] o.e.m.o.s.c.t.u.UascClientMessageHandler : No pending request with requestId=381 for ReadResponse
2020-12-05 16:48:39.333  WARN 45456 --- [d-thread-pool-1] o.e.m.o.s.c.t.u.UascClientMessageHandler : No pending request with requestId=395 for ReadResponse
2020-12-05 16:48:39.403  WARN 45456 --- [d-thread-pool-1] o.e.m.o.s.c.t.u.UascClientMessageHandler : No pending request with requestId=397 for ReadResponse
2020-12-05 16:48:39.454  WARN 45456 --- [d-thread-pool-1] o.e.m.o.s.c.t.u.UascClientMessageHandler : No pending request with requestId=384 for ReadResponse
[... 12 lines cut ...]
2020-12-05 16:48:40.232  WARN 45456 --- [d-thread-pool-1] o.e.m.o.s.c.t.u.UascClientMessageHandler : No pending request with requestId=410 for ReadResponse
2020-12-05 16:48:40.345  WARN 45456 --- [d-thread-pool-1] o.e.m.o.s.c.t.u.UascClientMessageHandler : No pending request with requestId=412 for ReadResponse
2020-12-05 16:48:40.395  WARN 45456 --- [d-thread-pool-1] o.e.m.o.s.c.t.u.UascClientMessageHandler : No pending request with requestId=399 for ReadResponse
2020-12-05 16:48:40.454  WARN 45456 --- [d-thread-pool-1] o.e.m.o.s.c.t.u.UascClientMessageHandler : No pending request with requestId=413 for ReadResponse
2020-12-05 16:48:40.525  WARN 45456 --- [d-thread-pool-1] o.e.m.o.s.c.t.u.UascClientMessageHandler : No pending request with requestId=415 for ReadResponse
2020-12-05 16:48:40.579  WARN 45456 --- [d-thread-pool-1] o.e.m.o.s.c.t.u.UascClientMessageHandler : No pending request with requestId=402 for ReadResponse
2020-12-05 16:48:40.632  WARN 45456 --- [d-thread-pool-1] o.e.m.o.s.c.t.u.UascClientMessageHandler : No pending request with requestId=416 for ReadResponse
2020-12-05 16:48:40.706  WARN 45456 --- [d-thread-pool-1] o.e.m.o.s.c.t.u.UascClientMessageHandler : No pending request with requestId=405 for ReadResponse
2020-12-05 16:48:40.759  WARN 45456 --- [d-thread-pool-1] o.e.m.o.s.c.t.u.UascClientMessageHandler : No pending request with requestId=418 for ReadResponse
2020-12-05 16:48:40.816  WARN 45456 --- [d-thread-pool-1] o.e.m.o.s.c.t.u.UascClientMessageHandler : No pending request with requestId=419 for ReadResponse
2020-12-05 16:48:40.881  WARN 45456 --- [d-thread-pool-1] o.e.m.o.s.c.t.u.UascClientMessageHandler : No pending request with requestId=408 for ReadResponse
2020-12-05 16:48:40.931  WARN 45456 --- [d-thread-pool-1] o.e.m.o.s.c.t.u.UascClientMessageHandler : No pending request with requestId=421 for ReadResponse
2020-12-05 16:48:40.990  WARN 45456 --- [d-thread-pool-1] o.e.m.o.s.c.t.u.UascClientMessageHandler : No pending request with requestId=422 for ReadResponse
2020-12-05 16:48:41.051  WARN 45456 --- [d-thread-pool-1] o.e.m.o.s.c.t.u.UascClientMessageHandler : No pending request with requestId=411 for ReadResponse
2020-12-05 16:48:41.103  WARN 45456 --- [d-thread-pool-1] o.e.m.o.s.c.t.u.UascClientMessageHandler : No pending request with requestId=424 for ReadResponse
[... cut ...]
2020-12-05 16:48:41.498  WARN 45456 --- [d-thread-pool-1] o.e.m.o.s.c.t.u.UascClientMessageHandler : No pending request with requestId=417 for ReadResponse
2020-12-05 16:48:41.559  WARN 45456 --- [d-thread-pool-1] o.e.m.o.s.c.t.u.UascClientMessageHandler : No pending request with requestId=431 for ReadResponse
2020-12-05 16:48:41.632  WARN 45456 --- [d-thread-pool-1] o.e.m.o.s.c.t.u.UascClientM
**Additional context**
Version of Eclipse Milo you are using, OPC UA client or software server you are interacting with, etc...

kevinherron, 2020-12-16:
The server you are running BrowseNodeExample against seems to be responding slow and the requests are timing out.

Try modifying this line and increasing the timeout to 60000 or more: https://github.com/eclipse/milo/blob/05ae1706d7b039c7bd64d3e7b946e6325c9b9349/milo-examples/client-examples/src/main/java/org/eclipse/milo/examples/client/ClientExampleRunner.java#L86

bigdatasunil, 2020-12-17:
Does not help much. same issue. :-(

kevinherron, 2020-12-17:
Do you know how to apply a patch? Can you try this change and see if it helps?

```
Index: opc-ua-sdk/sdk-client/src/main/java/org/eclipse/milo/opcua/sdk/client/AddressSpace.java
IDEA additional info:
Subsystem: com.intellij.openapi.diff.impl.patch.CharsetEP
<+>UTF-8
===================================================================
diff --git a/opc-ua-sdk/sdk-client/src/main/java/org/eclipse/milo/opcua/sdk/client/AddressSpace.java b/opc-ua-sdk/sdk-client/src/main/java/org/eclipse/milo/opcua/sdk/client/AddressSpace.java
--- a/opc-ua-sdk/sdk-client/src/main/java/org/eclipse/milo/opcua/sdk/client/AddressSpace.java	
+++ b/opc-ua-sdk/sdk-client/src/main/java/org/eclipse/milo/opcua/sdk/client/AddressSpace.java	
@@ -12,6 +12,7 @@
 
 import java.util.ArrayList;
 import java.util.Collections;
+import java.util.LinkedList;
 import java.util.List;
 import java.util.Objects;
 import java.util.Optional;
@@ -605,71 +606,73 @@
 
         CompletableFuture<List<ReferenceDescription>> browse = BrowseHelper.browse(client, browseDescription);
 
-        return browse.thenCompose(references -> {
-            List<CompletableFuture<? extends UaNode>> cfs = references.stream()
-                .map(reference -> {
-                    NodeClass nodeClass = reference.getNodeClass();
-                    ExpandedNodeId xNodeId = reference.getNodeId();
-                    ExpandedNodeId xTypeDefinitionId = reference.getTypeDefinition();
+        return browse.thenCompose(this::createNodesFromReferences);
+    }
+
+    private CompletableFuture<List<? extends UaNode>> createNodesFromReferences(
+        List<ReferenceDescription> references
+    ) {
+
+        return createNodesFromReferences(
+            Collections.synchronizedList(new LinkedList<>(references)),
+            Collections.synchronizedList(new ArrayList<>())
+        );
+    }
+
+    private CompletableFuture<List<? extends UaNode>> createNodesFromReferences(
+        List<ReferenceDescription> references,
+        List<UaNode> nodes
+    ) {
+
+        if (references.isEmpty()) {
+            return CompletableFuture.completedFuture(nodes);
+        } else {
+            ReferenceDescription reference = references.remove(0);
+
+            logger.info("Creating UaNode for {}...", reference.getNodeId());
+
+            return createNodeFromReference(reference).thenCompose(node -> {
+                logger.info("...Created UaNode for {}", reference.getNodeId());
+                nodes.add(node);
+
+                return createNodesFromReferences(references, nodes);
+            });
+        }
+    }
+
+    private CompletableFuture<? extends UaNode> createNodeFromReference(ReferenceDescription reference) {
+        NodeClass nodeClass = reference.getNodeClass();
+        ExpandedNodeId xNodeId = reference.getNodeId();
+        ExpandedNodeId xTypeDefinitionId = reference.getTypeDefinition();
 
-                    switch (nodeClass) {
-                        case Object:
[... 61 lines cut ...]
+                    logger.warn("Failed to create Node from Reference to {}", reference.getNodeId(), ex);
+                    return null;
+                });
+            }
         }
-
-        @SuppressWarnings("rawtypes")
-        CompletableFuture[] fa = cfs.toArray(new CompletableFuture[0]);
-
-        return CompletableFuture.allOf(fa).thenApply(v -> {
-            List<UaNode> results = new ArrayList<>(cfs.size());
-
-            for (CompletableFuture<? extends UaNode> cf : cfs) {
-                UaNode node = cf.join();
-                if (node != null) {
-                    results.add(node);
-                }
-            }
-
-            return results;
-        });
     }
 
     private static CompletableFuture<? extends UaNode> unwrap(
```

bigdatasunil, 2020-12-17:
I am unable to patch this, Any easy way to fix this? OR 
is it possible to provide me the Updated jar file ?

kevinherron, 2020-12-17:
You can try this branch instead: https://github.com/eclipse/milo/tree/create-nodes-serially

bigdatasunil, 2020-12-19:
Thanks for quick replies. 
I have generated the package "sdk-client-0.5.4-SNAPSHOT.jar" and replaced this with my older version. I still see the same issue. 
what else could be issue??

kevinherron, 2020-12-19:
I think you’ll need to get a Wireshark capture of the traffic between the client and server to diagnose further.

I suspect you are still not running the new code but we’ll see. The existing version has a step where it turns references returned by the browse service into UaNode instances and it does this in parallel, which seems to be overloading the server you are connected to.

bigdatasunil, 2020-12-20:
you are right. I m not running with new code. Now issues is fixed. I dont see any exceptions. thank you so much. 

Still i have a doubt, why broweExample.java is not working for my OCPUA server which has tags with "/channel/sensor1"

kevinherron, 2020-12-20:
It might depend on the ReferenceType used by the server to point to these Nodes, or it might be a bug in the server.

`BrowseExample` asks for all forward references of the base `Identifiers.Reference` type or its subtypes.

`BrowseNodesExample` asks for all forward references of the `Identifiers.HierarchicalReferences` type or its subtypes.

What `BrowseExample` asks for _should_ be a superset of what `BrowseNodesExample` asks for but it seems like the server you're connected to is responding incorrectly.

Try changing the `BrowseDescription` in `BrowseExample` to:
```java
        BrowseDescription browse = new BrowseDescription(
            browseRoot,
            BrowseDirection.Forward,
            Identifiers.HierarchicalReferences,
            true,
            uint(NodeClass.Object.getValue() | NodeClass.Variable.getValue()),
            uint(BrowseResultMask.All.getValue())
        );
```

does anything change?

The `nodeClassMask` parameter is also different - do you know what `NodeClass` the missing Nodes are?

closed 2021-01-11 by kevinherron