eclipse-milo/milo #738: Configuration of OpcUaServer scheduler

Ashiiro, 2020-12-17:
Hello,

Is it possible for you to make the scheduler in the configuration of the OpcUaServer instead of using the static final field retrieving the scheduler from Stack.

I've tried to figure out something using reflexion but it's not verry safe and it's still impossible to close every ressources that i allocate in a clean way if i open multiple servers.

It is important for me to insulate my different servers (thread wise).

(It is the same for any not configurable ressources taken from the Stack class).

Thank you for this library,
Ashii

kevinherron, 2020-12-17:
This sounds reasonable, I'll look into changing it.

Ashiiro, 2021-01-04:
Thank you verry much, do you know when the official 0.5.4 will be released ?

kevinherron, 2021-01-04:
I will do a release next week when I'm back from vacation. I found a couple other places in the codebase where the executor that gets used is not configurable so I will change those as well.

PR #739 (2021-01-11): Allow ScheduledExecutorService used by OpcUaServer to be configured
fixes #738
commit 05c276c2 (2021-01-11): Allow ScheduledExecutorService used by OpcUaServer to be configured (#739)

fixes #738
closed 2021-01-11 by kevinherron