eclipse-milo/milo #743: Bad_TooManySubscriptions

Ozzzzy, 2020-12-29:
Hello,

I encountered a problem. I created a subscription and subscribed to more than 300 nodes on the OPC UA server, but every time I start the program, the error Bad_TooManySubscriptions is returned. I want to know why this is my problem or What about the server problem? The same code was normal some time ago. Nowadays, it can only subscribe successfully occasionally, most of the time it returns to Bad_TooManySubscriptions.

Thank you for this library,
Ozz

kevinherron, 2020-12-29:
This is a response from the server. It’s common for the servers e.g. in PLCs to limit the number of sessions, subscriptions, and/or monitored items that are allowed to exist at once.

You might check if there are other clients connected, if you are cleanly disconnecting when you shut your program down and not leaving orphaned sessions and subscriptions, and if verify that you are only creating _ONE_ subscription and adding multiple monitored items to it, rather than creating a subscription per monitored item.

Ozzzzy, 2020-12-30:
Thank you ~ so how do I determine if there are legacy orphaned sessions and subscriptions?If I leave an isolated session and subscription, how do I close it completely?

kevinherron, 2020-12-30:
Calling `OpcUaClient::disconnect` will close your session and delete the subscriptions.

You can try connecting to the server with a client like UaExpert and use its diagnostics view to figure out if there are other sessions and subscriptions active, assuming the server supports diagnostics.

Ozzzzy, 2020-12-30:
Thank you. I'll have a try.

Ozzzzy, 2020-12-30:
I have one more question for you. What is the meaning and function of "future.complete(client);"?

kevinherron, 2020-12-30:
The “runner” that runs each example passes a `CompletableFuture` into the example being run that it should complete when it’s finished running so the runner knows to clean up. It’s not something that translates into real world usage, you can ignore it.

closed 2021-01-11 by kevinherron