eclipse-milo/milo #751: Just some theoretical clarification

RakshanPremsagar, 2021-01-19:
Hello,
Sorry to be asking this theoretical question here.. needed some clarity over discovery mechanism.
I have a LDS enabled with Multicast extension, running at 4840 port. I also have another server with mDNS enabled running at 4841. Also it is not registered to the LDS yet!
Now, running find Servers on Network, would result in details of both these servers mentioned above.

However, theoretically should it not be the case that the server at 4841 needs to register to the LDSME(port 4840) and only then the server(4841) is discovered?
Because the find Servers On network is always directed to the discovery endpoint right?

Regards and sincere apologies,
Rakshan

kevinherron, 2021-01-19:
I'm not an expert on the LDS-ME stuff, but it sounds like when you call FindServersOnNetwork against the LDS-ME server on 4840, it is sending a multicast probe and receiving a multicast announcement from the server on 4841, which has sufficient information for the server at 4841 to be returned in the response to FindServersOnNetwork.

[image]

RakshanPremsagar, 2021-01-19:
Yes thanks Kevin, your statement makes sense. 
But the next sentence of the image you posted explains scenario is seen only when the server uses the register server right? 
[image]

kevinherron, 2021-01-19:
Yes, it does sound like that. I'm not sure how to explain what you're seeing. Who wrote the LDS-ME implementations you are using?

RakshanPremsagar, 2021-01-19:
It is actually from the open62541 stack. There, generally nobody responds :/
(I am using the code from the examples section -> discovery-> server_lds.c and server_multicast.c)

RakshanPremsagar, 2021-01-19:
Oh yes, I am running on the same host. Possibly your second sentence is what is occuring in my case.

kevinherron, 2021-01-19:
I briefly toyed with writing an LDS-ME server using Milo but it never went anywhere. I found the spec to be a little bit lacking in descriptions and examples when it comes to multicast. It's a bit frustrating to work with.

RakshanPremsagar, 2021-01-19:
I agree!
 Anyway thanks for your response. I will close this issue.

closed 2021-01-19 by RakshanPremsagar