eclipse-milo/milo #752: StatusCode{name=Bad_TypeMismatch, value=0x80740000, quality=bad}

zetanet, 2021-01-29:
Hi,
I'm using milo SDK ver 0.3.4 JDK8.
I need to write a datetime (DTL) node like this when browsing server node..

```
Node=data DTL in scrittura NodeId=ExpandedNodeId{ns=4, id=43, serverIndex=0}
              Node=YEAR NodeId=ExpandedNodeId{ns=4, id=44, serverIndex=0}
              Node=MONTH NodeId=ExpandedNodeId{ns=4, id=45, serverIndex=0}
              Node=DAY NodeId=ExpandedNodeId{ns=4, id=46, serverIndex=0}
              Node=WEEKDAY NodeId=ExpandedNodeId{ns=4, id=47, serverIndex=0}
              Node=HOUR NodeId=ExpandedNodeId{ns=4, id=48, serverIndex=0}
              Node=MINUTE NodeId=ExpandedNodeId{ns=4, id=49, serverIndex=0}
              Node=SECOND NodeId=ExpandedNodeId{ns=4, id=50, serverIndex=0}
              Node=NANOSECOND NodeId=ExpandedNodeId{ns=4, id=51, serverIndex=0}

```
so inizialize my Variant node like this..

```
            List<NodeId> nodeIds = ImmutableList.of(
            new NodeId(4, 3), //Node=Q.tà in Silo 1
            new NodeId(4, 4), //Node=Q.tà in Silo 2
            new NodeId(4, 5), //Node=Q.tà in Silo 3
            new NodeId(4, 6), //Node=Q.tà in Silo 4
            new NodeId(4, 7), //Node=Q.tà in Silo 5
            new NodeId(4, 8), //Node=Q.tà in Silo 6
            new NodeId(4, 9), //Node=Q.tà in Silo 7
            new NodeId(4, 10), //Node=Q.tà in Silo 8
            new NodeId(4, 11), //Node=Q.tà in Silo 9
            new NodeId(4, 15), //Node=id commessa in            
            new NodeId(4, 12),  //Node=in Commesse
            new NodeId(4, 44),  //Node=yyyy
            new NodeId(4, 45),  //Node=MM
            new NodeId(4, 46),  //Node=dd
            new NodeId(4, 48),  //Node=HH
            new NodeId(4, 49),  //Node=mm
            new NodeId(4, 50)  //Node=ss                   
            );                    

```

```
   LocalDateTime datetime= LocalDateTime.parse(arrOfStr[10], DateTimeFormatter.ofPattern("dd/MM/yyyyHH:mm",Locale.getDefault()));               

            Variant vDTLYear = new Variant((Integer) datetime.getYear());            
            Variant vDTLMonth = new Variant((Integer) datetime.getMonth().getValue());            
            Variant vDTLDay = new Variant((Integer) datetime.getDayOfMonth());            
            Variant vDTLHour = new Variant((Integer) datetime.getHour());            
            Variant vDTLMinute = new Variant((Integer) datetime.getMinute());            
            Variant vDTLSecond = new Variant((Integer) datetime.getSecond());  

```

```
List<DataValue> valueIds=ImmutableList.of(

                    DataValue.valueOnly(vDTLYear),
                    DataValue.valueOnly(vDTLMonth),
                    DataValue.valueOnly(vDTLDay),
                    DataValue.valueOnly(vDTLHour),
                    DataValue.valueOnly(vDTLMinute),
                    DataValue.valueOnly(vDTLSecond)                    
            );            
          
```
but getting always..

[DEBUG] 2021-01-29 17:03:57.509 DataValue{value=Variant{value=100.0}, status=null}
[DEBUG] 2021-01-29 17:03:57.509 DataValue{value=Variant{value=200.0}, status=null}
[DEBUG] 2021-01-29 17:03:57.509 DataValue{value=Variant{value=0.0}, status=null}
[DEBUG] 2021-01-29 17:03:57.509 DataValue{value=Variant{value=100.0}, status=null}
[DEBUG] 2021-01-29 17:03:57.509 DataValue{value=Variant{value=0.0}, status=null}
[DEBUG] 2021-01-29 17:03:57.509 DataValue{value=Variant{value=0.0}, status=null}
[DEBUG] 2021-01-29 17:03:57.509 DataValue{value=Variant{value=0.0}, status=null}
[DEBUG] 2021-01-29 17:03:57.509 DataValue{value=Variant{value=0.0}, status=null}
[DEBUG] 2021-01-29 17:03:57.509 DataValue{value=Variant{value=0.0}, status=null}
[DEBUG] 2021-01-29 17:03:57.509 DataValue{value=Variant{value=121}, status=null}
[DEBUG] 2021-01-29 17:03:57.509 DataValue{value=Variant{value=Ordine nr.729/20 - Finissimo di coke sfuso}, status=null}
[DEBUG] 2021-01-29 17:03:57.509 DataValue{value=Variant{value=2020}, status=null}
[DEBUG] 2021-01-29 17:03:57.509 DataValue{value=Variant{value=12}, status=null}
[DEBUG] 2021-01-29 17:03:57.509 DataValue{value=Variant{value=24}, status=null}
[DEBUG] 2021-01-29 17:03:57.509 DataValue{value=Variant{value=14}, status=null}
[DEBUG] 2021-01-29 17:03:57.509 DataValue{value=Variant{value=2}, status=null}
[DEBUG] 2021-01-29 17:03:57.509 DataValue{value=Variant{value=0}, status=null}
[DEBUG] 2021-01-29 17:03:57.509 StatusCode{name=Good, value=0x00000000, quality=good}
[DEBUG] 2021-01-29 17:03:57.509 StatusCode{name=Good, value=0x00000000, quality=good}
[DEBUG] 2021-01-29 17:03:57.509 StatusCode{name=Good, value=0x00000000, quality=good}
[DEBUG] 2021-01-29 17:03:57.509 StatusCode{name=Good, value=0x00000000, quality=good}
[DEBUG] 2021-01-29 17:03:57.509 StatusCode{name=Good, value=0x00000000, quality=good}
[DEBUG] 2021-01-29 17:03:57.509 StatusCode{name=Good, value=0x00000000, quality=good}
[DEBUG] 2021-01-29 17:03:57.509 StatusCode{name=Good, value=0x00000000, quality=good}
[DEBUG] 2021-01-29 17:03:57.509 StatusCode{name=Good, value=0x00000000, quality=good}
[DEBUG] 2021-01-29 17:03:57.509 StatusCode{name=Good, value=0x00000000, quality=good}
[DEBUG] 2021-01-29 17:03:57.509 StatusCode{name=Good, value=0x00000000, quality=good}
[DEBUG] 2021-01-29 17:03:57.509 StatusCode{name=Good, value=0x00000000, quality=good}
[DEBUG] 2021-01-29 17:03:57.509 StatusCode{name=Bad_TypeMismatch, value=0x80740000, quality=bad}
[DEBUG] 2021-01-29 17:03:57.509 StatusCode{name=Bad_TypeMismatch, value=0x80740000, quality=bad}
[DEBUG] 2021-01-29 17:03:57.509 StatusCode{name=Bad_TypeMismatch, value=0x80740000, quality=bad}
[DEBUG] 2021-01-29 17:03:57.509 StatusCode{name=Bad_TypeMismatch, value=0x80740000, quality=bad}
[DEBUG] 2021-01-29 17:03:57.509 StatusCode{name=Bad_TypeMismatch, value=0x80740000, quality=bad}
[DEBUG] 2021-01-29 17:03:57.509 StatusCode{name=Bad_TypeMismatch, value=0x80740000, quality=bad}

How can write this kind of node???...

Thanks.

kevinherron, 2021-01-29:
You'll need to read the DataType attribute (or view it in another client like UaExpert) for all of these Nodes to know what type of value they are expecting.

zetanet, 2021-02-05:
Hi, installed UAExpert on Customer server and browed the desired not called "data DTL in scrittura" witch is **DTL** data type 

[image]

it's like a complex value with subnode YEAR,MONTH,DAY etc...

I can see that single node ex. DAY,MONTH,HOUR,MINUTE,SECOND is **Byte** data type 

[image]

so i've changed my code to..

            Variant vDTLMonth = new Variant(Byte.parseByte(String.valueOf(datetime.getMonth().getValue())));            
            Variant vDTLDay = new Variant(Byte.parseByte(String.valueOf(datetime.getDayOfMonth())));            
            Variant vDTLHour = new Variant(Byte.parseByte(String.valueOf(datetime.getHour())));            
            Variant vDTLMinute = new Variant(Byte.parseByte(String.valueOf(datetime.getMinute())));            
            Variant vDTLSecond = new Variant(Byte.parseByte(String.valueOf(datetime.getSecond())));            

but writing this single node value always give me ..

[DEBUG] 2021-02-05 10:45:22.775 StatusCode{name=Bad_TypeMismatch, value=0x80740000, quality=bad}

i dont' undestand how write this single node.

Tnks.

kevinherron, 2021-02-05:
An OPC UA `Byte` is an unsigned value. You need to use `UByte` from Milo instead of a Java `Byte`, which is signed and would be an OPC UA `SByte`.

zetanet, 2021-02-05:
Tnks @kevinherron ..

following your hint (except for YEAR with is 2 byte long = UShort), the error message changed in ..

[DEBUG] 2021-02-05 14:51:29.110 StatusCode{name=Bad_NotWritable, value=0x803B0000, quality=bad}
[DEBUG] 2021-02-05 14:51:29.110 StatusCode{name=Bad_NotWritable, value=0x803B0000, quality=bad}
[DEBUG] 2021-02-05 14:51:29.110 StatusCode{name=Bad_NotWritable, value=0x803B0000, quality=bad}
[DEBUG] 2021-02-05 14:51:29.110 StatusCode{name=Bad_NotWritable, value=0x803B0000, quality=bad}
[DEBUG] 2021-02-05 14:51:29.110 StatusCode{name=Bad_NotWritable, value=0x803B0000, quality=bad}
[DEBUG] 2021-02-05 14:51:29.110 StatusCode{name=Bad_NotWritable, value=0x803B0000, quality=bad}

seems to be the single subnode is not writable ...

Any way to write all the 12 bytes of DTL structure???. a string like DTL#yyyy-mm-dd-hh:mm:ss.sss ???

[image]

tnks again.

kevinherron, 2021-02-05:
It might be possible but a little difficult. 

Before trying to get into figuring out of you can read or write the structure value from Milo you should see if UaExpert is able to write the whole structured value or not.

zetanet, 2021-02-05:
Actually i can read the single subnode value witch is written by PCL.. but is very difficoult to write it...
OK. l'll try to write directly with UaExpert.

zetanet, 2021-02-05:
OWWWWWWWWWW.. UAExpert give me same error on single node YYYY..

Write to node 'NS4|Numeric|44' failed [ret = BadNotWritable]._

kevinherron, 2021-02-05:
What about on the structure itself, which for some reason UaExpert read as `Null`? `ns=4;i=30` in your first screenshot.

zetanet, 2021-02-05:
yes.. ns=4;i=30  ... Value = Null on whole structure.. i don't undestand..

kevinherron, 2021-02-05:
Sorry, sounds like a problem with this server and it’s not going to allow you to write to any of these Nodes.

zetanet, 2021-02-05:
Tnks @kevinherron ..
l'll ask to PCL programmer to verify the writable configuration of this node..
for now.. tnks from Italy... 👍 ..

closed 2021-02-07 by kevinherron