eclipse-milo/milo #753: Milo-client "hangs" when server "dies"

dkahlert, 2021-01-30:
We have a client app using milo 0.5.3 that read data from a server permanently. It looks like the connection to the server is not stable. (The server version or used library version is unknown)
Extract from the logfile:
2021-01-29 12:59:58,751 DEBUG o.e.m.o.s.c.s.OpcUaSubscriptionManager [milo-shared-thread-pool-5] getTimeoutHint() maxKeepAlive=5000.0 maxPendingPublishes=2 timeoutHint=15000.0
2021-01-29 12:59:58,751 DEBUG o.e.m.o.s.c.s.OpcUaSubscriptionManager [milo-shared-thread-pool-5] Sending PublishRequest, requestHandle=3003, acknowledgements=[id=39/seq=2498]
2021-01-29 12:59:59,760 DEBUG o.e.m.o.s.c.s.OpcUaSubscriptionManager [milo-shared-thread-pool-6] Received PublishResponse, sequenceNumber=2499
2021-01-29 12:59:59,760 DEBUG o.e.m.o.s.c.s.OpcUaSubscriptionManager [milo-shared-thread-pool-5] onPublishComplete() response for subscriptionId=39
2021-01-29 12:59:59,760 DEBUG o.e.m.o.s.c.s.OpcUaSubscriptionManager [milo-shared-thread-pool-5] [id=39] PublishResponse sequence=2499, available sequences=[id=39/seq=2499]
2021-01-29 12:59:59,760 DEBUG o.e.m.o.s.c.s.OpcUaSubscriptionManager [milo-shared-thread-pool-5] onPublishComplete(), subscriptionId=39, sequenceNumber=2499, publishTime=DateTime{utcTime=132563952072346990, javaDate=Fri Jan 29 13:00:07 CET 2021}
2021-01-29 12:59:59,760 DEBUG o.e.m.o.s.c.s.OpcUaSubscriptionManager [milo-shared-thread-pool-6] Received 11 MonitoredItemNotifications
2021-01-29 12:59:59,760 DEBUG d.w.c.o.u.e.DataReceiver [milo-shared-thread-pool-6] subscription value received: ...

2021-01-29 12:59:59,760 DEBUG o.e.m.o.s.c.s.OpcUaSubscriptionManager [milo-shared-thread-pool-5] getTimeoutHint() maxKeepAlive=5000.0 maxPendingPublishes=2 timeoutHint=15000.0
2021-01-29 12:59:59,760 DEBUG o.e.m.o.s.c.s.OpcUaSubscriptionManager [milo-shared-thread-pool-5] Sending PublishRequest, requestHandle=3004, acknowledgements=[id=39/seq=2499]
<no publish response>

2021-01-29 14:00:02,910 DEBUG o.e.m.o.s.c.SessionFsm [milo-shared-thread-pool-12] [2] S(CreatingWait)          x E(CreatingWaitExpired)   = S'(Creating)            
2021-01-29 14:00:02,910 DEBUG o.e.m.o.s.c.SessionFsm [milo-shared-thread-pool-12] [2] Sending CreateSessionRequest...
2021-01-29 14:00:02,910 DEBUG o.e.m.o.s.c.t.t.OpcTcpTransport [milo-netty-event-loop-1] Write failed, channel closed; retrying...
2021-01-29 14:00:03,911 DEBUG o.e.m.o.s.c.t.t.OpcTcpTransport [milo-netty-event-loop-1] Write failed, request=CreateSessionRequest, requestHandle=3904
2021-01-29 14:00:03,911 DEBUG o.e.m.o.s.c.UaStackClient [milo-shared-thread-pool-12] sendRequest() failed, request=CreateSessionRequest, requestHandle=3904
java.util.concurrent.CompletionException: java.nio.channels.ClosedChannelException
	at java.util.concurrent.CompletableFuture.encodeRelay(Unknown Source) ~[?:1.8.0_131]
	at java.util.concurrent.CompletableFuture.completeRelay(Unknown Source) ~[?:1.8.0_131]
	at java.util.concurrent.CompletableFuture.uniRelay(Unknown Source) ~[?:1.8.0_131]
	at java.util.concurrent.CompletableFuture$UniRelay.tryFire(Unknown Source) ~[?:1.8.0_131]
	at java.util.concurrent.CompletableFuture.postComplete(Unknown Source) ~[?:1.8.0_131]
	at java.util.concurrent.CompletableFuture.completeExceptionally(Unknown Source) ~[?:1.8.0_131]
	at org.eclipse.milo.opcua.stack.client.transport.AbstractTransport.lambda$sendRequest$6(AbstractTransport.java:107) ~[?:?]
	at io.netty.util.concurrent.DefaultPromise.notifyListener0(DefaultPromise.java:577) ~[?:?]
	at io.netty.util.concurrent.DefaultPromise.notifyListenersNow(DefaultPromise.java:551) ~[?:?]
	at io.netty.util.concurrent.DefaultPromise.notifyListeners(DefaultPromise.java:490) ~[?:?]
	at io.netty.util.concurrent.DefaultPromise.setValue0(DefaultPromise.java:615) ~[?:?]
	at io.netty.util.concurrent.DefaultPromise.setFailure0(DefaultPromise.java:608) ~[?:?]
	at io.netty.util.concurrent.DefaultPromise.tryFailure(DefaultPromise.java:117) ~[?:?]
	at io.netty.channel.AbstractChannel$AbstractUnsafe.safeSetFailure(AbstractChannel.java:993) ~[?:?]
	at io.netty.channel.AbstractChannel$AbstractUnsafe.write(AbstractChannel.java:865) ~[?:?]
	at io.netty.channel.DefaultChannelPipeline$HeadContext.write(DefaultChannelPipeline.java:1367) ~[?:?]
	at io.netty.channel.AbstractChannelHandlerContext.invokeWrite0(AbstractChannelHandlerContext.java:715) ~[?:?]
	at io.netty.channel.AbstractChannelHandlerContext.invokeWriteAndFlush(AbstractChannelHandlerContext.java:762) ~[?:?]
	at io.netty.channel.AbstractChannelHandlerContext$WriteTask.run(AbstractChannelHandlerContext.java:1089) ~[?:?]
	at io.netty.util.concurrent.AbstractEventExecutor.safeExecute(AbstractEventExecutor.java:164) ~[?:?]
	at io.netty.util.concurrent.SingleThreadEventExecutor.runAllTasks(SingleThreadEventExecutor.java:472) ~[?:?]
	at io.netty.channel.nio.NioEventLoop.run(NioEventLoop.java:500) ~[?:?]
	at io.netty.util.concurrent.SingleThreadEventExecutor$4.run(SingleThreadEventExecutor.java:989) ~[?:?]
	at io.netty.util.internal.ThreadExecutorMap$2.run(ThreadExecutorMap.java:74) ~[?:?]
	at java.lang.Thread.run(Unknown Source) [?:1.8.0_131]
Caused by: java.nio.channels.ClosedChannelException
	at io.netty.channel.AbstractChannel$AbstractUnsafe.newClosedChannelException(AbstractChannel.java:957) ~[?:?]
	... 11 more
2021-01-29 14:00:03,911 DEBUG o.e.m.o.s.c.SessionFsm [milo-shared-thread-pool-12] [2] CreateSession failed: java.nio.channels.ClosedChannelException
java.util.concurrent.CompletionException: java.nio.channels.ClosedChannelException
...
2021-01-29 14:00:03,911 DEBUG o.e.m.o.s.c.SessionFsm [milo-shared-thread-pool-10] [2] S(Creating)              x E(CreateSessionFailure)  = S'(CreatingWait)        
2021-01-29 14:00:19,911 DEBUG o.e.m.o.s.c.SessionFsm [milo-shared-thread-pool-10] [2] S(CreatingWait)          x E(CreatingWaitExpired)   = S'(Creating)            
2021-01-29 14:00:19,911 DEBUG o.e.m.o.s.c.SessionFsm [milo-shared-thread-pool-10] [2] Sending CreateSessionRequest...
2021-01-29 14:00:19,911 DEBUG o.e.m.o.s.c.t.t.OpcTcpTransport [milo-netty-event-loop-1] Write failed, channel closed; retrying...
2021-01-29 14:00:20,912 DEBUG o.e.m.o.s.c.t.t.OpcTcpTransport [milo-netty-event-loop-1] Write failed, request=CreateSessionRequest, requestHandle=3905
2021-01-29 14:00:20,912 DEBUG o.e.m.o.s.c.UaStackClient [milo-shared-thread-pool-10] sendRequest() failed, request=CreateSessionRequest, requestHandle=3905
java.util.concurrent.CompletionException: java.nio.channels.ClosedChannelException
	at java.util.concurrent.CompletableFuture.encodeRelay(Unknown Source) ~[?:1.8.0_131]
	at java.util.concurrent.CompletableFuture.completeRelay(Unknown Source) ~[?:1.8.0_131]
	at java.util.concurrent.CompletableFuture.uniRelay(Unknown Source) ~[?:1.8.0_131]
	at java.util.concurrent.CompletableFuture$UniRelay.tryFire(Unknown Source) ~[?:1.8.0_131]
	at java.util.concurrent.CompletableFuture.postComplete(Unknown Source) ~[?:1.8.0_131]
	at java.util.concurrent.CompletableFuture.completeExceptionally(Unknown Source) ~[?:1.8.0_131]
	at org.eclipse.milo.opcua.stack.client.transport.AbstractTransport.lambda$sendRequest$6(AbstractTransport.java:107) ~[?:?]
	at io.netty.util.concurrent.DefaultPromise.notifyListener0(DefaultPromise.java:577) ~[?:?]
	at io.netty.util.concurrent.DefaultPromise.notifyListenersNow(DefaultPromise.java:551) ~[?:?]
	at io.netty.util.concurrent.DefaultPromise.notifyListeners(DefaultPromise.java:490) ~[?:?]
	at io.netty.util.concurrent.DefaultPromise.setValue0(DefaultPromise.java:615) ~[?:?]
[... 4 lines cut ...]
	at io.netty.channel.DefaultChannelPipeline$HeadContext.write(DefaultChannelPipeline.java:1367) ~[?:?]
	at io.netty.channel.AbstractChannelHandlerContext.invokeWrite0(AbstractChannelHandlerContext.java:715) ~[?:?]
	at io.netty.channel.AbstractChannelHandlerContext.invokeWriteAndFlush(AbstractChannelHandlerContext.java:762) ~[?:?]
	at io.netty.channel.AbstractChannelHandlerContext$WriteTask.run(AbstractChannelHandlerContext.java:1089) ~[?:?]
	at io.netty.util.concurrent.AbstractEventExecutor.safeExecute(AbstractEventExecutor.java:164) ~[?:?]
	at io.netty.util.concurrent.SingleThreadEventExecutor.runAllTasks(SingleThreadEventExecutor.java:472) ~[?:?]
	at io.netty.channel.nio.NioEventLoop.run(NioEventLoop.java:500) ~[?:?]
	at io.netty.util.concurrent.SingleThreadEventExecutor$4.run(SingleThreadEventExecutor.java:989) ~[?:?]
	at io.netty.util.internal.ThreadExecutorMap$2.run(ThreadExecutorMap.java:74) ~[?:?]
	at java.lang.Thread.run(Unknown Source) [?:1.8.0_131]
[... cut ...]
2021-01-29 14:00:36,912 DEBUG o.e.m.o.s.c.t.t.OpcTcpTransport [milo-netty-event-loop-1] Write failed, channel closed; retrying...
2021-01-29 14:00:37,913 DEBUG o.e.m.o.s.c.UaStackClient [milo-shared-thread-pool-12] sendRequest() failed, request=CreateSessionRequest, requestHandle=3906
java.util.concurrent.CompletionException: java.nio.channels.ClosedChannelException

At this point, there is no return from the Exception. The subscription listener is defined but will not be reached. Haw can this be handled?

Thanks.
dkahlert.

kevinherron, 2021-01-30:
Could you possibly turn the ChannelFsm logger on DEBUG or TRACE as well? And maybe get a Wireshark capture while this is happening?

dkahlert, 2021-02-01:
Hi, trace is enabled, We'll waiting for the next error.
Best regards
dkahlert.

kevinherron, 2021-02-17:
We can re-open this if/when the issue occurs again or there is more information.

closed 2021-02-17 by kevinherron