eclipse-milo/milo #758: OPC UA Client state unknown

sensubha, 2021-02-08:
When connect to OPC UA TIA server, OPC UA Client is running and exchanging data for 7-8 hours. After that getting error in client as "OPC UA Client state unknown" and tag reading is stopped. What would be the probable reason and how to resolve?

kevinherron, 2021-02-08:
You’ll have to provide a lot more information, log messages, Wireshark captures, etc...

“OPC UA Client state unknown” is not even a message that originates front his library.

sensubha, 2021-02-09:
2021-02-09 02:30:49.5388 DEBUG [OPC]: Object NodeId: ns=3;s="OPCMethodAdhData_IDB", Method NodeId: ns=3;s="OPCMethodAdhData_IDB".Method.
2021-02-09 02:30:49.5520 DEBUG [OPC]: Object NodeId: ns=3;s="OPCMethodAdhData_IDB", Method NodeId: ns=3;s="OPCMethodAdhData_IDB".Method. Successful
2021-02-09 02:30:49.5520 DEBUG [OPC] Monitored tag event received: 127
2021-02-09 02:30:49.5520 DEBUG [OPC]: Object NodeId: ns=3;s="OPCMethodAdhData_IDB", Method NodeId: ns=3;s="OPCMethodAdhData_IDB".Method.
2021-02-09 02:30:49.6293 ERROR BadInvalidState
Opc.Ua.ServiceResultException: BadInvalidState
   at Opc.Ua.Client.Session.Call(NodeId objectId, NodeId methodId, Object[] args)
   at CUS.L1A.Handlers.OPCHandler.OPCHandler.CallAdhMethod(String methodIdString, String objectIdString) in D:\ProcessExpert\Standard\Src\CUS.Handlers\CUS.L1A.Handlers\OPCHandler\OPCHandler.cs:line 589
   at CUS.L1A.Managers.L1SignalManager.CallAdhMethod() in D:\ProcessExpert\Standard\Src\CUS.Managers\CUS.L1A.Managers\L1SignalManager.cs:line 720
2021-02-09 02:30:49.6762 DEBUG [OPC]: Object NodeId: ns=3;s="OPCMethodAdhData_IDB", Method NodeId: ns=3;s="OPCMethodAdhData_IDB".Method. Successful
2021-02-09 02:30:49.6762 DEBUG [OPC] Monitored tag event received: 130
2021-02-09 02:30:49.6762 DEBUG [OPC]: Object NodeId: ns=3;s="OPCMethodAdhData_IDB", Method NodeId: ns=3;s="OPCMethodAdhData_IDB".Method.
2021-02-09 02:30:49.6918 ERROR [OPC] Error during calling ADH method

sensubha, 2021-02-09:
When restart the program, it again start tag reading and opc method calling.

kevinherron, 2021-02-09:
These log messages are not from this library. I think you may be in the wrong place. Looks like .NET code.

sensubha, 2021-02-09:
I am getting the error when I tried to connect S7 1500 PLC opc server from C# client which is developed using OPC UA Client

kevinherron, 2021-02-09:
This is a Java library.

closed 2021-02-09 by kevinherron