eclipse-milo/milo #760: getNode() throws status=Bad_NodeClassInvalid (not seems ok in other clients)

faljse, 2021-02-11:
We are developing a simple client to test/benchmark our opcua server.
I'm somewhat lost in this opc-ua stuff, so this might not be a bug at all (sorry).

Calling getNode() on most of the nodes in our namespaces throws an exception:
"status=Bad_NodeClassInvalid, description=The node class is not valid."

https://github.com/eclipse/milo/blob/7d1a6ef8ba649964b3a7177e40b7ba1b5468867c/opc-ua-sdk/sdk-client/src/main/java/org/eclipse/milo/opcua/sdk/client/AddressSpace.java#L806
getValue() returns null.

Ist this an issue with our namespace, the way i query the node, or something else?

The nodes seem to work fine in uaexpert (but they do have some invalid arguments)
client lib is milo 5.3

kevinherron, 2021-02-11:
Hmm, well it sounds like your server is not returning a valid value for NodeClass, which isn't an optional attribute.

What does UaExpert show in the attributes pane for this Node?

faljse, 2021-02-15:
The error ("UaException: status=Bad_NodeClassInvalid, message=The node class is not valid.")
is thrown if the nodeid isnt found in the namespace.
`UaNode node = as.getNode(NodeId.parse("ns=4;s=asdf"));`

Is this the expected behavior? I would prefer a more specific error message.

kevinherron, 2021-02-15:
I agree a better error message should be returned if possible. I have prepared an easy fix for this.

kevinherron, 2021-02-15:
Pending PR: https://github.com/eclipse/milo/pull/769

Was this the cause of your original issue or do you still need assistance?

faljse, 2021-02-15:
Yes, that caused my original issue; Thx!

closed 2021-02-15 by kevinherron