eclipse-milo/milo #775: [Feature] Support for module-info.java structure (aka Jigsaw)

Adito5393, 2021-02-22:
Thanks for sharing your work with the rest of the world! 

I will look into the source code these days and attempt to add [a module-info.java](http://openjdk.java.net/projects/jigsaw/) file for the client part (and all the components that come with it).

Have you tried adding this before? 

Atm, I am looking into module jakarta.activation that gives the following error in my module java application:

```bash
Error occurred during initialization of boot layer
java.lang.module.ResolutionException: Modules org.eclipse.milo.opcua.sdk.client.dictionaryreader and 
org.eclipse.milo.opcua.sdk.client export package 
org.eclipse.milo.opcua.sdk.client to module jakarta.activation
```

A possible explanation could be [this one](https://github.com/eclipse-ee4j/jaf/issues/38).

Would you be interested in a future PR, if it all works out?

kevinherron, 2021-02-22:
I don't know if adding `module-info.java` will solve your problem (not sure what will). 

`Automatic-Module-Name` entries are already being made in `MANIFEST.MF` files when the JARs are built.

You're welcome to try it and see, but be aware that this library must currently compile and run on JDK 8 so I think you'd have to configure multi-release JARs so that the `module-info.java` can exist independent of the rest of the code compiled for JDK 8.

Adito5393, 2021-02-22:
Good to know you want to continue the support of JDK 8. Then I will include the multi-release JARs pom.xml configuration before creating the PR.

I will also try using the [latest jakarta.activation v2.0.0](https://mvnrepository.com/artifact/com.sun.activation/jakarta.activation/2.0.0) that supports [module definition](https://github.com/eclipse-ee4j/jaf/blob/master/activation/src/main/java/module-info.java). Do you have any technical reasons for sticking with 1.2.1? (otherwise, I will find out while building & testing the new jars).

kevinherron, 2021-02-22:
> Do you have any technical reasons for sticking with 1.2.1? (otherwise, I will find out while building & testing the new jars).

I think it's largely because that's what was available at the time, and because that's the version `jaxb-runtime` depends on.

```
[INFO] --- maven-dependency-plugin:2.8:tree (default-cli) @ bsd-core ---
[INFO] org.eclipse.milo:bsd-core:jar:0.5.5-SNAPSHOT
[INFO] +- org.eclipse.milo:stack-core:jar:0.5.5-SNAPSHOT:compile
[INFO] |  +- org.bouncycastle:bcprov-jdk15on:jar:1.61:compile
[INFO] |  +- org.bouncycastle:bcpkix-jdk15on:jar:1.61:compile
[INFO] |  +- com.google.guava:guava:jar:26.0-jre:compile
[INFO] |  +- io.netty:netty-codec:jar:4.1.45.Final:compile
[INFO] |  |  +- io.netty:netty-common:jar:4.1.45.Final:compile
[INFO] |  |  +- io.netty:netty-buffer:jar:4.1.45.Final:compile
[INFO] |  |  \- io.netty:netty-transport:jar:4.1.45.Final:compile
[INFO] |  |     \- io.netty:netty-resolver:jar:4.1.45.Final:compile
[INFO] |  +- io.netty:netty-handler:jar:4.1.45.Final:compile
[INFO] |  \- org.slf4j:slf4j-api:jar:1.7.25:compile
[INFO] +- com.sun.activation:jakarta.activation:jar:1.2.1:compile
[INFO] \- org.glassfish.jaxb:jaxb-runtime:jar:2.3.2:compile
[INFO]    +- jakarta.xml.bind:jakarta.xml.bind-api:jar:2.3.2:compile
[INFO]    +- org.glassfish.jaxb:txw2:jar:2.3.2:compile
[INFO]    +- com.sun.istack:istack-commons-runtime:jar:3.0.8:compile
[INFO]    +- org.jvnet.staxex:stax-ex:jar:1.8.1:compile
[INFO]    +- com.sun.xml.fastinfoset:FastInfoset:jar:1.2.16:compile
[INFO]    \- jakarta.activation:jakarta.activation-api:jar:1.2.1:compile
```

kevinherron, 2021-02-22:
I see there's a version `3.0.0` of `jaxb-runtime` available now too.

Any of these dependencies can be updated as needed as long as nothing breaks, though I'd probably merge those version changes into the `dev/0.6` branch.

Adito5393, 2021-02-22:
Great, I will use that branch for my local fork also to bump the versions!

Compiling my application now with your `0.5.4` version, I get this error:

```bash
Error occurred during initialization of boot layer java.lang.module.ResolutionException: 
Modules org.eclipse.milo.opcua.sdk.client and
org.eclipse.milo.opcua.sdk.client.dictionaryreader 
export package org.eclipse.milo.opcua.sdk.client 
to module org.eclipse.milo.opcua.binaryschema.parser
```

This makes me think that bumping the versions of other jar dependencies to module-info support is not enough and I need module-info within the parent structure. 

I see that [the jakarta.activation's pom.xml](https://github.com/eclipse-ee4j/jaf/blob/master/pom.xml) also defines a multi-release jar within the maven-compiler-plugin. I will attempt something similar...

kevinherron, 2021-02-22:
Does anything happen if you change the automatic module name from `org.eclipse.milo.opcua.sdk.client.dictionaryreader` to something like `org.eclipse.milo.opcua.sdk.dictionaryreader`?

Adito5393, 2021-02-22:
Thanks for the recommendation!

I get similar error, now regarding `module netty.channel.fsm`:

```bash
Error occurred during initialization of boot layer java.lang.module.ResolutionException: 
Modules org.eclipse.milo.opcua.sdk.client and 
org.eclipse.milo.opcua.sdk.dictionaryreader 
export package org.eclipse.milo.opcua.sdk.client 
to module netty.channel.fsm
```

The automatic module name is exporting all packages from within and looks like `sdk.client` and `dictionaryreader` contain the same package names within. I will try to rename `dictionaryreader` packages and see how that goes....

kevinherron, 2021-02-22:
If that works, then it's possible the original automatic module name given to the `dictionary-reader` module was fine but its classes need to be in a package that matches rather than in `org.eclipse.milo.opcua.sdk.client`.

Clearly I haven't done any projects with the module system :)

Adito5393, 2021-02-22:
Yeah, that was the issue. 

Exactly as you thought, the name of the module should match the package it contains, according to [the OpenJDK quick start](http://openjdk.java.net/projects/jigsaw/quick-start): "By convention, the source code for the module is in a directory that is the name of the module."

So far I only had to modify `dictionary-reader` and `dictionary-manager` modules.

kevinherron, 2021-02-22:
So it seems all that's needed is to refactor the package name for `dictionary-reader` and `dictionary-manager` for the 0.6 release then? 

No need for `module-info.java` and multi-release JARs?

Adito5393, 2021-02-22:
This is enough to run any application build on top of your library as a module using module-path to run it.

Unfortunately, this is not enough to really enjoy the Java Platform Module System (JPMS) benefits, like using the [jlink tool](https://docs.oracle.com/javase/9/tools/jlink.htm) to generate a stand-alone application (no JDK required). 

In the end, that's what I would like to achieve! And I look forward to contributing.

But for now, I am getting an error from `bouncycastle` lib: `java.lang.NoClassDefFoundError: org/bouncycastle/cert/X509v3CertificateBuilder`
I am running the [ReadExample.java](https://github.com/eclipse/milo/blob/master/milo-examples/client-examples/src/main/java/org/eclipse/milo/examples/client/ReadExample.java) main. I have modified `ClientExample.java` to use the `SecurityPolicy.Aes128_Sha256_RsaOaep` and `getEndpointUrl()` returns the URL of a real OPC UA server I have on the network online (with username and password defined in the `KeyStoreLoader.java`). I don't know what `IdentityProvider` I should return with the `getIdentityProvider()` method, given my use case.

Any recommendations on how to figure this out?

kevinherron, 2021-02-22:
The `IdentityProvider` you use doesn't really have anything to do with the `SecurityPolicy` you choose.

Each endpoint will enumerate what kinds of authentication are allowed. You can use `CompositeProvider` to build one that works for Anonymous and Username/Password (or whatever else).

kevinherron, 2021-02-22:
I don't know if you'll ever get to fully realize JPMS benefits and use jlink until all Milo dependencies are also using explicit modules. Not sure what the status of that is.

I use the [Badass Runtime Plugin](https://badass-runtime-plugin.beryx.org/releases/latest/) to generate a stand-alone application with bundled runtime. Seems to work well. It's used in https://github.com/digitalpetri/opc-ua-demo-server

Adito5393, 2021-02-22:
Thanks for the hint, never heard or worked with Badass Runtime Plugin, I will look into it.

I have manage to fix the `bouncycastle` issue --> for anyone in the future having problems with it, all I had to do was to add the `requires org.bouncycastle.pkix;` line to my module-info and add the pom dependency for maven `Maven: org.bouncycastle:bcpkix-jdk15on:1.61`.

Now I'm getting some errors with the netty framework... issues with the communication

Adito5393, 2021-02-23:
I found another clash of package naming from 2 modules:

* org.eclipse.milo.opcua.sdk.core
* org.eclipse.milo.opcua.sdk.server

They both contain: `org.eclipse.milo.opcua.sdk.server.util` package. Which one do you prefer to rename?

kevinherron, 2021-02-23:
Move `GroupMapCollate` into `org.eclipse.milo.opcua.sdk.core.util`

Adito5393, 2021-02-25:
Just a quick update on the issue: 

I have currently managed to write the `module-info.java` files for all the relevant modules of the project (skipping the build-tools and the test/integration modules). The changes are available on [my fork (dev/0.6.500 branch)](https://github.com/Adito5393/milo/tree/dev/0.6.500). I've used the `jdeps` tool provided by Java11 via a [nice GUI](https://github.com/lanthale/ModitectGen) developed for the [moditect maven plugin](https://github.com/moditect/moditect).

The project builds without errors, only warnings, I will look into those, these days.

kevinherron, 2021-02-26:
I think I'd like to see this as 2 PRs to keep things clean and reduce the chance that the introduction of `module-info.java` breaks the build or something else.

1st PR should have all the package name changes to get automatic module names working.
2nd PR, based on the work in the 1st, should attempt to introduce `module-info.java` and multi-release JAR files. This one I'm more worried about.

kevinherron, 2021-03-05:
@Adito5393 did you get anywhere with this?

closed 2021-03-09 by kevinherron

Adito5393, 2021-03-16:
Sorry for the late reply. 

I was able to build the module-info.java with jdk11 from a seperated directory and the rest is jdk8. It passed all the tests. 

The problem was 2 dependencies jars that were old and the maven compiler plugin had to inject an automatic module name and printed out a warning not to publish the jar publicly. (later i will edit this msg and add the jar names...) 

I didn't had time to fully test the jars within my application such that i can verify that the module-info is indeed useful. It is definitely useful for when you will bump the Java version. 

I can make the PR for the dev/0.7 if you will soon release 0.6 version. This week I expect to have time to make some progress.

kevinherron, 2021-03-16:
Okay, targeting the next version should be fine. 

I made all the necessary package changes and refactoring for automatic module names to hopefully work in the 0.6 release.

Adito5393, 2021-03-16:
Alright, I will update my fork with the latest changes from the remote side and manually resolve conflicts. And keep an eye for when the `dev/0.7`appears 😃 

I will open a new issue if I encounter any problems with the current `dev/0.6`. 

Regarding the `jsr305-3.0.2.jar` dependency, would you be willing to switch to [the jetbrains annotations](https://github.com/JetBrains/java-annotations) (as mention in a [previous issue](https://github.com/eclipse/milo/pull/99)) because it is still maintained and offers an automatic-module-name?

kevinherron, 2021-03-16:
It's not too late to change these dependencies for 0.6. The JetBrains annotations are fine for now. In the future I hope that [jspecify](https://github.com/jspecify/jspecify) succeeds and that everybody can use that instead.

`strict-machine` and `netty-channel-fsm` are both my projects so I can publish new versions that have automatic module names.

Adito5393, 2021-03-16:
Awesome! Thanks for willing to help out!

Adito5393, 2021-03-18:
Hey @kevinherron why did you choose to use org.jetbrains:annotations:16.0.2? (latest version available is 20.1.0)

Looking at [the CHANGELOG](https://github.com/JetBrains/java-annotations/blob/master/CHANGELOG.md) I can see they added `Automatic-Module-Name` to MANIFEST.MF in version 16.0.3.

I'm now locally installing your master branch and I will test it within my modular application.

Adito5393, 2021-03-18:
I have some issues with Intellij IDE at the moment with updating the new module names for jars. The IDE does not recognize the `org.eclipse.milo.opcua.sdk.client.dtd` module. Doing the official cache clean (File | Invalidate Caches/Restart) did not solve it.

I manually remove the **Maven** and **index** folder from `%LocalAppData%\JetBrains\IdeaIC2020.3\` and the IDE still does not load the new name. 

I think it would be better if you do not name it `org.eclipse.milo.opcua.sdk.client.dtd` because this implies that it is within the `org.eclipse.milo.opcua.sdk.client` module. This is also related with the package naming that you use, that matches the module name. 

PS: At the moment this is only an IDE issue. If I run `mvn compile` it compiles without any issues. I also tried running the class with `mvn exec` as a module and it runs fine (an example that does not use `org.eclipse.milo.opcua.sdk.client.dtd.DataTypeDictionarySessionInitializer`)

kevinherron, 2021-03-18:
I think I got the version from this page: https://www.jetbrains.com/help/idea/annotating-source-code.html

I'll update version to 20.1.0.

kevinherron, 2021-03-18:
`dictionary-manager` and `dictionary-reader` use sub packages of `server` and `client` respectively because logically they _are_ (and _were_) part of those. They require the server or client and were only moved to separate Maven modules because they come with heavy XML dependencies that one day I'd like to make optional.

Adito5393, 2021-03-18:
Okey, no need to change the module name or the package structure since my issue is an IDE issue and probably a cache issue.

kevinherron, 2021-03-18:
@Adito5393 I am going to publish a 0.6.0-M1 release today if there are no other issues relating to automatic module names.

kevinherron, 2021-03-19:
[This article](https://www.theoryofgeek.com/articles/fomo-java-module-edition) is somewhat compelling to me. I am leaning towards having correctly configured automatic module names and staying away from actual modularization and using `module-info.java` for now.

Adito5393, 2021-03-29:
The big difference I see between the article's application and `milo` is that he is heavily dependent on Spring.io while you have a standalone application (except for the dependency on the netty framework). 

The article's developer tackles the `java module` file creation as it was something to be fixed (via automation) instead of understanding what each line means and what it is useful for. I see this opportunity as a way to better understand the library API: what packages are supposed to be used by the user and which ones are for internal use. Even though the file also tells you the libraries dependencies (via `requires`) this is not meant to replace the Maven dependency system, but to work together. 

I will be more than happy to keep the `module-info.java` support alive in the snapshot phase until netty implements natively `module-info.java`. It will also give me more time to test it and make changes accordingly. This way I will be able to switch to stable mode at any moment and use the `automatic module name` for troubleshooting.

This way you will have a starting point that was tested and verified by someone who is using your library. Towards the end of the development, together we can choose what packages should each module export. For now, I will export everything and only have a working Proof Of Concept.

Do you think I could automate the merging process from your dev branch to my dev branch via Github actions? Or it would be better to handle it manually for the possible merge conflicts that could appear?

kevinherron, 2021-03-29:
I don't know if GitHub Actions can do it for you but it shouldn't be too hard to maintain by hand.

PR #784 (2021-03-09): Fix compatibility issues with JPMS
dictionary-manager and dictionary-reader needed to use package names that
matched their automatic module names and GroupMapCollate needed to be moved to
an appropriate package for the module it resides in.

fixes #775
commit fe994779 (2021-03-09): Fix compatibility issues with JPMS (#784)

dictionary-manager and dictionary-reader needed to use package names that
matched their automatic module names and GroupMapCollate needed to be moved to
an appropriate package for the module it resides in.

fixes #775