eclipse-milo/milo #779: BigInteger.longValueOf() not implemented in Android

GerkeBF, 2021-03-05:
**Problem**
Thanks for sharing your work with the rest of the world!

I am using the milo library in an Android Project. Unfortunately the method BigInteger.longValueOf() is not yet implemented in Android. It will be added with Android 12, which will probably run stable in about a year. 

The exact usage is in OpcUaSubscriptionManager on line 226. 

**Logs**
java.lang.NoSuchMethodError: No virtual method longValueExact()J in class Ljava/math/BigInteger; or its super classes (declaration of 'java.math.BigInteger' appears in /apex/com.android.art/javalib/core-libart.jar)
        at org.eclipse.milo.opcua.sdk.client.subscriptions.OpcUaSubscriptionManager.getLifetimeCount(OpcUaSubscriptionManager.java:226)
        at org.eclipse.milo.opcua.sdk.client.subscriptions.OpcUaSubscriptionManager.lambda$LUctLZChikw3npQWbJvSPICy8NM(Unknown Source:0)
        at org.eclipse.milo.opcua.sdk.client.subscriptions.-$$Lambda$OpcUaSubscriptionManager$LUctLZChikw3npQWbJvSPICy8NM.apply(Unknown Source:10)
        at org.eclipse.milo.opcua.sdk.client.subscriptions.OpcUaSubscriptionManager.createSubscription(OpcUaSubscriptionManager.java:149)
        at org.eclipse.milo.opcua.sdk.client.subscriptions.OpcUaSubscriptionManager.createSubscription(OpcUaSubscriptionManager.java:110)

kevinherron, 2021-03-05:
I'm hesitant to change this because I have no intentions of supporting Android and there will certainly just be another thing that needs changing after this one...

If you fork Milo, modify it to avoid this usage, and then build your application using the fork do you run into other errors or does it work?

kevinherron, 2021-03-05:
I think it could be modified to look like this:
```java
    private UInteger getLifetimeCount(double publishingInterval, UInteger maxKeepAliveCount) {
        // Lifetime must be 3x (or greater) the keep-alive count.
        BigInteger lifetimeCount = maxKeepAliveCount
            .toBigInteger()
            .multiply(BigInteger.valueOf(6))
            .min(BigInteger.valueOf(UInteger.MAX_VALUE));

        return uint(lifetimeCount.longValue());
    }
```

GerkeBF, 2021-03-08:
Thanks for responding that fast. As I see, you created a pull request and plan to have this in Version 0.6.0, correct? Can you give me a rough estimation when the pull request will be merged?

kevinherron, 2021-03-08:
This week, sooner if you can verify it fixes the issue for you and let me know if there are any others.

PR #783 (2021-03-09): Avoid the use of BigInteger::longValueExact
This function doesn't exist on Android (yet) and there is an easy enough
alternative implementation of getLifetimeCount.

fixes #779
commit f2906ad2 (2021-03-09): Avoid the use of BigInteger::longValueExact (#783)

This function doesn't exist on Android (yet) and there is an easy enough
alternative implementation of getLifetimeCount.

fixes #779
closed 2021-03-09 by kevinherron