eclipse-milo/milo #788: Nodeset loading compatibility

faljse, 2021-03-12:
Tried to load our nodeset with the code mentioned in #713 
Seems to work fine, but in the end the namespace is empty.
Nodes are created and they get added to the namespace, but nothing visible to the client.
Do i need to add a rootnode or sth?
Should this code still be compatible with head?

Our nodeset xml:
https://gist.github.com/faljse/bf698f0ea27953f68ffc274bb2cde9a5

kevinherron, 2021-03-12:
Hard to say what’s wrong without seeing your code. Did you create a Namespace implementation based on the examples linked to in the nodeset-parser repository? Has that Namespace been added to the server?

What are you expecting your client to see and where are you browsing? Most of these information models just define a bunch of types, which you would only see if you started browsing the type hierarchy from the client. You wouldn’t see instances of these types unless you’ve instantiated and added them yourself.

faljse, 2021-03-12:
The Nodeset works in open62541;
Rootnode is "Machines" (folder) with "MachineTool1" (object) as child
"MachineTool1" is described in the xml, "Machines" is not.
So could that be the issue; i have to add this root folder manually somehow?

Namespace, etc:
https://gist.github.com/faljse/be5b8a35afbfa2b3dbd5df3761bd206c#file-mynamespace-java
Namespace as seen from open62541:
[image]

GoetzGoerisch, 2021-03-12:
The _Machines_ folder is defined by the [OPC 40001-1 UA for Machinery 9](https://reference.opcfoundation.org/v104/Machinery/v101/docs/9.1/).
In your addressspace this companion spec is missing.

kevinherron, 2021-03-12:
This model references THREE other models that you are missing.

kevinherron, 2021-03-12:
Also I have just updated the ModeledAddressSpace and BacnetNamespace gists for compatibility with the current Milo 0.5.x release.

https://gist.github.com/kevinherron/b4268328c3daeec74343ee82b872aef0
https://gist.github.com/kevinherron/3215a4c624f35fdec0d75faaa442f1b3

faljse, 2021-03-12:
indeed...  :)
any hints on how to load/merge multiple noadset files?

kevinherron, 2021-03-12:
Your example also does not need these lines:
```
            DataTypeInitializer.initialize(
                    server.getNamespaceTable(),
                    server.getDataTypeManager()
            );
            ObjectTypeInitializer.initialize(
                    server.getNamespaceTable(),
                    server.getObjectTypeManager()
            );
            VariableTypeInitializer.initialize(
                    server.getNamespaceTable(),
                    server.getVariableTypeManager()
            );
```
In the BacnetNamespace example those are references to initializers that were generated for BACnet types with a prototype code generator. They are not the initializers from the stack and you should not need to do this.

You don't need to merge NodeSet files because you aren't generating code. You should be able to get away with just loading each NodeSet separately, i.e. create a Namespace implementation based on `ModeledAddressSpace` for each of the NodeSets and add them all to your server.

faljse, 2021-03-12:
Thank you, seems to be working fine now!

kevinherron, 2021-03-12:
Support for modeling and code generation will take priority hopefully in the near future. I need to add OPC UA 1.04 support first.

kevinherron, 2021-03-12:
I put together an example server that loads all of these namespaces: https://github.com/kevinherron/machine-example

The models do not define any _instances_, but the Machines folder and others are present.

[image]

faljse, 2021-03-12:
For now we would be very happy if we could connect some VariableNodes from that namespace to an external datasource.
I havent found a read() method or any callbacks in the node itself; so im lost again :(

kevinherron, 2021-03-12:
Well, the easiest way to hook into the read behavior of a Node is using an AttributeFilter: https://github.com/eclipse/milo/blob/f581575cfbf53e0b7d4528ef0bbd515cf7781930/milo-examples/server-examples/src/main/java/org/eclipse/milo/examples/server/ExampleNamespace.java#L486-L492

This likely will not scale very far though. More efficient read/write must be handled by in your AddressSpace implementation by overriding `AttributeServices::read` and `AttributeServices::write` methods and deciding what the best approach is based on the Node(s) being read or written.

GoetzGoerisch, 2021-03-12:
Sample Machine Tool Model can be found [here](https://github.com/OPCFoundation/UA-Nodeset/blob/v1.04/MachineTool/Machinetool-Example.xml).

faljse, 2021-03-12:
we need to block in getValue; which is our main issue with single threaded open62541
is this an issue in milo also?

kevinherron, 2021-03-12:
You can block but it won't scale very well. That is where you need to consider implementing the AddressSpace read/write methods yourself. There is no example of this unfortunately.

kevinherron, 2021-03-12:
Thank you @GoetzGoerisch. I will see if I can load this model as well. What is your relationship with OPC UA / 62541 / modeling? Always interesting to find others who are working in this area...

faljse, 2021-03-12:
Does that mean its also only one thread or is there some worker mechanism?

GoetzGoerisch, 2021-03-12:
https://github.com/GoetzGoerisch/GoetzGoerisch/blob/main/README.md

kevinherron, 2021-03-12:
No the server uses a thread pool, but if a request to read many nodes comes in and all of the nodes have a blocking AttributeFilter installed then the response may be slow because each AttributeFilter will have getValue called sequentially. 

A smart implementation of your AddressSpace would recognize when Nodes are being read belong to an external source, would know how to batch them together, and would then read the underlying data in some efficient manner and complete then complete the request. Without using an AttributeFilter.

kevinherron, 2021-03-12:
@GoetzGoerisch your model seems to load 👍 
[image]

However this is where the modeling support in Milo starts to fall apart. Since these instances were loaded from a modeled namespace rather than instantiated in a programmatically created AddressSpace/Namespace there is no easy opportunity to install behavior on the instances as they are created...

closed 2021-03-17 by kevinherron