eclipse-milo/milo #805: Variant cannot contain DataValue

nicolatimeus, 2021-03-24:
**Describe the bug**
We have the requirement to read the Value attribute of a Variable node whose DataType is DataValue, this seems to fail due to the following assertion while decoding the ReadResponse:

https://github.com/eclipse/milo/blob/60ac9ebfe700ed15ab342f4d2b10553533290abf/opc-ua-stack/stack-core/src/main/java/org/eclipse/milo/opcua/stack/core/types/builtin/Variant.java#L47

Currently we tried to use the `UaClient.read()` method for this:

https://github.com/eclipse/milo/blob/60ac9ebfe700ed15ab342f4d2b10553533290abf/opc-ua-sdk/sdk-client/src/main/java/org/eclipse/milo/opcua/sdk/client/api/services/AttributeServices.java#L53

Is there another way for reading such value?
If not, is the assumption that a Variant cannot contain a DataValue correct? 

**Additional context**
Client: Milo 0.2.4 (the same assertion seems to be also on master)
Server: https://help.emsprocloud.com/index.html?opc-ua.html

kevinherron, 2021-03-24:
This is explicitly disallowed by the spec.

From UA 1.03, Part 6:
> DataValue and DiagnosticInfo types only have meaning when returned in a response message
with an associated StatusCode. As a result, Variants cannot contain instances of DataValue or
DiagnosticInfo. This requirement means that if an Attribute supports the writing of a null value
it shall also support writing of an empty Variant and vice versa.

and further clarified in UA 1.04:
> Values of Attributes are always returned in instances of DataValues. Therefore, the DataType
of an Attribute cannot be a DataValue. Variants can contain DataValue when used in other
contexts such as Method Arguments or PubSub Messages. The Variant in a DataValue cannot,
directly or indirectly, contain another DataValue.

kevinherron, 2021-03-24:
I am going to remove this check for the 0.6.0 release, however I would recommend you report this to and push back against your server vendor because a VariableNode is not allowed to have a DataType of DataValue.

closed 2021-03-24 by kevinherron

nicolatimeus, 2021-03-24:
Thank you for the quick reply and your clarifications.

PR #806 (2021-03-24): Remove check preventing Variant construction with DataValue
OPC UA 1.04 further clarifies that it is legal to construct a Variant
containing a DataValue for use in e.g. method arguments or PubSub but it is
not legal for a DataValue to contain such a Variant. Without knowledge of
where the Variant will be used it does not make sense to restrict construction
of such a Variant.

fixes #805
commit 53210d79 (2021-03-24): Remove check preventing Variant construction with DataValue (#806)

OPC UA 1.04 further clarifies that it is legal to construct a Variant
containing a DataValue for use in e.g. method arguments or PubSub but it is
not legal for a DataValue to contain such a Variant. Without knowledge of
where the Variant will be used it does not make sense to restrict construction
of such a Variant.

fixes #805